revk/xESP32-Faikout #566: [BUG]

matteustace, 2024-11-28:
**Faikin hardware**
3 Faikins purchased from Amazon recently and 1 purchased less recently

**Daikin hardware**
S21 lead to FXM-N units

**Describe the bug**
Some of the sensors, notably the liquid temperatures and fan Hz readings, possibly some others too, seem to be intermittent over MQTT. Having just power cycled the air conditioning, all of the Faikins are now appearing in Home Assistant as not including those sensors any more, but the readings were fine prior to the reboot. The readings do appear fine on the Faikin webpage, but seemingly are not being provided to Home Assistant, or Home Assistant isn't detecting them, or something. If this is a Home Assistant MQTT bug then please feel free to reject.

**To Reproduce**
Steps to reproduce the behavior:
1. Power cycle air conditioning
2. Observe missing stats in Home Assistant

**Expected behavior**
I expected to still see these sensors. Other sensors are stlil working.

**Additional context**
Thankyou for creating these excellent little devices.

revk, 2024-11-29:
Can you try latest beta.

matteustace, 2024-11-30:
Just tried the beta on an additional one (one of the set I bought was DoA, then replacement got lost by the courier, now I have another replacement but the snap lines on the PCB are partially missing.... )

Took approximately 7 hours to start reporting comp and fanfreq metrics but everything else worked straight away.

revk, 2024-11-30:
I have issued the new code now, so same as beta.

But weird if there is a delay like that. Is it now Ok though?

matteustace, 2024-11-30:
They all seem to have a several hour delay after a reboot before they report those metrics, all the other metrics work straight away. But once they start then it's fine until the faikin is next rebooted. It just means that if I use them in automations, the automations break for several hours.... not a major issue just a bit strange!

revk, 2024-11-30:
I really am puzzled by that. I'll recheck here with HA (I don't really use HA) and see if I can find the cause. We had found one before but had fixed it, I thought.

matteustace, 2024-11-30:
So I've just updated Home Assistant Core and suddenly the items that were missing after a power cycle earlier today have re-appeared.... so I am wondering if this is a Home Assistant bug rather than a Faikin one....

revk, 2024-12-01:
Well possibly a combination of the two. I know the Faikin was, in one release, not sending all sensors initially. Maybe HA was caching something. If it is now working, that is great.