revk/xESP32-Faikout #574: [BUG] Heating over Target

Ulf349, 2024-12-11:
**Faikin hardware**
Self created PCB with esp32-s3-wroom-1_wroom-1u

**Daikin hardware**
Daikin FVXM25R with S21 lead and parallel working Onecta to Daikin cloud.

**Describe the bug**
Faikin auto mode ist used with +/- 0,5°C on target 20°C.
It worked from 30.November very good. From 08. December the temperature increase over the target approx 3°C. I couldn´t understand why that happens so I turned off the Faikin auto mode with the Off button on the Web page and after some hours I turned the Faikin auto mode on, back to +/-0,5°C. Same procedure, it overheats the target.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Faikin web page'
2. Click on 'Faikin auto mode off'
3. Waiting some hours
4. Click on 'Faikin auto mode on +/-0,5°C'

**Expected behavior**
I could try to restart the Faikin but I didn´t do that yet.

**Additional context**
Here you can see the increased room temperature, logged every 10 seconds. The value is from the BLE temperarure sensor, placed on the wall in 1,5m height, approx 5m distance from the Daikin. The sensor is not in the direct fan flow.
[image]

revk, 2024-12-11:
Ok a lot of settings and controls you can play with to do whatever you want there, and documentation of how it works.

closed 2024-12-11 by revk

revk, 2024-12-11:
Best wishes working it out - if you think a bug, please do say.

Ulf349, 2024-12-15:
Solution found. After changing the Rx under Tx to other pins the connection over S21 died.
I changed some values over Onecta. Waiting some minutes and changed back to the correct Rx and Tx.
From that point the Faikin works great again. 
I don't know why that happened