revk/xESP32-Faikout #576: [FEATURE] Set external temp via MQTT (not just BLE)

GeertvanHorrik, 2024-12-12:
**Daikin hardware**

S21, purchased in official UK store (via Amazon)

**Is your feature request related to a problem? Please describe.**

The BLE external temp idea is great. But I would like to submit it via MQTT instead of BLE. I have temp sensors in every room where I can use the average to more reliably update current climate info. I could, for example, use HA & Node Red to update it every 30 seconds (or on every change, depending on the number of changes).

**Describe the solution you'd like**

Allow the external room temp to be set via MQTT. 

**Describe alternatives you've considered**

-

**Additional context**

I (think I) checked the MQTT options. Getting the temp is supported, setting it (the env temp) is not.

revk, 2024-12-12:
What, you mean like it explains in https://github.com/revk/ESP32-Faikin/blob/main/Manuals/Advanced.md

[image]

closed 2024-12-12 by revk

GeertvanHorrik, 2024-12-15:
Thank you for your response.

I am looking at these parameters:

https://github.com/revk/ESP32-Faikin/blob/main/Manuals/Advanced.md#aircon-control

I am confused about the difference between `temp`, `target` and `env`. If I understand correctly, the `target` is *what we want it to be*, either 1 or 2 (range) values. `env` is the environment temp, but is this only used by Faikin auto mode or also set to Daikin as "environment temp"? If both of my assumptions are correct, then what is `temp`?

revk, 2024-12-15:
You want just `target` an `env`. IIRC `temp` sets the temp target in the Daikin like many other controls. `env` and `target` allow Faikin auto control to work and set the various settings as needed.