revk/xESP32-Faikout #579: [Question] Communication over Rest API like legacy Daikin Wifi module

cmonty14, 2024-12-14:
Hi,
I'm aware that this Faikin-module can communicate over MQTT.
However I would like to address this question:
Is communication over Rest API like legacy Daikin Wifi module working, too?

THX

revk, 2024-12-15:
Pretty much all of the simple legacy http Daikin functions work, so generally legacy integrations still work.

Raspdomo, 2024-12-27:
First of all: many many thanks for this Faikin module. I have 8 indoor units (on 2 multisplit outdoor units) - one of them had no LAN API. After installing a Faikin module the integration with Domoticz (based on the HTTP LAN interface - not via MQTT) was dead easy and works like a charm. Now my whole system is fully automated at any time of the year.

There's one thing that I really miss: the Domoticz devices created to collect heating and cooling power consumption remain at zero. That is probably due to the **/aircon/get_day_power_ex** not being implemented in Faikin.
This API returns the consumption per hour in multiples of 100W:
ret=OK,curr_day_heat=0/0/0/0/0/1/0/3/3/2/5/0/6/4/0/0/0/0/0/0/0/0/0/0,prev_1day_heat=0/0/0/0/0/0/1/1/1/2/0/0/2/3/0/1/3/0/3/1/5/4/0/0,curr_day_cool=0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0,prev_1day_cool=0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0

> Would it be possible to add this to the development backlog ?

revk, 2024-12-27:
That sounds like something we would have to collect and store rather that something we can simply extract from the aircon. Not out of the question but very much not the way the Faikin works at present. So I'll have to ponder this one.

Raspdomo, 2024-12-27:
It didn't occur to me before but indeed the Faikin unit only manages "current" settings.
There are two other functionalities /get_week_power and /get_year_power that also contain historical information but I think Domoticz only uses the /get_day_power_ex and then from there keeps history inside the Domoticz db.

I wonder how this is managed in the Daikin LAN module - it must store some history about power consumption, roughly about 130 values:

- daily: 24 heating values per hour, 24 cooling values, keeping also the values of yesterday (see above)

- weekly: 7 values, heating and cooling combined:
ret=OK,today_runtime=0,datas=10500/10100/11100/10100/7900/9500/10400

- yearly: 12 values heating and cooling combined, keeping also the values of last year
ret=OK,previous_year=490/287/240/107/0/28/0/0/0/76/229/322,this_year=461/182/196/129/13/31/79/98/24/72/240/243

revk, 2024-12-28:
Indeed, it would not be impossible to store such values. I try and avoid using the flash unnecessarily, and storing in RAM would lose data on a restart - but that may be reasonably OK as these do not normally get reset often (occasional s/w updates).

It would probably be simpler to collect using HA as that will happily store and graph the values over time I believe.

Raspdomo, 2024-12-28:
### Keep only 96 values in RAM ?
My guess is that Domoticz only uses _/get_day_power_ex_.
If you only implement _/get_day_power_ex_ you could keep the values in RAM instead of flash. that would be 24 x 4 = 96 values Too bad if these data are lost on a (rare) restart.

### ... or only 48 ??
I'm not even sure if Domoticz uses the values of yesterday, so an option would be to include those in the HTTP response but hardcoded at zero. That way there would only be 48 values (24 heat and 24 cool) to keep.

A real HTTP response captured today at 15:30 is (though I overwrote prev_1day_heat values with zero):
ret=OK,curr_day_heat=0/1/0/0/0/0/0/3/2/3/3/4/1/0/1/0/0/0/0/0/0/0/0/0,prev_1day_heat=0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0,curr_day_cool=0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0,prev_1day_cool=0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0
The first 15 _curr_day_heat_ values are real consumption data, the remaining 9 are still at zero as yet to come.

And this is the response at 16:10. Now the consumption of the 16th hour of the day is also filled in (value 6 = 600kWh).
ret=OK,curr_day_heat=0/1/0/0/0/0/0/3/2/3/3/4/1/0/1/**_6_**/0/0/0/0/0/0/0/0,prev_1day_heat=0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0,curr_day_cool=0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0,prev_1day_cool=0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0
So each hour a value is added - the remaining values for the day still at zero, the previous values stay the same.

### ... or maybe even only one ???
Maybe Domoticz only picks the value of the "current minus one" hour and adds it to the domoticz.db to keep history. That way there would only be one value to be stored in RAM (or 2: one for heat and one for cool). The HTTP response for the 16th hour today in heating mode could then be
ret=OK,curr_day_heat=0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/**_6_**/0/0/0/0/0/0/0/0,prev_1day_heat=0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0,curr_day_cool=0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0,prev_1day_cool=0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0/0

I'm always willing to beta-test a few bits and ends if you decide to give it a go :-) With a bit of luck the "only one" option is enough !

revk, 2024-12-28:
That is probably lot too hard.

Raspdomo, 2024-12-28:
Then maybe let's start from the beginning: is there any way at all to get any power consumption information over the S21 interface ?

revk, 2024-12-29:
Yes it reports it. We already do that. I think it's a cumulative counter so could easily be partitioned on the hour.

Raspdomo, 2024-12-29:
Academic precision isn't required, so losing the stats for a couple of hours isn't a showstopper at all.

In the github documentation I only found something about energy in the S21.md manual: _GM | Hex, Wh total energy usage_.
Couldn't find any other info (sorry if I overlooked anything).

- Is this value available via MQTT only ? (which I haven't installed on my system - I use the standard Domoticz Daikin integration via HTTP)
- What does this value represent ? E.g. the Grand Total of the energy usage since the Daikin unit was first started, or today's usage, or ... ?

revk, 2024-12-30:
At present MQTT (which reports using MQTT to HA) as far as I am aware.

And yes, a grand total energy!

So if we logged this every hour to a table we could provide per hour usage total.

MuzzaMcD, 2024-12-30:
Just to add a little value to this topic of the Daikin Energy Counter from my observation only...
It is my experience that the Energy Counter is only changed when it accumulates 100Wh or 0.1kWh. So when running idle or indeed mode=Off/Standby (ie low power consumption) you will only see a change to the value over an extended period of time. It can be confusing when you see a jump in the graph and wonder if the unit was turned on randomly ? ! ! 
ALSO, the counter value is 'hard written' (to survive power down) even less often than 100Wh. Not sure but maybe every 1000Wh? So sometimes if the unit powers off for some reason, you will see the next reported value DECREASE a notch, which can be a pain in the bottom for HA statistics and graphs etc.

Raspdomo, 2024-12-30:
@MuzzaMcD : at first the 0.1kWh increments confused me a bit too, For units that are turned off I typically see one increment per day. It's the power usage of the WiFi module + other circuit board. I never witnessed a decrease (yet).

The Daikin HTTP _/aircon/get_day|week|year_power_ex_ responses differentiate between heating and cooling. I wonder how that is done if the S21 port only provides a single grand total value...

@revk How does it currently work ? Do you publish one MQTT message per hour with total power usage ?
If so, even with one single grand total coming in it should be possible to keep track of per hour usage in HA/Domoticz. Differentiation between heating and cooling should also be possible as it is known in which mode the system was during the last hour (apart from the rare/weird occasion where there would have been both heating and cooling over the last hour).

Raspdomo, 2024-12-30:
I set up an MQTT broker and connected the Faikin. Didn't connect to Domoticz yet, I first want to observe a bit.

Under the Faikin header I found the actual data - being updated every 60 seconds. There's effectively an energy value, labeled Wh.
Looking at the message history (see below) there are some strange things to observe:
1. Mostly the message contains one Wh value, but sometimes there are three values.
2. The values are around 4535000. That can't possibly be the "lifetime energy" of the unit. The unit is very rarely on and typically consumes around 50kW per year (I kept track of it from 2022 till mid 2024, as long as I used Onecta). Its lifetime energy usage at this moment should rougly be somewhere between 175 and 225kWh.
3. In a timespan of exactly one hour the value increased by 700. The unit being off, that can't possibly be 0,7kWh. It might be 7Wh though, times 24 that gives 168 and we know that a unit that is off uses around 100-200Wh per day.
But then still I can't explain the total of around 4535000... unless it is the total of this year which I expect to be around 45kWh! Coincidence ? If in two days this value falls down to zero we'll know for sure.

MQTT Faikin message log between 16:15 and 17:00 (only those messages where the Wh value changed):
- 16:15: "Wh": 4535400
- 16:19: "Wh": 4535500
- 16:27: "Wh": [ 4535500, 4535598, 4535600 ]
- 16:28:  "Wh": 4535600
- 16:34: "Wh": [ 4535600, 4535698, 4535700 ]
- 16:42: "Wh": [ 4535700, 4535796, 4535800 ]
- 16:43: "Wh": 4535800
- 16;51: "Wh": [ 4535800, 4535896, 4535900 ]
- 16:52: "Wh": 4535900
- 17:00: "Wh": [ 4535900, 4535995, 4536000 ]
- ...
- 17:15: "Wh": 4536100

revk, 2024-12-31:
These are my regular stats reports - you can change the format - but normally its is one value if only that value for period, or three (min, ave,max) if it changed during the period. It may be better to set live status updates to see more clearly.

Raspdomo, 2024-12-31:
After further observation today I'm convinced that the reported Wh accounts for the whole multisplit system - i.e. all indoor units combined.
> Anybody with a multisplit and more than one unit with a Faikin ? Do they all report the same Wh value ?

There's a difference with what Daikin reports (through HTTP into Domoticz). The Faikin total is roughly 10% lower (observed today as well as for the whole lifetime). I'm not looking for academic precision so either will do for me. Though I have more trust in the Faikin value :-)

MuzzaMcD, 2025-01-01:
@Raspdomo I am not running any multi splits so I can't help with that. However, I do have a Shelly Power Meter attached to one of my Faikin controlled units.  
The Faikin does report a raw energy value from the Daikin that is a little lower than the measured Energy from the Shelly 1PM.  I have always put this down to the 100Wh 'precision' that the Daikin delivers.
My experience with various power/energy integrations before I moved to Faikin was that some attempt to improve the accuracy of the Raw Energy data by 'estimating the slope' and applying a compensating factor. (You might like to google and cast your eyes over the pydaikin/power.py code to see an example. I also believe that the Raw Data is used when 'logically' dividing/attributing energy usage to heat/cool and even days/weeks etc. 
I am currently using Shelly Power meters where appropriate to measure Power and store Energy data and more accurately apply Electricity tariff costs to work out true cost  .... but that's a whole other story :)
I think history data like Power and Energy is best stored centrally like in Home Assistant. 
I suspect that the Daikin Raw Energy counter is simply implemented by Daikin to primarily record how hard a compressor unit has worked over its lifetime.  ... and not necessarily to give us users any accurate energy data :)

Raspdomo, 2025-01-02:
I come to the same conclusions.

Raspdomo, 2025-01-04:
In Faikin.c line
` jo_string (j, "prevr_year_cool", "0/0/0/0/0/0/0/0/0/0/0/0");`
should be
` jo_string (j, "prev_year_cool", "0/0/0/0/0/0/0/0/0/0/0/0");`
though this doesn't cause any trouble in Domoticz.

I went through the code trying to understand a bit more before asking just any silly question here - but my C++ is **very** rusty  :-)

Reason is that I'm getting weird behaviour in MQTT. I'm using MQTT explorer to observe the Mosquitto MQTT broker on my Raspberry.
A couple of hours ago all MQTT config messages (for homeassistant autodiscovery) had disappeared. In the meantime some are back (energy, demand, sensor, comfort, quiet, econo), but there are no config messages for powering on/off, neither for indoor and outdoor temperature, fan settings, mode, liquid, compressor frequency, ...).
Note that I had switched on haswitches and then off again.
Initially there was also had a topic Faikin, but that one has completely disappeared from the MQTT broker also.

Furthermore the state topic doesn't receive any new messages (those used to arrived every minute) -  EXCEPT when I change something e.g. switching nofaikinauto on. I then get exactly four messages in only 2 seconds time:

> {"ts":"2025-01-04T**16:10:07**Z","id":"983DAEF01904","up":true,"uptime":6142,"mqtt-up":6140,"**restart":2**,"reason":"Settings changed (Change nofaikinauto)","online":true,"power":false,"target":16.00,"temp":18.50,"outside":2.50,"liquid":35.50,"demand":55,"Wh":4600100,"fanfreq":0.0,"comp":30,"mode":"off","fan":"auto","streamer":true,"quiet":false,"econo":false,"comfort":false,"powerful":false,"sensor":false,"swing":"off","preset":"home"}

> {"ts":"2025-01-04T**16:10:08**Z","id":"983DAEF01904","up":true,"uptime":6143,"mqtt-up":6141,"**restart":1**,"reason":"Settings changed (Change nofaikinauto)","online":true,"power":false,"target":16.00,"temp":18.50,"outside":2.50,"liquid":35.50,"demand":55,"Wh":4600100,"fanfreq":0.0,"comp":30,"mode":"off","fan":"auto","streamer":true,"quiet":false,"econo":false,"comfort":false,"powerful":false,"sensor":false,"swing":"off","preset":"home"}

> {"ts":"2025-01-04T**16:10:09**Z","id":"983DAEF01904","up":true,"uptime":6144,"mqtt-up":6142,"**restart":0**,"reason":"Settings changed (Change nofaikinauto)","online":true,"power":false,"target":16.00,"temp":18.50,"outside":2.50,"liquid":35.50,"demand":55,"Wh":4600100,"fanfreq":0.0,"comp":30,"mode":"off","fan":"auto","streamer":true,"quiet":false,"econo":false,"comfort":false,"powerful":false,"sensor":false,"swing":"off","preset":"home"}

> {"ts":"2025-01-04T**16:10:09**Z",**"up":false**,"reason":"Settings changed (Change nofaikinauto)"}

Switching livestatus on or off doesn't make a difference.
I'm still using the HTTP interface to interact with Domoticz, I'm only observing the MQTT messages before I switch to the MQTT interface.
On the Faikin web interface as well as in Domoticz via the HTTP interface everything seems normal.
The unit is an FTXM20.

I have no idea what's going on...

revk, 2025-01-05:
Very rusty, as this is C not C++ :-)
Typo fixed.

But no idea why the normal MQTT updates would have stopped, that is weird. I don't even have a setting to do that - if we are connected to MQTT and online with the aircon (as we clearly are) you should get the updates.

I mean check the settings on web pages anyway incase somehow something has changed.

Failing that maybe a factory reset (can be done on latest boards by shorting the two circuit pads on the back three times).

Raspdomo, 2025-01-05:
I also looked into the Daikin implementation of Domoticz (Daikin.cpp, meaning C++ I dare to guess :-))
Instead of having
`jo_string (j, "curr_year_heat", "0/0/0/0/0/0/0/0/0/0/0/0");`
would it be possible to return
`jo_string (j, "curr_year_heat", "[Wh]/0/0/0/0/0/0/0/0/0/0/0");`
where [Wh] = the value  that comes out of the S21 interface (not multiplied by 100).
That would allow to get the Wh into Domoticz through the standard HTTP interface. The fact that there's no differentiation per month doesn't matter as Domoticz adds up the monthly values from the HTTP response anyway. And the fact that there's no differentiation between heat and cool isn't all that important...

As for the MQTT questions: this is what I've been doing.
- Domoticz addressed the Faikin module through the HTTP interface but on port 20 instead of 80 for a couple of minutes (I made a mistake). Gave error messages in Domoticz. Though I can't see how this could have influenced the MQTT thing, I mention it for the sake of completeness - and because around that time (I think ?) the MQTT stuff got messed up.
- haenable on/off
- haswitches on/off => I tried this one after the config messages had disappeared. I think switching this on put config messages for _sensor, comfort, quiet and econo_ into the MQTT. I bit later the energy and demand config topics also appeared, and that's it.
- livestatus on/off => didn't make any difference
- debug on/off => no difference
- reporting: changed value from 60 to 120, waited a couple of minutes, and back to 60 => no difference

Switches nodemand until noquiet are all off. noled and nofaikinauto are on.
All I can think of is something wrong with the value of _daikin.status_known_ but I can't dump or log it.

I'm on _Faikin-S3-MINI-N4-R2: 9f65e5b 2024-11-30T08:32:55_ .
I can't fiddle around too much with the module itself (factory reset) as I'm reluctant to have trouble in this period as the airco's are my sole heating system in the house and I can't have anything break down during this time of the year.
A sofware factory reset option would be very handsome...

I also looked into the Daikin implementation of Domoticz (Daikin.cpp, meaning C++ I dare to guess :-))
Instead of having
`jo_string (j, "curr_year_heat", "0/0/0/0/0/0/0/0/0/0/0/0");`
would it be possible to return
`jo_string (j, "curr_year_heat", "[Wh]/0/0/0/0/0/0/0/0/0/0/0");`
where [Wh] = the value  that comes out of the S21 interface (not multiplied by 100).
That would allow to get the Wh into Domoticz through the standard HTTP interface. The fact that there's no differentiation per month doesn't matter as Domoticz adds up the monthly values anyway. And the fact that there's no differentiation between heat and cool isn't all that important...

As for the MQTT questions: this is what I've been doing.
- Domoticz addressed the Faikin module through the HTTP interface but on port 20 instead of 80 for a couple of minutes (I made a mistake). Gave error messages in Domoticz. Though I can't see how this could have influenced the MQTT thing, I mention it for the sake of completeness - and because around that time (I think ?) the MQTT stuff got messed up.
- haenable on/off
- haswitches on/off => I tried this one after the config messages had disappeared. I think switching this on put config messages for _sensor, comfort, quiet and econo_ into the MQTT. I bit later the energy and demand config topics also appeared, and that's it.
- livestatus on/off => didn't make any difference
- debug on/off => no difference
- reporting: changed value from 60 to 120, waited a couple of minutes, and back to 60 => no difference

Switches nodemand until noquiet are all off. noled and nofaikinauto are on.
All I can think of is something wrong with the value of _daikin.status_known_ or some of the constants _CONTROL_xxxxx_ being corrupted or so but I can't dump or log.
But then the weird thing is that after a reboot I do get 3 status messages in a row containing all the normal values (temps and all the stuff), followed by a message {"ts":"2025-01-05T10:09:48Z","up":false,"reason":"Settings changed (Change nofaikinauto)"}.

I'm on _Faikin-S3-MINI-N4-R2: 9f65e5b 2024-11-30T08:32:55_
I can't fiddle around too much with the module itself (factory reset) as I'm reluctant to have trouble in this period as the airco's are my sole heating system in the house and I can't have anything break down during this time of the year.
A sofware factory reset option would be very handsome...

==================================================
Besides returning the Wh as mentioned above it might be interesting to enhance the interface a bit at some point in the future:
- provide the cmpfreq through get_sensor_info instead of a dash: `ret=OK,htemp=20.50,hhum=-,otemp=10.00,err=0,cmpfreq=-`
- implementing the HTTP demand control interface:
SET: http://Faikin-IP/aircon/set_demand_control?lpw=&en_demand=1&mode=0&type=1&max_pow=[30-100]
OFF: http://Faikin-ip/aircon/set_demand_control?lpw=&en_demand=0
STATE: http://Faikin-ip/aircon/get_demand_control?lpw=
which returns _ret=OK,type=1,en_demand=[0 or 1 - if you need to hardcode something then return 1],mode=0,max_pow=[demand control value from S21],scdl_per_day=4,moc=0,tuc=0,wec=0,thc=0,frc=0,sac=0,suc=0_
but anything after the pow value can be discarded (it concerns an demand control automation in the Daikin module).

revk, 2025-01-05:
"A sofware factory reset option would be very handsome..."

You can send the `command` `factory` with payload with MAC and `Faikin`

[image]

revk, 2025-01-05:
OK beta in a moment, various things. `set_demand_control` was already in there.  Do try.

closed 2025-01-05 by revk

Raspdomo, 2025-01-05:
... again a couple of hours later ... all back to normal in MQTT ...

I haven't done anything to the Faikin module in the meantime. So about the only possibility left is that the binoculars I was watching through (MQTT explorer) gave me a quirky view. Even after after 35 years in IT I can still be flabbergasted :-)
In any case: sorry to have bothered you, and many many thanks for being so helpful and for making available Faikin to the community.

BTW: Amazon offers the Faikin module for sale in France and Germany, but unfortunately not in Belgium (where I live). Any idea why ?

revk, 2025-01-05:
Amazon are a law unto themself. Last I looked they said I need a Belgium VAT number, and then that I don't need one. We are always trying to improve the availability, but Amazon do not make anything simple. I'd keep an eye out anyway in case it magically appears.