revk/xESP32-Faikout #583: [FEATURE] Send BLE information via MQTT triggerd by change values of BLE

AndrCarl, 2024-12-19:
**Daikin hardware**
S21

**Is your feature request related to a problem? Please describe.**
No MQTT messages if AC power off

**Describe the solution you'd like**
It would be great if the BLE sensor values change would trigger a distribution via MQTT (also when AC is powerd off or not in cooling/heating mode).

revk, 2024-12-19:
Not a bad idea, I'll make a beta to report env changing as a status change. Give it a few minutes.

AndrCarl, 2024-12-20:
Cool thank you for implementing that fast :-)

Additional question: When I understand the change correctly you are now updating the parameter "env" on change BLETemp and sending a message to topic faikin
`{"protocol":"S21","ts":"2024-12-20T09:35:00Z","env":[19.38,19.39,19.41],"target":[19.00,20.00],"online":true,"control":false,"home":22.50,"heat":true,"fanrpm":0,"comp":0,"outside":[4.50,4.80,5.00],"liquid":21.00,"anglev":494,"Wh":161300,"power":false,"mode":"H","temp":20.00,"demand":30,"fan":"A","swingh":false,"swingv":false,"econo":false,"powerful":false,"comfort":false,"streamer":true,"sensor":true,"quiet":true}`

the message itself in topic state.status, where the BLE values are published is not updated; last message is from 7:50AM
`{"protocol":"S21","env":20.1,"online":true,"control":false,"home":22.5,"heat":true,"fanrpm":350,"comp":44,"outside":3.5,"liquid":21.0,"anglev":183,"Wh":161300,"power":false,"mode":"H","temp":20.0,"demand":30,"fan":"A","swingh":false,"swingv":false,"econo":false,"powerful":false,"comfort":false,"streamer":true,"sensor":true,"quiet":true,"ble":{"temp":20.14,"hum":56.51,"bat":2014},"autob":"A4C13.......","autor":0.5,"autot":19.5,"auto0":"00:00","auto1":"00:00","autop":false}`

Is it possible to publish the state.status topic as well? or send BLE in Faikin topic additionally?

Onother question to temperature: Is there a way to implement an offset for home temperature. As you see in the messages the home temp is quite wrong (home 22.5 compared to env 19.4).

revk, 2024-12-20:
The problem is that there are two type of messages, as the original code has one type and then later code for HA had another.

I could probably add `env`.

The auto mode with external reference, uses the home temp as reference for setting the target on the aircon, so does not matter if way off, generally. No I don't have an offset.

revk, 2024-12-20:
Wait the stuff has `"ble":{"temp":20.14,"hum":56.51,"bat":2014}`

Is that not what you want?

(I see `bat` is wrong, so fixing that - see next beta).

AndrCarl, 2024-12-20:
the message state.status has the BLE, but the message is not updated if the the ac indoor is powered off or not currently heating/cooling. For me it would be enough if is this message is comming regulary. Alternatively we could add the ble:"" to faiking topic as this is comming every minute.

revk, 2024-12-20:
There is a setting to force the status message on every change rather than periodically, `livestatus`. Try that.

AndrCarl, 2024-12-20:
oh, yes, 'livestatus' is working. Thank you

AndrCarl, 2024-12-20:
`"ble":{"temp":20.08,"hum":54.83,"bat":83,"volt":28.70}`
volt must be like this 2870 mV or 2.87 V

Thank you very much for support :-)

closed 2024-12-20 by AndrCarl

revk, 2024-12-20:
LOL on the volt... I think it usually works. Maybe this BLE sensor reports in an unexpected way, sorry. I mean it could be a bug in one of the formats we handle. What sensor is it?

AndrCarl, 2024-12-20:
https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html
Status: Hardware Version: LYWSD03MMC B1.5, Software Version: 4.8
Advertising type: PVVX (Custom)
Vbat: 3032 mV , Temp: 21.67°C, Humi: 52.06%,

revk, 2024-12-20:
I have checked, and the code does indeed expect mV, and so my report is out by 10. Issuing a new beta now.