revk/xESP32-Faikout #586: [BUG] mqtt commands

jsapede, 2024-12-21:
**Faikin hardware**
amazon.uk buy
Faikin-S3-MINI-N4-R2: 9f65e5b 2024-11-30T08:32:55

**Daikin hardware**
Daikin Siesta / S21

**Describe the bug**
trting to send mqtt commands, to activate / deactivate auto mode (both via autor:0 and via env temp, but nothing changes. 

**To Reproduce**

Publish mqtt command : 

either on HA  : 
[image]

or in mqttexplorer : 

[image]

but nothing changes on faikin settings and autor stays here at 0.5 : 

[image]

[image]

ive als tried using the method you gave here : https://github.com/revk/ESP32-Faikin/issues/506#issuecomment-2381272643

bot no more effect.

Am i missing something ?

revk, 2024-12-22:
Several things I thing.

The `control` message with `target` is for remote control setting, you don't *also* set Faikin auto mode, it replaces the auto mode on the web interface. But you have to send those commands regularly (every few minutes at least), e.g. from some external temperature monitoring/control. You *also* have BLE, so the remote `env` would clash with that. Which do you want, BLE or a remote control `env` level?

`autor` is a `setting` so you need to send a `setting/...` not a `command/...`. `auto0` and `auto1` have no colons. You can, of course, simply change these settings via the web interface, which is usually a lot simpler.

jsapede, 2024-12-22:
hello, sorry, maybe i misunderstood the docs.

[image]

as its written to use a "control command"

ok the idea is to use BLE and the internal auto mode of the faikin witch gives very stable temperature and to trun it on from homeassistant through mqtt command as it seems not native in the homeassistant interface.

so it would be : setting/Faikin/control with {"autor": 0.0} to deactivate auto mode ?

revk, 2024-12-22:
Why not just set using web interface, way simpler. But yes.

jsapede, 2024-12-22:
because everything is automated within homeassistant and online google calendar starts house preheating 24h before our arrival in holidays. nothing more to do than add a period of presence in the calendar and the house power up iteself
then all temeratures during staying time are automated through scheduler 

moreover i have an interrogation about the "homeassistant climate" interface for faikin. Usually in HA climates, preset "home / eco / away ..." are for temperature levels, say 21 / 20 / 16 ... but in the Faikin climate presets are used for extra modes life "boost" or "eco" witch are only power related settings but not temperature ones. Why not keeping the usual HA climate consturction ?

revk, 2024-12-22:
OK yes, try that

YSIGE123, 2025-01-14:
> because everything is automated within homeassistant and online google calendar starts house preheating 24h before our arrival in holidays. nothing more to do than add a period of presence in the [...]

Did you get this working? Sorry I'm a bit of a beginner at MQTT and I've hacked together the below from the comments above and the HA documentation. I've put this in the config yaml - but its not working. Is this how its supposed to look?

mqtt:
  - switch:
      unique_id: LivingAC_auto_switch
      name: "Living AC Auto Mode"
      state_topic: "setting/Faikin/control"
      command_topic: "setting/Faikin/control/autor"
      payload_on: "2.0"
      payload_off: "2.0"
      state_on: "ON"
      state_off: "OFF"
      retain: true

revk, 2025-01-14:
I did presets based on a request from a user. I am sure HA can set things sensibly.

YSIGE123, 2025-01-15:
Thank you for all the work you've done on this - Faikin is brilliant and its really appreciated. 

I managed to get this working after figuring out that mqtt is case sensitive and reading the response to my LivingAC/setting/autor message which was an error stating the settings password wasnt provided. I'm not sure how to include that in the JSON as I couldnt find any documentation. I then removed the settings password and then it worked perfectly with the below :) 

Thank you as I can now turn the auto mode on and off from HA without going to the web interface which I'm having trouble accessing with the http bug (havent tried the beta yet) https://github.com/revk/ESP32-Faikin/issues/299

mqtt:
  - switch:
      unique_id: LivingAC_auto_switch
      name: "Living AC Auto Mode"
      state_topic: "LivingAC/setting/autor"
      command_topic: "LivingAC/setting/autor"
      payload_on: "2.0"
      payload_off: "0.0"
      state_on: "2.0"
      state_off: "0.0"
      retain: true

revk, 2025-01-15:
When sending settings you simply need to include `"password":"..."` with your password in the settings if you have a settings password set.