revk/xESP32-Faikout #587: Help: Programming Faikin (blank ordered from JLCPCB)

sellw88, 2024-12-23:
Hi,
I ordered a set of 5 PCB's designed as per this Git from JLCPCB as I require 5 off and worked out cheaper.
I then also ordered the Tasmotizer to program.

Being here a month later with these parts to hand: I now realise this board version needs to be programmed using the TC2030 pinout.

Is there any way of programming the board using the Tasmotizer & Dupont leads I have? If I can get contact on the lads.

I've had a look and it's over £50 from the US for the TC2030 Cable.
I'm £95 into this and don't know which was to turn 😲.

Any help would be appreciated.

revk, 2024-12-23:
Sorry, no, the GPIOs for serial are no longer on the board - they caused problems sometimes shorting on back of processor, hence the change to TC2030.

Some people have soldered USB tails to TC2030, it is the standard USB layout for TC2030, so not too hard.

revk, 2024-12-23:
Well, you have the KiCAD files...

[image]

sellw88, 2024-12-23:
Thankyou! 
I appeared to be overcomplicating things by looking up other 2030 connection cable pin outs (which are nothing like that)
My mistake 🤣. 

Thankyou so much 😁.

Great project and thankyou - I've reached my witts end with HA and Onecta integration .
Have a great Christmas ☺️

closed 2024-12-23 by sellw88

revk, 2024-12-23:
Happy New Year!

sellw88, 2024-12-23:
Hi RevK,
One last one before I think I have to write the PCB's off (doh)!

I've soldered onto the connector pads, and in Windows I'm getting the USB jtag/Serial port appear for a second, disconnect, reconnect, disconnect etc.

I've tried multiple PCB's and 2 different cables with the same issue.

I've tried looking online and the only suggestions I can find are the pull up GPIO0 - then release, which isnt accessible on the PCB.

Am I doing something silly before I write these off and order precoded from you?

darylc, 2026-02-02:
For others who find this closed issue in future, I have the same problem on 5 Faikin boards from jlcpcb (pcb rev 2024-07-15) ordered in July 2024 using the TC2030-USB-NL cable, they appear to be a bin job.

revk, 2026-02-02:
I never have a problem on my Mac, but windows seems way more fussy.

These days I actually use a custom flasher board https://flasher.revk.uk

This powers on and resets USB and connects all in a controlled sequence.

darylc, 2026-02-02:
For what it's worth I tried both Windows and Linux.  Got the first one installed and working, awesome product.