revk/xESP32-Faikout #588: [FEATURE] Improve swing mode Home Assistant Integration

Floyd-Schramm, 2024-12-25:
**Daikin hardware**
Indoor Unit FTXA25C2V1BB, Faikin connected to free S21 port, original WiFi module BRP069C4x installed

**Is your feature request related to a problem? Please describe.**
After integrating the Faikin modules in Home Assistant I noticed that the swing modes are not translated correctly. As seen in the below screenshots, the modes available are named "H" instead of "Horizontal" for example. This difference to the HA default swing also causes the problem that there is no icon available for the swing mode "H".
[image]
[image]

**Describe the solution you'd like**
A better, Home Assistant specific, integration using the default swing mode values described in the official [HA docs](https://developers.home-assistant.io/docs/core/entity/climate/#swing-modes)
`SWING_OFF`, `SWING_VERTICAL`, `SWING_HORIZONTAL`, `SWING_BOTH`.
I am not sure how to integrate the comfort mode as there is no default value in HA, but all other standard Daikin swing modes should be implemented as stated above.

**Describe alternatives you've considered**
Live with the current situation, as renaming or overwriting within HA is too much trouble :D

revk, 2024-12-26:
I cannot easily test as I don't have that icon/option on my HA for some reason.

If you set `otahost` to `test.revk.uk` and turn off `otabeta` and upgrade you will see changes.

Let me know if they work, and I can make a beta.

revk, 2024-12-26:
I was able to test, and seems to work, bus my HA is not showing anything special apart from those names, e.g. `SWING_HORIZONTAL`. Does yours show any better?

I have made a beta release now, so just try a normal update to beta.

Floyd-Schramm, 2024-12-26:
If that is the case, I will have to open a feature request for Home Assistant directly.
I just tried to update to the beta release by enabling OTA beta updates, then pressing update. Now the ESP does not start anymore, and I get a Timeout when trying to reach it:
[image]

revk, 2024-12-26:
Yeh, I also applied some changes to try and get TCP to work better and concerned it may have upset things. Investigating now.

revk, 2024-12-26:
My initial testing worked, the beta did not, which is a problem.

revk, 2024-12-26:
OK, the beta is fixed - do you have the means to solder USB leads to the board, to reflash it?

revk, 2024-12-26:
This looks like a bug in ESP IDF that has caught me a few times now.

Floyd-Schramm, 2024-12-26:
I just got the Faikin Module (also pulled out the whole indoor unit in the process, which needed some replacement wall anchors apparently). Do you have a wiring diagram for which pin is which for the USB cable?

Floyd-Schramm, 2024-12-26:
Can I directly solder the leads from a USB cable to the PCB? Or do I need to use a programmer? If so, I only have ESP01 USB programmers here, does it work with this kind of programmer?

revk, 2024-12-26:
OK programming is explained end of https://github.com/revk/ESP32-RevK which can be done using Chrome.

This is pin out.

[image]

Floyd-Schramm, 2024-12-26:
[image]
[image]
Flashed with bootloader and firmware and its alive again :) Just needs to be installed again

revk, 2024-12-26:
Thanks, sorry for the hassle.

Floyd-Schramm, 2024-12-26:
Like you mentioned, there are still no Icons for the swing modes:
[image]

revk, 2024-12-26:
So do we think some other config would work? Maybe "Horizontal" not "SWING_HORIZONTAL" or some such?

revk, 2024-12-26:
If it does not help I'll change back so it is what people are used to.

But if you can find a config that works better, let me know.

gentlecat, 2024-12-26:
@revk somewhat off-topic, but seems I got unlucky with automatic upgrade timing. 2 out 3 of my Faikin boards probably got the upgrade with that bad firmware you mentioned earlier (they are no longer connecting to network). I had beta upgrades enabled at some point which is probably not the best idea in retrospect. I'm assuming the only way out of this is to re-flash the firmware?

I saw [these instructions for re-flashing](https://github.com/revk/ESP32-Faikin/blob/main/Manuals/Setup.md#re-flash-firmware), but don't have a USB to serial converter around. I do have a Raspberry Pi though, just don't know if I can use it to flash ESP32 or how to do it, if that's possible. Do you have any pointers?

Floyd-Schramm, 2024-12-26:
I was able to directly solder an cut USB cable to the module. This is probably dependent on the board version however.
[image]
From my understanding, using an raspberry pi as a serial programmer should work.

Floyd-Schramm, 2024-12-26:
> So do we think some other config would work? Maybe "Horizontal" not "SWING_HORIZONTAL" or some such?

We could try "Horizontal", but my guess is that the Homeassistant internal link from SWING_HORIZONTAL to a display name and a icon is not yet made.

gentlecat, 2024-12-26:
@Floyd-Schramm thanks! I have older version of the board and no soldering tools anyway, which is why I'm hoping to get away with some jumper wires and another board. Just need to figure out how...

sm-Fifteen, 2025-01-09:
> > So do we think some other config would work? Maybe "Horizontal" not "SWING_HORIZONTAL" or some such?
> 
> We could try "Horizontal", but my guess is that the Homeassistant internal link from SWING_HORIZONTAL to a display name and a icon is not yet made.

@Floyd-Schramm: Maybe try [`"vertical"`, `"horizontal"` and `"both"`](https://github.com/home-assistant/core/blob/ec37e1ff8d1daac5137edf15dabbbe46886de25a/homeassistant/components/climate/const.py#L71-L76)?

> I am not sure how to integrate the comfort mode as there is no default value in HA, but all other standard Daikin swing modes should be implemented as stated above.

"Comfort" [appears to be an operation mode in HAss](https://github.com/home-assistant/core/blob/ec37e1ff8d1daac5137edf15dabbbe46886de25a/homeassistant/components/climate/const.py#L46-L47), which actually somewhat matches how it works on the remote (mutually exclusive with powerful mode, disregards vertical swing config).