revk/xESP32-Faikout #589: [BUG] "The entity does not have a unique id" HA error

eshroom, 2024-12-25:
**Faikin hardware**
Amazon purchase 12/23

**Daikin hardware**
X50 connection to FXNQ series

**Describe the bug**
Have been getting intermittent Home Assistant error "The entity does not have a unique id" - "Unknown error" - symptom is while mode and temperature are selectable, fan speed is not and I get this error. But then it semingly resolves itself.

**To Reproduce**
Steps to reproduce the behavior:
It really appears and disappears.

**Expected behavior**
All functions, incluiding fan speed to work.

**Additional context**
All functions works fine within the Faikin control page, this error only presents when controlling via MQTT in Home Assistant

revk, 2024-12-26:
This may need someone that knows HA better than I.

I'll have a look though.

revk, 2024-12-26:
I am not seeing obvious duplicates - does the debug information say what ID is not unique?