revk/xESP32-Faikout #590: [BUG] IllegalInstruction reboots in loo[

Jerther, 2024-12-27:
**Faikin hardware**
I bought the module on Tindie

[image]

**Daikin hardware**
My Daikin unit is a FTX12NMVJUA
I have the faikin module connected according to this thread I made: https://github.com/revk/ESP32-Faikin/issues/208

**Describe the bug**
The faikin module stopped connecting to my wifi for no apparent reason. It just died on december 26th at 5:45 AM EST. We were sleeping. Power cycling does not help. The LED on the module is off.

[image]

**Additional context**
I managed to pull the module out of my Daikin unit, connect a FTDI to the  module [with these instructions](https://github.com/revk/ESP32-Faikin/blob/main/Manuals/Setup.md#connection) and see if there was any life left, and thankfully, there is. The ESP is stuck in a boot loop caused by some "IllegalInstruction". Here's the complete message:

```
08:52:18.001 -> ���ESP-ROM:esp32s3-20210327
08:52:18.001 -> Build:Mar 27 2021
08:52:18.001 -> rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
08:52:18.033 -> Saved PC:0x40376e2c
08:52:18.033 -> SPIWP:0xee
08:52:18.033 -> mode:DIO, clock div:2
08:52:18.033 -> load:0x3fce3818,len:0x3c8
08:52:18.033 -> load:0x403c9700,len:0x4
08:52:18.033 -> load:0x403c9704,len:0x9ac
08:52:18.033 -> load:0x403cc700,len:0x27b8
08:52:18.033 -> entry 0x403c9868
08:52:18.417 -> Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
08:52:18.514 -> Memory dump at 0x420074d8: 00007cc7 3040900d a0b20ca0
08:52:18.514 -> Core  0 register dump:
08:52:18.514 -> PC      : 0x420074dc  PS      : 0x00060d30  A0      : 0x803768d2  A1      : 0x3fceb2a0  
08:52:18.514 -> A2      : 0x00000000  A3      : 0x00000020  A4      : 0x00000000  A5      : 0xfc000000  
08:52:18.514 -> A6      : 0x0000003f  A7      : 0x0000000c  A8      : 0x82004b10  A9      : 0x3fceb270  
08:52:18.514 -> A10     : 0x3fceb2a0  A11     : 0xfffe0000  A12     : 0x00000000  A13     : 0x3fca2cbc  
08:52:18.546 -> A14     : 0x00000000  A15     : 0x00000005  SAR     : 0x00000006  EXCCAUSE: 0x00000000  
08:52:18.546 -> EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  
08:52:18.546 -> 
08:52:18.546 -> 
08:52:18.546 -> Backtrace: 0x420074d9:0x3fceb2a0 0x403768cf:0x3fceb2d0 0x403cd70e:0x3fceb340 0x403cd95a:0x3fceb380 0x403c98bd:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710
08:52:18.546 -> 
08:52:18.546 -> 
08:52:18.546 -> 
08:52:18.546 -> 
08:52:18.546 -> ELF file SHA256: 
08:52:18.546 -> 
08:52:18.546 -> Rebooting...
08:52:18.546 -> ���ESP-ROM:esp32s3-20210327
08:52:18.546 -> Build:Mar 27 2021
...
```

My guess is that the module attempted an auto-update during the night and it failed. The module was very stable prior to that.

revk, 2024-12-27:
There is some information on flashing code at the bottom of including a link to a Chrome based flashing too.

Jerther, 2024-12-27:
Here's what I did:

- Install esptool (pip install esptool)
- Find the correct binaries to download [here](https://github.com/revk/ESP32-Faikin/tree/main/ESP/release). My ESP module is a S3-MINI-1 so I guessed I needed the S3-MINI-N8 version.
- Flash the ESP with this command: `python -m esptool -p /dev/ttyUSB0 write_flash 0x1000 Faikin-S3-MINI-N8-bootloader.bin 0x8000 partition-table.bin 0xd000 ota_data_initial.bin 0x10000 Faikin-S3-MINI-N8.bin`
- I get an error saying "Guru Meditation Error detected". I figured I needed to put the ESP into programming mode.
- I tried to connect the "0" pin of the Faikin module to the DTR pin of my FTDI but that didn't help. So I tried the manual way:
- Disconnect power from the faikin module
- Bridge pin "0" of the faikin module to ground
- Connect power to the esp module. Meanwhile I had a serial monitor opened and it said "waiting for download"
- Remove the bridge
- Reissue the esptool command. Upload went well.

Now the ESP still reboots in a loop, but for a different reason. Here a loop:

```
09:30:55.625 -> ESP-ROM:esp32s3-20210327
09:30:55.657 -> Build:Mar 27 2021
09:30:55.657 -> rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
09:30:55.657 -> Saved PC:0x40043ac8
09:30:55.657 -> SPIWP:0xee
09:30:55.657 -> mode:DIO, clock div:2
09:30:55.657 -> load:0x3fce3818,len:0x3c8
09:30:55.657 -> load:0x403c9700,len:0x4
09:30:55.657 -> load:0x403c9704,len:0x9ac
09:30:55.657 -> load:0x403cc700,len:0x27b8
09:30:55.657 -> Checksum failure. Calculated 0x73 stored 0x3a
09:30:55.657 -> ets_main.c 329
```

revk, 2024-12-27:
I think you have an N4R2 (very small print on bottom of chip front).

But puzzled at that error.

Jerther, 2024-12-27:
There is conflicting information about the offsets to use in these pages:

https://github.com/revk/ESP32-Faikin/tree/main/ESP/release#esptool
https://github.com/revk/ESP32-Faikin/blob/main/Manuals/Setup.md#connection
https://github.com/revk/ESP32-RevK?tab=readme-ov-file#flashing-code

Two of them say the first offset to use is 0x0 so I guess it's the good one. I tried to flash again with

`python -m esptool -p /dev/ttyUSB0 write_flash 0x0 Faikin-S3-MINI-N8-bootloader.bin 0x8000 partition-table.bin 0xd000 ota_data_initial.bin 0x10000 Faikin-S3-MINI-N8.bin`

Now I get this:

```
09:39:30.749 -> ESP-ROM:esp32s3-20210327
09:39:30.781 -> Build:Mar 27 2021
09:39:30.781 -> rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
09:39:30.781 -> Saved PC:0x40376d50
09:39:30.781 -> SPIWP:0xee
09:39:30.781 -> mode:DIO, clock div:2
09:39:30.781 -> load:0x3fce2810,len:0x48c
09:39:30.781 -> load:0x403c8700,len:0x4
09:39:30.781 -> load:0x403c8704,len:0xa50
09:39:30.781 -> load:0x403cb700,len:0x26f8
09:39:30.781 -> entry 0x403c886c
09:39:31.202 -> [0;31mE (421) spi_flash: Detected size(4096k) smaller than the size in the binary image header(8192k). Probe failed.[0m
09:39:31.202 -> 
09:39:31.267 -> assert failed: __esp_system_init_fn_init_flash startup_funcs.c:95 (flash_ret == ESP_OK)
09:39:31.267 -> 
09:39:31.267 -> 
09:39:31.267 -> Backtrace: 0x40376e16:0x3fceb0f0 0x4038202d:0x3fceb110 0x40387055:0x3fceb130 0x420049a6:0x3fceb250 0x42005512:0x3fceb270 0x4200555d:0x3fceb290 0x420055c7:0x3fceb2b0 0x40376a0e:0x3fceb2d0 0x403cc69e:0x3fceb340 0x403cc916:0x3fceb380 0x403c88c1:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710
09:39:31.267 -> 
09:39:31.267 -> 
09:39:31.267 -> 
09:39:31.267 -> 
09:39:31.267 -> ELF file SHA256: 1095f61dd9fda880
09:39:31.267 -> 
09:39:31.267 -> Rebooting...
09:39:31.267 -> ESP-ROM:esp32s3-20210327
...
```

I will try with the N4R2 binaries.

Jerther, 2024-12-27:
Here's my little setup:

[image]

So for the N4R2 binaries I issued this command:

```
python -m esptool -p /dev/ttyUSB0 write_flash 0x0 Faikin-S3-MINI-N4-R2-bootloader.bin 0x8000 partition-table.bin 0xd000 ota_data_initial.bin 0x10000 Faikin-S3-MINI-N4-R2.bin
```

Aaaaaaand we're back! :smile: 

[image]

Any idea what might've happened, and how to prevent it from happening again?

MartynasZilinskas, 2025-01-24:
I have one faikin that also "died". I suspect that was failed OTA update as well. Thanks for @Jerther step by step guide how to reflash it.

My board: Faikin-S1-PICO 2023-06~

revk, 2025-01-25:
You managed to fix OK?

MartynasZilinskas, 2025-01-25:
Yeah, I meant I have successfully reflashed. If I will encounter this again I will try to record output.