revk/xESP32-Faikout #592: [BUG / HELP] Device went offline

shaohung, 2024-12-31:
**Faikin hardware**
Amazon.co.uk, purchased on Feb 2 2024

**Daikin hardware**
Probably unrelated

**Describe the bug**
I got two Faikins in my house, and they have been working great since Feb. Recently I went on a trip and one of the stopped working on 12-26-2024 around 11am UTC. I read the release note and notice the line of

> Please note, 2024-12-26 for a short period the beta did not work and would break your unit.

My guess is that that unit got updated to the bad beta version... Now I'm wondering, is there a way to get it back to working? or my best option is to get new one (although it no longer ship to US directly :( ). I can't seem to find any discussion on this beside the release note

Thanks

**To Reproduce**
N/A

**Expected behavior**
N/A

**Additional context**
N/A

revk, 2024-12-31:
That is unfortunate timing, sorry, it was a couple of hours at most.

Basically, it means soldering a USB lead and refreshing (which can be done from Chrome). The couple of other people that have had this have managed to do it.

More instructions at bottom of https://github.com/revk/ESP32-RevK

shaohung, 2024-12-31:
Gotcha, thanks for the quick reply and awesome project!

closed 2024-12-31 by shaohung

revk, 2024-12-31:
Let me know if you get stuck.

shaohung, 2025-01-08:
first time doing soldering - luckily everything worked out, and the board is back in service now :D thanks again!

revk, 2025-01-08:
Thanks for letting me know - I was obviously rather concerned that I bricked some boards.