revk/xESP32-Faikout #594: [BUG] demand reported 100 by Faikin, but Device seems to work with previous setting

JoergBo, 2024-12-31:
**Faikin hardware**
5x Faikin Device(s) are build in China/private, latest modell and latest firmware (Faikin-S3-MINI-N4-R2: 9f65e5b 2024-11-30T08:32:55)

**Daikin hardware**
Daikin Indoor unit is FTXM25R2V1B, connected to the S21 Port, works smooth (Multisplit 2 units and Multisplit 3 units here)

**Describe the bug**
When i total power off the Main Unit and repower it after some time, Faikin reports the demand setting to be 100.
The Daikin device in difference works further with the previaus seting. This is definitly shown by the power consumption of the Daikin main unit.
Faikin should report 50, when the unit is working with 50 after power up.

**To Reproduce**
Steps to reproduce the behavior:
1. set Faikin demand to 50
2. Power off the Outdoor Daikin unit
3. Wait 2 min. and power on the Main unit again.
4. Check the demand setting of the Faikin adapter, here allways 100

**Expected behavior**
Faikin should report the true demand setting

revk, 2025-01-01:
It reports what comes in on S21. You can enable debug (to mqtt) and see what is being sent by the unit itself.

JoergBo, 2025-01-01:
Hi, thanks for your hint to the debug option.
After shutoff with demand set to 50 and powering the outdoor unit again:
14:03:40.790

`14:03:40.790
Faikin_FaikinWG_F13598
info/FaikinWG/s21
{"G1":"04DA","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u0000\u00800","G5":"7?0\u0080","G6":"0\u00B008","G7":"00NA","GA":"CC60","GB":"0440","GG":"0000","GK":"q{51","GM":"4623","GN":"0000","GP":"E400","GQ":"7500","GS":"2000","GT":"3000","SG":"A"}
14:03:42.806
Faikin_FaikinWG_F13598
info/FaikinWG/s21
{"G1":"04DA","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u0000\u00800","G5":"7?0\u0080","G6":"0\u00B008","G7":"00NB","GA":"CC60","GB":"0440","GG":"0000","GK":"q{51","GM":"4623","GN":"0000","GP":"E400","GQ":"7500","GS":"2000","GT":"3000","SM":"521+"}
14:03:44.808
Faikin_FaikinWG_F13598
info/FaikinWG/s21
{"G1":"04DA","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u0000\u00800","G5":"7?0\u0080","G6":"0\u00B008","G7":"00NB","GA":"CC60","GB":"0440","GG":"0000","GK":"q{51","GM":"4623","GN":"0000","GP":"E400","GQ":"7500","GS":"2000","GT":"3000","SX":"052+"}
14:03:46.802
Faikin_FaikinWG_F13598
info/FaikinWG/s21
{"G1":"04DA","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u0000\u00800","G5":"7?0\u0080","G6":"0\u00B008","G7":"00NB","GA":"CC60","GB":"0440","GG":"0000","GK":"q{51","GM":"4623","GN":"0000","GP":"E400","GQ":"7500","GS":"2000","GT":"3000","SD":"000"}
14:03:48.807
Faikin_FaikinWG_F13598
info/FaikinWG/s21
{"G1":"04DA","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u0000\u00800","G5":"7?0\u0080","G6":"0\u00B008","G7":"00NB","GA":"CC60","GB":"0440","GG":"0000","GK":"q{51","GM":"4623","GN":"0000","GP":"E400","GQ":"7500","GS":"2000","GT":"3000","SH":"031+"}
14:03:50.808
Faikin_FaikinWG_F13598
info/FaikinWG/s21
{"G1":"04DA","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u0000\u00800","G5":"7?0\u0080","G6":"0\u00B008","G7":"00NB","GA":"CC60","GB":"0440","GG":"0000","GK":"q{51","GM":"4623","GN":"0000","GP":"E400","GQ":"7500","GS":"2000","GT":"3000","SI":"091+"}
14:03:52.809
Faikin_FaikinWG_F13598
info/FaikinWG/s21
{"G1":"04DA","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u0000\u00800","G5":"7?0\u0080","G6":"0\u00B008","G7":"00NB","GA":"CC60","GB":"0440","GG":"0000","GK":"q{51","GM":"4623","GN":"0000","GP":"E400","GQ":"7500","GS":"2000","GT":"3000","Sa":"580+"}
14:03:54.868
Faikin_FaikinWG_F13598
info/FaikinWG/s21
{"G1":"04DA","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u0000\u00800","G5":"7?0\u0080","G6":"0\u00B008","G7":"00NB","GA":"CC60","GB":"0440","GG":"0000","GK":"q{51","GM":"4623","GN":"0000","GP":"E400","GQ":"7500","GS":"2000","GT":"3000","SL":"000"}
14:03:56.805
Faikin_FaikinWG_F13598
info/FaikinWG/s21
{"G1":"04DA","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u0000\u00800","G5":"7?0\u0080","G6":"0\u00B008","G7":"00NB","GA":"CC60","GB":"0440","GG":"0000","GK":"q{51","GM":"4623","GN":"0000","GP":"E400","GQ":"7500","GS":"2000","GT":"3000","Sd":"000"}
14:03:58.811
Faikin_FaikinWG_F13598
info/FaikinWG/s21
{"G1":"04DA","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u0000\u00800","G5":"7?0\u0080","G6":"0\u00B008","G7":"00NA","GA":"CC60","GB":"0440","GG":"0000","GK":"q{51","GM":"4623","GN":"0000","GP":"E400","GQ":"7500","GS":"2000","GT":"3000","SN":"521+"}
14:04:00.822
Faikin_FaikinWG_F13598
info/FaikinWG/s21
{"G1":"04DA","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u0000\u00800","G5":"7?0\u0080","G6":"0\u00B008","G7":"00NA","GA":"CC60","GB":"0440","GG":"0000","GK":"q{51","GM":"4623","GN":"0000","GP":"E400","GQ":"7500","GS":"2000","GT":"3000","SG":"A"}
14:04:02.826
Faikin_FaikinWG_F13598
info/FaikinWG/s21
{"G1":"04DA","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u0000\u00800","G5":"7?0\u0080","G6":"0\u00B008","G7":"00NB","GA":"CC60","GB":"0440","GG":"0000","GK":"q{51","GM":"4623","GN":"0000","GP":"E400","GQ":"7500","GS":"2000","GT":"3000","SM":"521+"}
14:04:05.125
Faikin_FaikinWG_F13598
info/FaikinWG/s21
{"G1":"04DA","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u0000\u00800","G5":"7?0\u0080","G6":"0\u00B008","G7":"00NB","GA":"CC60","GB":"0440","GG":"0000","GK":"q{51","GM":"4623","GN":"0000","GP":"E400","GQ":"7500","GS":"2000","GT":"3000","SX":"052+"}
14`

and debug data after mode on:
`14:08:40.789
Faikin_FaikinWG_F13598
info/FaikinWG/s21
{"G1":"04DA","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u0000\u00800","G5":"7?0\u0080","G6":"0\u00B008","G7":"00N@","GA":"CC60","GB":"0440","GG":"0000","GK":"q{51","GM":"4623","GN":"0000","GP":"E400","GQ":"7500","GS":"2000","GT":"3000","SG":"A"}
14:08:42.802
Faikin_FaikinWG_F13598
info/FaikinWG/s21
{"G1":"04DA","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u0000\u00800","G5":"7?0\u0080","G6":"0\u00B008","G7":"00N@","GA":"CC60","GB":"0440","GG":"0000","GK":"q{51","GM":"4623","GN":"0000","GP":"E400","GQ":"7500","GS":"2000","GT":"3000","SM":"521+"}
14:08:45.526
Faikin_FaikinWG_F13598
info/FaikinWG/s21
[... 48 lines cut ...]
{"G1":"14DA","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u0000\u00800","G5":"7?0\u0080","G6":"0\u00B008","G7":"00N@","GA":"CC60","GB":"0441","GG":"0000","GK":"q{51","GM":"4623","GN":"0000","GP":"E400","GQ":"7500","GS":"2000","GT":"3000","SH":"031+"}
14:09:11.014
Faikin_FaikinWG_F13598
info/FaikinWG/s21
{"G1":"14DA","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u0000\u00800","G5":"7?0\u0080","G6":"0\u00B008","G7":"00N@","GA":"CC60","GB":"0441","GG":"0000","GK":"q{51","GM":"4623","GN":"0000","GP":"E400","GQ":"7500","GS":"2000","GT":"3000","SI":"081+"}
14:09:12.810
Faikin_FaikinWG_F13598
info/FaikinWG/s21
[... cut ...]
{"G1":"14DA","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u0000\u00800","G5":"7?0\u0080","G6":"0\u00B008","G7":"00NC","GA":"CC60","GB":"0441","GG":"0000","GK":"q{51","GM":"4623","GN":"0000","GP":"E400","GQ":"7500","GS":"2000","GT":"3000","Sd":"440"}
`

One addition to my previous Bug Report:
I was wrong, that the unit starts with 50 but reports 100! The unit starts with 100 and reports 100 ... 
My home automation set the demand to 50 after a few minutes, but I didn't notice. 
So may be, the only "problem" is, that the unit switches back to demand 100 after powering off.

Greetings!

revk, 2025-01-01:
`G7` reports demand, and first digit is `0` meaning demand is `100`

This is before and after. So it should have shown 100 the whole time, apart from maybe briefly 50 whilst it tried (and failed) to set to 50.