revk/xESP32-Faikout #595: [BUG] “System is offline”

peter-forrest, 2025-01-01:
[image]
[image]
[image]
**Faikin hardware**
Purchased direct from Faikin website. 

**Daikin hardware**
FTXM25YVMA

**Describe the bug**
Installed fine onto the connector shown in the guides here for the S403 connector. I put the metal cover back on and now it flashes red (there’s no error code mentioned here for flashing red and the others that contain flashing red can’t be it like heating as it’s not working) but I can no longer reach it from Home Assistant. I can reach it via the web browser so wifi must still be working and I notice that on the main page it says “system is offline” - see screenshot. When I press the top toggle button that presumably is the power button, it looks like it stays on for about 1ms then turns back off again. I tried a few times and same result. I tried changing the host name so that it would reboot itself but same result.

revk, 2025-01-01:
What button? Oh, and which web site did you buy from exactly?

Off line is usually a failed connection to the air-con, but it can mean you need the 5V reference on pin 1.

peter-forrest, 2025-01-01:
The "button" or toggle looking icon in the web interface from my screenshot. 

Sorry I thought I bought it from your website but I just double checked and I got them from Tindie.com

revk, 2025-01-01:
The buttons don't do anything if off line!

revk, 2025-01-01:
Looks like it shows timeouts so the comms are not working, issue with tx/rx or a need for the 5V pull up maybe. Not sure which. Check wiring carefully.

peter-forrest, 2025-01-01:
Haha! Yep that makes sense! 

Ok thanks. I'll double check the wiring in the morning. 

Here's a video of the flashing:

https://share.icloud.com/photos/0595-GDRkaUFbN6Z7OjJ_Evsg

I'll get a better video in the morning when I take the cover off to inspect the wiring.

peter-forrest, 2025-01-02:
Well that'll do it! It must have pulled on the cables a bit when I put the cover back on. 

[image]

I was hoping to have it hanging outside the cover so the metal cover wouldn't block the wifi signal but it's actually doing just fine inside the cover now. 

[image]

Thanks so much for your help Revk!!!

I have another 10 or so to do, this being my first. I'm wondering if a few tweaks to this amazing chip might be possible in upcoming versions?

The warning message in the web interface - would it be possible to change it to something like "no data detected on RX pin 34".  Would this be a possibility to help us in the future? (I wouldn't have needed to post here if I saw that)

And add an error code to the wiki for flashing red? 

And then after installation is complete it would be great to disable that LED, is it possible?

revk, 2025-01-03:
I'll tweak the message.

Disable LED in normal working by setting `dark` setting.

I'll have to make new 3D case, sorry, will do.

closed 2025-01-09 by peter-forrest