revk/xESP32-Faikout #596: Help: Energy Consumption monitoring in HASS

RacerBas, 2025-01-01:
I've been trying to get energy consumption trough mqtt in HASS but I can't quite wrap my head around it.
Has anyone else already achieved this?

I found this but I can't quite get a grasp on it and how the values were found: https://github.com/revk/ESP32-Faikin/issues/101

sm-Fifteen, 2025-01-06:
If your unit supports energy metering, Faikin should report it automatically as the lifetime energy usage of the device. You can just add it as-is to your energy dashboard. Not all units support that command, though (mine doesn't, even though I've only had it for 3 years), so there's not much you can do to fix it if it doesn't work out of the box.

The best alternate solution would be to monitor your house's power usage and try to guess the relation between that and the compressor speed, but that's easier said than done.

RacerBas, 2025-01-06:
I'm fairly certain it does since the cloud integration is reporting that metric in HASS. Do you still have the Daikin app and did/does it report energy usage there for you?

I just don't know how to get it out of MQTT and into HASS since it doesn't seem to discover that by itself. I'm trying to get rid of the cloud integration and this is the only thing missing.

The previous link mentions this:
```
mqtt:
sensor:
- name: "AC-Studio - Lifetime Energy"
state_topic: "A412FA97EC1C"
unit_of_measurement: "kWh"
value_template: "{{ (value_json.Wh | float) / 1000 }}"
device:
identifiers:
- A412FA97EC1C
name: AC-Studio
```

I'm fairly certain this is what I need, I just don't know what that state_topic is or where to my version for it is.
Now that I'm looking at it again though it seems an awful lot like a MAC address...

I'll try to see if I can find that this week. If I find it I'll try to write a tutorial that could be added to the documentation.