revk/xESP32-Faikout #598: [BUG] MQTT commands for swingh and swingv not working

eireit, 2025-01-04:
I try to control my Daikin FTXTA30C2V1BB/RXTA30C2V1B trou MQTT commands.

I'm using the following commands:

not working commands:
daikin/command/swingh 1 - Not working from MQTT, but works by using buttons in Faikin UI
daikin/command/swingv 1 - Not working from MQTT, but works by using buttons in Faikin UI

example for Working commands:
daikin/command/sensor 1 - working both from MQTT and Faikin UI
daikin/command/comfort 1 - working both from MQTT and Faikin UI
daikin/command/streamer 1 - working both from MQTT and Faikin UI
daikin/command/econo 1 - working both from MQTT and Faikin UI

The http command 
IP..../aircon/set_control_info?f_dir=1 is working

Is there any other commands that can activate the swing h/v?

revk, 2025-01-05:
I believe it is currently a command of `swing` and value of `H`, `V`, `H+V`

eireit, 2025-01-05:
Hi, thanks for info. This is working. Maybe the wiki shoud have been changed

from:
swingh - Boolean - horizontal louvre swing
swingv - Boolean - vertical louvre swing

to
swing - H, V, H+V (Horisontal,Vertikal, Horisontal+Vertikal)

revk, 2025-01-05:
I'll have to find that. I suspect it was changed due to home assistant integration.

revk, 2025-01-05:
OK the manual is right, it says `control` with a JSON payload including Boolean value `swingh` and `swingv`

The `swing` command is for HA, and *documented* as part of the HA config messages.

closed 2025-01-05 by revk

eireit, 2025-01-05:
As described in this first post i used the manual and the swingh and swingv did not work. ;-)

revk, 2025-01-05:
Yes you used as a command not the control command with them as json payload as per manual.

eireit, 2025-01-07:
Will this be correct?

Topic: daikin/command/control 
Payload: {"swingh": 1}

revk, 2025-01-07:
Assuming that is how you have the command topic, though for a Boolean it should be `{"swingh":true}`

eireit, 2025-01-08:
Thanks, now its working...