revk/xESP32-Faikout #601: [FEATURE] Additional functions

eireit, 2025-01-08:
For my Daikin FTXTA30C2V1BB/RXTA30C2V1B i have some additional function that should have been included. 

**Heat source** - External heat source, for example a fireplace
**Quiet** - Reduces the outdoor unit's noise level by changing the outdoor unit's frequency and fan speed.

Is is possible to include these functions? 

[image]

revk, 2025-01-08:
If there is an S21 message for it, then yes.

There is a lot of work on S21 messages, but not sure if quiet is one that we know. No idea on heat source.

sm-Fifteen, 2025-01-08:
I don't believe "quiet mode" (for the outdoor unit) is one that has been isolated. Interesting that the intelligent eye toggle is also there, we've been looking for that one for a while and it still hasn't turned up either. Having it in that screenshot at least indicates that it's *supposed* to be there.

As for external heat source, I think you might be the first one here to have that.

sm-Fifteen, 2025-01-09:
> Is there a easy way to test and logg? I can activate intelegent eye in the daikin app, but how to see what the s21 protocol is sending to the unit?
> 
> The Econo mode, powerful mode, Vertical swing, horizontal swing, confort mode, stream/fliter mode, quiet outdoor is working perfect. The sensore mode is the only mode not working. Intelegent eye and heat source is missing.

[So far](https://github.com/revk/ESP32-Faikin/issues/408), we've mostly just been changing the unit's configuration using the official remote and polling all known S21 commands (using the `command/myDaikinACUnitName/send` MQTT command) and looking for changes in the values, or tapping into the lines of the official controller to see what it's sending (which requires more advanced wiring and equipment; @Sonic-Amiga said he had to do at least some soldering). RevK has [a passthrough-style alternative board for signal reverse engineering](https://github.com/revk/ESP32-Faikin/blob/main/Manuals/Advanced.md#debug), but these aren't directly for sale and only available on demand, possibly only in the UK.

Most of what we've found so far is documented [here](https://github.com/revk/ESP32-Faikin/wiki/S21-Protocol).

Sonic-Amiga, 2025-01-09:
Hello! I indeed did some soldering, but that's trivial. I simply made a tap adapter so that i could tap a wire from either rx or tx line. The tap wire is routed to a USB -Serial (TTL level) adapter, so that i could set up RealTerm to capture raw data and analyze it later by hands.