revk/xESP32-Faikout #603: [Request for advice] - Demand control will not set to 100%

matteustace, 2025-01-11:
Hi,

I am not sure if this is a bug with the Faikin code or an issue with the air conditioning system itself - quite possibly the latter... but I am not sure.

Essentially I cannot select 100% demand control. If I select it on any of my 5 units through the Faikin, it just jumps back to the previous setting after a few seconds.

I don't know if the command the Faikin is sending is somehow incorrect, or if the indoor unit is just not accepting it for some reason.

Any advice would be much appreciated!

revk, 2025-01-11:
"jumping back" happens if the setting does not "stick", i.e. it sends several times but the value reported has not changed. So usually means the aircon is not playing.

Obviously if you can find an error in the S21 or a fix, do let me know.

closed 2025-01-11 by revk

dahadd, 2025-01-11:
Perhaps the demand control is not at 100% in another unit? Then the lowest value is assumed or the value jumps back again.

matteustace, 2025-01-11:
Thanks both - I had tried various ways of updating it on all the unit at once via scripts etc, whatever I did it would not switch to 100.

Power cycling it however has fixed it and it will now switch between modes without a problem.... so I suspect this is a Daikin bug rather than a Faikin bug!