revk/xESP32-Faikout #607: [BUG][HELP} Increased overall AC standby power consuption

Meadowtec, 2025-01-16:
Around the (auto) upgrade of my 5 Faikin boards (Amazon purched, S21 connected) to version 2024-11-30 the stanby power consuption of my AC system went up from typical 18W to typical 54W. I did not intentionally change any settings. All Faikin boards as well as the AC in generall are working perfectly.

Has anybody else observed a similar behaviour?
My only option to track down the culprit would be to dismantle all indoor units and temporarely disconnect the Faikin boards. Or has anybody a better / less elaborate idea?

Power is measured with a permanently installed Modbus connected power meter right after the circuit breaker which exclusively feeds the AC system.

MuzzaMcD, 2025-01-16:
I have not observed this on any of my Daikin systems. I have Shelly power meters on 4 of them, three are small 'back to back' systems (ie none are multi-split). The 3 small units each consistently report ~1.4w with a peak at ~5w every hour for a few minutes only while in standby. The Shelly PMs report every 20secs so effectively continuously reporting into HomeAssistant. My graphs do not show any changes. But there are many different Daikin systems so you might want to share your '5 Faikin boards' configuration. 
I have one larger Daikin system which is an outside unit connected to a single Ceiling Cassette internal unit. This is not running Faikin wireless (no S21 port) and it is using ~7w continuously in Standby.
Only suggestion I can make is to unplug Faikins and get measurements temporarily to compare with No Faikin?
Also how are the 18W / 54W power usage reported? Is it a peak power over time period? What is the reported time period of measurement?  . . .  Just some ideas, regards

Meadowtec, 2025-01-16:
@MuzzaMcD Thank you for your ideas :-) I didn't want to overload my initial post. Here are the details:
It's a multi-split system
Outdoor: 5MXM90A2V1B9
Indoor: 3x FTXM35A2V1B and 2x FTXM25A2V1B

Meetering: EASTRON SDM630 (only one phase in use)
Reporting interval 60secs to Iobroker / Influxdb / Grafana
The meter reports not just power (W) but energy (Wh) as well. The two values are in line over a 24h period. (1.3kWh in 24h = 54W average)

BUT !!! You triggered something with your ideas .....
I just created a diagram with outside temperatures and power overlayed and made a very interesting discovery: I always had those power spikes as you discribe them above. Mid November I started to have more of those spikes. Beginning of December, as mentioned in my initial post, the power consuption started to permanently be at the peak level of those (before) spikes.
If I correlate the spikes and the frequency of them and the later permanent high consumption with the outside temperature then there is a clear relation: Whenever the outside temperature drops below 12 C the consumption goes up! Currently we constatntly have below 12 C - in fact always around 0 C - outside. Therefore it's impossible to verify my assumption. But I think the 54W standby consumption could in fact be Daikin system related and simply a consequence of the outside temperature. 
We will start to see temperature above 12 C in only two to three months. Since disconnecting the Faikin boards is a hell of a work, I will therefore live with the current status until then and review the situation again once it gets warmer.
I will report back my findings.
Thank you once again for sharing your ideas! It's priceless to get a second view of a situation!

closed 2025-01-16 by Meadowtec