revk/xESP32-Faikout #608: [FEATURE] Additional HA/Mqtt Sensors

apos-t, 2025-01-17:
Can you please add the following data to HA as new sensor:

- Internal Temperature measurement - when you use a BLE Sensor, you just get the BLE Temp and not the internal data.
- The Set Temperature for the indoor unit

Alternatively, it would be enough to add them to the state mqtt message

[image]

Thank you

revk, 2025-01-17:
Pretty sure it is all in state message, the web uses the same set of fields.

apos-t, 2025-01-17:
I've dubble check the state message, but both sensors are missing.
And temp is equal to bletemp, which is fine for ha etc. 

Topic: state/klima_gym
{"ts":"2025-01-17T16:17:01Z","id":"XXXXXXXXX","up":true,"uptime":446113,"mqtt-up":101892,"app":"Faikin","version":"9f65e5b","build-suffix":"-S1-PICO","build":"2024-11-30T08:31:55","flash":8388608,"rst":3,"mem":40172,"spi":2069992,"ssid":"XXXXXX","bssid":"XXXXXX","rssi":-68,"chan":6,"ipv4":"XXXXXXXXX","online":true,"power":true,"target":17.00,"temp":17.37,"outside":2.00,"liquid":-18.00,"demand":60,"Wh":1454000,"fanfreq":0,"comp":79,"bletemp":17.37,"blehum":41.47,"blebat":70,"mode":"heat","fan":"auto","streamer":false,"quiet":false,"econo":false,"comfort":false,"powerful":false,"sensor":false,"swing":"off","preset":"home"}

revk, 2025-01-18:
I am sure there is an mqtt message with them, but I'll have to look.

revk, 2025-01-18:
OK look for `state/klima_gym/status`

revk, 2025-01-18:
But adding explicitly as an HA sensor is an option I guess.

revk, 2025-01-18:
OK try latest beta

apos-t, 2025-01-18:
Thank you for the quick implementation, there are two issues

- In HA there is only one new sensor AC-Home - which has no value yet, when I look into the mqtt debug log there is a value for ac-home - looks like the mapping is not working yet
- AC-Target is missing as a sensor in HA and has the wrong value, the value is the ble_temp not the target which should be 18 C for the ac.

You can see it in the attached screenshot

[image]

revk, 2025-01-18:
Interesting. I'll double check.

revk, 2025-01-18:
OK I see one bug, but I'll change names of fields as well in case HA has odd ideas. May need device reloading in HA. try next beta in a few minutes.

apos-t, 2025-01-18:
Thank you, ac-home is looking good, the sensor is showing in ha and has the right value
ac-target shows the right value in the mqtt message, but there is no sensor in ha for ac-target

revk, 2025-01-18:
Check the HA config message that is being sent, it may show why, but these are added as a generic tool so no way they should not work. Deleting and re-adding the device to HA may work?

apos-t, 2025-01-18:
there is no message for homeassistant/sensor/*MAC*actarget/config I've checked it directly on the mqtt server, I see the message for homeassistant/sensor/*MAC*achome/config this should be the reason why the sensor is not created. Took a look at your code and maybe found something:

first you're working with  CONTROL_target and later with CONTROL_temp as input

addtemp (daikin.status_known & **CONTROL_target**, "actarget", "AC-Target", "mdi:thermometer");

and then you work with 

   if (daikin.status_known & **CONTROL_temp**)
      jo_litf (j, "actarget", "%.2f", daikin.temp);     // The actual target

revk, 2025-01-18:
Ah, that is probably it

revk, 2025-01-18:
OK try that.

apos-t, 2025-01-18:
perfect :-), working now thank you!

closed 2025-01-18 by apos-t