revk/xESP32-Faikout #609: [FEATURE] Move External BLE Sensor Selection to Settings

benswinburne, 2025-01-17:
**Daikin hardware**
Faikin-S3-MINI-N4-R2: 9f65e5b 2024-11-30T08:32:55 S21

**Is your feature request related to a problem? Please describe.**

The web pages to control Faikin is exposed on my network, and also my guest network so users without access to home assistant can control the HVAC units. The BLE sensor selection is available on the homepage of the Faikin interface, not protected by the settings password. This allows any user on my network to choose the wrong temperature sensor on any HVAC page, i.e. the office HVAC could be pointed at the bedroom external sensor by a non admin user, resulting in the wrong temperatures being passed to the HVAC unit.

I struggled with this setting this up this initially - reading the docs was clear until the end where suddenly it just stops. I turned on BLE then was digging through settings trying to find where "The device will appear with its MAC address as its name." appeared. I assumed "BLE sensor ID" was supposed to change to a drop down. I wondered what ID meant in this context and ultimately didn't put the mac address in there.

I found the actual drop down by accident later on the homepage - then realised that this fills in the mac address in the ID field in settings.

**Describe the solution you'd like**

- Move the BLE sensor selection to the BLE section in the settings to be password protected.
- Rename the description of `autob` to `External BLE Sensor MAC Address (Without : delimiters)`
- Move `autob` next to `ble` in settings
- Update documentation in [BLE][1] 
  - indicate you must turn on BLE
  - inform the reader they must select BLE device using the new dropdown (or from homepage if selection is not moved) or fill in `autob` with the mac address of the BLE device

**Describe alternatives you've considered**

Leave the BLE device dropdown on the homepage but make it read only. Make `autob` field the only way to set the value.

**Additional context**

As an aside, the process of updating the LYWSD03MMC devices is considerably more convoluted since the recent default firmware update which [only accepts signed OTA updates MI Home][2]. It's still very doable but might want to tweak your summary slightly?

Also, the custom firmware only offers "BTHome2" in terms of BTHome. Might want to add that you support BTHome and BTHome2 in the docs (Everything seems to be working on BTHome2, but as the docs didn't say it I spent time testing)

[1]: https://github.com/revk/ESP32-Faikin/blob/main/Manuals/BLE.md
[2]: https://github.com/pvvx/ATC_MiThermometer/issues/602#issue-2786915630

revk, 2025-01-18:
Some good suggestions, I'll look in to it.

revk, 2025-01-18:
The "ID" is only a MAC if no name provided, e.g. these are the IDs I see on my desk.

[image]

revk, 2025-01-18:
I'll put the pull down in the first settings page to make that easier.

[image]

revk, 2025-01-18:
I'll remove the whole Faikin-auto from main page if a password is set.

revk, 2025-01-18:
OK try next beta, I think I have addressed your concerns.

benswinburne, 2025-01-18:
> The "ID" is only a MAC if no name provided, e.g. these are the IDs I see on my desk.
> 
> <img alt="Image" width="144" src="https://private-user-images.githubusercontent.com/996983/404512232-48fccdc6-9e6a-47bd-b09d-6673438ce201.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzcyMTc1OTAsIm5iZiI6MTczNzIxNzI5MCwicGF0aCI6Ii85OTY5ODMvNDA0NTEyMjMyLTQ4ZmNjZGM2LTllNmEtNDdiZC1iMDlkLTY2NzM0MzhjZTIwMS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTE4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDExOFQxNjIxMzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01ZWUxMzJlZjg5NjdhNWU1ZTcxYmJjNWYxN2VjZGFkYzFhMzUxOTU0Mjc3MTVhMWFiNjI1YTE1Yzg1ODQ2YzE1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Mp-MjXoCJRVJQJcU6Q0Qacla8z-cw2WjsyI_vpNfIv4">

Interesting, my dropdown only shows MAC addresses. The devices are all named (I see their names when scanning for BLE devices outside of Faikin). I am using the ATC_v49.bin firmware on the sensors.

[image]

Which ends up in `autob`

[image]

Updated to the beta firmware, section has moved to behind password protected section so 👍 

Thanks for the speedy turnaround, much appreciated.

revk, 2025-01-18:
Yeh we don't do a secondary poll for a name so has to be in the basic advertisement. Else MAC

closed 2025-01-31 by revk