revk/xESP32-Faikout #610: Faikin continues to heat up with Set Temperatur of 10 °

bjappler, 2025-01-22:
Hello,
I have noticed a strange behaviour with the Faikin module.
In heating mode and Faikin-auto mode deactivated, my air conditioner continues to heat.
In the Faikin interface and the Daikin app, the heating temperature is set correctly to 10 degrees, so it shouldn't actually be heating
If I switch from heating mode to fan mode and then back to heating mode, heating mode is stopped.
What could be the reason for this?
I only noticed this after I had activated comfort mode, but I have no idea whether this is directly related.
I have now deactivated comfort mode again and will check whether the air conditioning starts to heat up again when the temperature is set to 10 degrees.

I use a total of 8 Faikin modules and have not noticed this behaviour so far.
Since using the Faikin modules, I have not installed any firmware updates to the DAIKIN air conditioning systems. Can install firmware updates from Daikin without hesitation or should I still leave the firmeware as it is?

Best regards,
Björn

[image]
[image]

revk, 2025-01-22:
Getting some debug S21 messages may help diagnose.

revk, 2025-01-28:
If Faikin auto is off, that just sends whatever you select to the air-con.

Latest alpha reports more in the logs, including the ac-target which is what is set, that may have clues.

bjappler, 2025-01-28:
Where can I find these reports?
Attached you will find all info which I was able to copy out of mqqt Explorer:

state:
`{"ts":"2025-01-28T09:21:23Z","id":"xxxxxxxx","up":true,"uptime":621868,"mqtt-up":621863,"app":"Faikin","version":"9f65e5b","build-suffix":"-S3-MINI-N4-R2","build":"2024-11-30T08:32:55","flash":4194304,"rst":3,"mem":72016,"spi":2068592,"ssid":"xxxxxx","bssid":"444E6DBA177B","rssi":-52,"chan":6,"ipv4":"192.168.2.173","online":true,"power":true,"target":10.00,"temp":19.33,"outside":6.50,"liquid":40.50,"demand":40,"Wh":1136900,"fanfreq":12.0,"comp":18,"bletemp":19.33,"blehum":51.95,"blebat":56,"mode":"heat","fan":"auto","streamer":false,"quiet":false,"econo":false,"comfort":true,"powerful":false,"sensor":false,"swing":"C","preset":"home"}`

info:
`{"G1":"140A","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u0000\u00800","G5":"000\u0080","G6":"p004","G7":"l84k","GA":"2360","GB":"0441","GG":"0000","GK":"q{51","GM":"96C2","GN":"0000","GP":"D700","GQ":"A800","GS":"0000","GT":"2000","Sa":"560+"}`

revk, 2025-01-28:
Depends what you are watching on MQTT, but latest beta should have an `"actarget"` in that status. Are you running the beta?

The target sent to aircon is `D1` command and `G1` status, you `G1` is `140A`

- `1` is power on
- `4` is, I think, *heat*
- `0` should be temp, but not sure `0` is valid, which is odd.
- `A` is fan I think (*auto*)

What target temp have you set it to?

If you enable debug we can see what `D1` is sent perhaps.

bjappler, 2025-01-28:
This is what I have set with Faikin at the moment.

[image]

I am controlling my Faikin with iObroker scripts, but they are disabled at the moment. I can install the new beta if this helps.

revk, 2025-01-28:
Well I would suggest beta for any debugging, but that temp looks odd in the `G1`. Seeing the `actarget` would help I expect.

bjappler, 2025-01-28:
Here are the states after installing the beta update. Currently the behaviour is normal, no heating.

state:
`{"ts":"2025-01-28T10:11:35Z","id":"xxx","up":true,"uptime":161,"mqtt-up":155,"mem":119352,"spi":2069388,"online":true,"power":true,"target":10.00,"actarget":10.00,"temp":18.74,"achome":23.50,"outside":7.50,"liquid":24.50,"demand":40,"Wh":1137000,"fanfreq":0.0,"comp":0,"bletemp":18.74,"blehum":52.25,"blebat":56,"mode":"heat","fan":"auto","streamer":false,"quiet":false,"econo":false,"comfort":true,"powerful":false,"sensor":false,"swing":"C","preset":"home"}` 

info:
`{"ts":"2025-01-28T10:11:35Z","id":"xxxx","up":true,"uptime":161,"mqtt-up":155,"mem":119352,"spi":2069388,"online":true,"power":true,"target":10.00,"actarget":10.00,"temp":18.74,"achome":23.50,"outside":7.50,"liquid":24.50,"demand":40,"Wh":1137000,"fanfreq":0.0,"comp":0,"bletemp":18.74,"blehum":52.25,"blebat":56,"mode":"heat","fan":"auto","streamer":false,"quiet":false,"econo":false,"comfort":true,"powerful":false,"sensor":false,"swing":"C","preset":"home"}`

`{"ts":"2025-01-28T10:11:07Z","url":"http://ota.revk.uk/Faikin-S3-MINI-N4-R2.bin?xxx","version":"9f65e5b","project":"Faikin","time":"08:32:55","date":"Nov 30 2024","was-version":"deb4b95"}`

`{"protocol":"S21","cmd":"D1","payload":"31343041","text":"140A"}`

revk, 2025-01-28:
OK The `D1` is right, the `0` does work out as 10℃, so the air-con is set to 10℃, and heat.

So is it really pumping out warm air?

If the temp it sees is 23.5℃ and you have set heat to 10℃ it should not be heating.

bjappler, 2025-01-28:
currently it is working fine and it is not heating. But this can be due to the rebooting after installing the beta.
I will keep in comfort mode and if I notice this bug again I will send you the states again..

bjappler, 2025-01-30:
Right now I noticed the same issue again and yes it is really pumping out warm air and consuming 300 w.
Oddly, my Faikin module had automatically installed the standard firmware again. I have now reinstalled the BETA and unchecked the auto-updates option.
Below are the errors, I hope I can now keep the beta longer without the firmware resetting to Nov again. I thought I had already deactivated the Updatefeature after installing Beta the last time..

`{"G1":"1407","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u0000\u00800","G5":"000\u0080","G6":"p004","G7":"v84S","GA":"9360","GB":"0441","GG":"0000","GK":"q{51","GM":"DAC2","GN":"0000","GP":"D700","GQ":"A800","GS":"0000","GT":"2000","Sa":"050+"}`

`{"ts":"2025-01-30T07:58:40Z","url":"http://ota.revk.uk/Faikin-S3-MINI-N4-R2.bin?xxx","version":"9f65e5b","project":"Faikin","time":"08:32:55","date":"Nov 30 2024","was-version":"deb4b95"}`

`{"protocol":"S21","cmd":"DH","payload":"31303030","text":"1000"}`

`{"ts":"2025-01-30T08:07:01Z","id":"xxx","up":true,"uptime":528,"mqtt-up":522,"app":"Faikin","version":"deb4b95","build-suffix":"-S3-MINI-N4-R2","build":"2025-01-26T12:08:19","flash":4194304,"rst":3,"mem":119528,"spi":2069372,"ssid":"xxx","bssid":"xxx","rssi":-50,"chan":6,"ipv4":"192.168.2.173","online":true,"power":true,"target":10.00,"actarget":10.00,"temp":19.51,"achome":23.00,"outside":5.50,"liquid":27.50,"demand":30,"Wh":1143700,"fanfreq":19.0,"comp":14,"bletemp":19.51,"blehum":55.61,"blebat":57,"mode":"heat","fan":"high","streamer":false,"quiet":false,"econo":false,"comfort":true,"powerful":false,"sensor":false,"swing":"C","preset":"home"}`

`{"protocol":"S21","env":19.5,"online":true,"home":23.0,"heat":true,"fanrpm":1140,"comp":14,"outside":5.5,"liquid":27.5,"anglev":276,"Wh":1143700,"power":true,"mode":"H","temp":10.0,"demand":30,"fan":"5","swingh":false,"swingv":false,"econo":false,"powerful":false,"comfort":true,"streamer":false,"sensor":false,"led":true,"quiet":false,"ble":{"temp":19.49,"hum":55.56,"bat":57},"autob":"xxx","autor":0.0,"autot":17.0,"auto0":"00:00","auto1":"00:00","autop":false}`

revk, 2025-01-30:
If beta is set that should not clear, so that is odd.

That shows `actarget` 10, and the message logs showed that is what is sent.

So you seem to have a Daikin that heats when target is set to 10C

The only other thing I can think is whether your air-con can actually be set to 10C? I don't think mine can. If using your normal remote does it allow you to set as low as 10C?

If not, maybe try setting to the lowest it actually works with?

bjappler, 2025-01-30:
Why I am confused is, as Faikin set temperature is displayed correctly so it should not head. You said, that 

> If Faikin auto is off, that just sends whatever you select to the air-con.

But shouldn't all the settings that were set on the air con be displayed correctly in Faikin?

[image]

bjappler, 2025-01-30:
As described, if I deactivate the comfort mode the air con does not heat at 10. So it seems to be a problem with comfort mode

`{"protocol":"S21","env":19.7,"online":true,"home":23.5,"heat":true,"fanrpm":0,"comp":19,"outside":5.0,"liquid":31.5,"anglev":126,"Wh":1143700,"power":true,"mode":"H","temp":10.0,"demand":30,"fan":"5","swingh":false,"swingv":false,"econo":false,"powerful":false,"comfort":false,"streamer":false,"sensor":false,"led":true,"quiet":false,"ble":{"temp":19.67,"hum":55.47,"bat":56},"autob":"

[image]

","autor":0.0,"autot":17.0,"auto0":"00:00","auto1":"00:00","autop":false}`

revk, 2025-01-30:
We are second guessing what your daikin is doing. The S21 shows we have sent 10C, and so it should not be heating.

If, with no Faikin at all, you manually set to 10C and heat and comfort mode, what does the air-con do?

bjappler, 2025-01-30:
Currently, after reactivating comfort mode and setting it to 10°, it's not heating. I don't know when it will start heating at 10° again. 

Maybe it is a bug with the Daikin AC. I have a Daikin EMURA 3.

Should I do a firmware update with the Daikin AC? I avoided it as I was afraid that Faikin could stop working with Daikin after installing a firmware update.

revk, 2025-01-30:
Probably sensible.