revk/xESP32-Faikout #613: [BUG] Can't leave beta channel

benswinburne, 2025-01-22:
**Faikin hardware**
Faikin-S3-MINI-N4-R2: 6c45736 2025-01-18T13:39:11

**Describe the bug**
I recently switched to the beta channel for updates on my Faikin to test an update from #609 - obviously to do that I had to toggle the "Download from beta" option (I forget the exact wording, it's gone now).

Having now downloaded and tested the beta, I don't want further beta updates so I hoped to disable this option and switch back to the stable channel but there's no option to do so.

Honestly I'm not sure this is a bug per se, but it's something I wanted to do and couldn't so wasn't sure if it was intentional.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable beta updates
2. Download an OTA update
3. Try to disable further beta updates
4. Not possible

**Expected behavior**

Ability to switch back to the stable version.

I understand that this may not necessarily give me the option to download an "older" version of non beta firmware, but it should perhaps download the stable version of the beta i'm on once the beta is promoted to the stable channel. This would allow me to test some beta firmware, say 1.0-beta, then when 1.0 stable comes out, i'm on stable and can continue to only auto update stable updates thereafter.

If switching to beta is permanent - this should be noted in the ui perhaps and point at the [reflashing documentation][1].

[1]: https://github.com/revk/ESP32-Faikin/blob/main/Manuals/Setup.md#re-flash-firmware

revk, 2025-01-23:
Oooh, interesting!

That may be a bug, in which case I'll make a new beta that will fix, later today.

revk, 2025-01-23:
OK silly error on my part, I moved `betarelease` to `release/beta` to reduce clutter, and so it thought there were no beta releases. Fix in latest beta now.

closed 2025-01-23 by revk

benswinburne, 2025-01-23:
Confirmed the Beta Release toggle is back in Faikin-S3-MINI-N4-R2: c632105 2025-01-23T07:43:52

Thanks very much