revk/xESP32-Faikout #614: Home Assistant 'set_temperature' not working?

JBecker32, 2025-01-22:
I try to set the required temperature in HA with node_red via climate.set_temperature, but this is not accepted!?!
Then in the HA GUI I only see  'eco' mode as a possible mode, not 'comfort' or 'away' or anything else. How am I supposed to set the AC to different temperatures at all?
I can set the temperature manually via the 'thermostat', so this is possible. But how can I achieve this via node_red?

BR, Jörg.

PS: extremely pleased with the product, much better than the Daikin units!!!

revk, 2025-01-23:
HA can normally control temperature setting with no problem. But I am not an expert on HA / node_red. Someone else may be able to help.

I suggest loading the latest beta release anyway as there have been minor improvements.

As for other modes, if you have them available via S21 (which the case for most) then they too should be settable.

JBecker32, 2025-01-23:
> HA can normally control temperature setting with no problem. But I am not an expert on HA / node_red. Someone else may be able to help.
>
I am so stupid! I wrote the command in a wrong way, so it was impossible to work correctly. Sorry for the noise I made. Setting temperature is working perfectly now!

> I suggest loading the latest beta release anyway as there have been minor improvements.
> 
> As for other modes, if you have them available via S21 (which the case for most) then they too should be settable.

I have not yet understood how I can find out which modes are available on my AC. But as setting temperature is working now, this can wait until later.

Thank you very much for your fast response and generally for the fantastic work!
Jörg.