revk/xESP32-Faikout #622: Daikin heats continouesly over target up to reset

Ulf349, 2025-01-28:
**Faikin hardware**
Faikin-S3-MINI-N4-R2, self created module

**Daikin hardware**
FVXM25

**Describe the bug**
Iobroker controlls via MQTT the faikin over S21
The rooms heats over the target Temperatur.

Target 20 degrees Celsius.
Rooms over BLE 23 °C

Only over reset could be the problem solved. 

This behavior was on more Faikins.
I have 9 working Faikins.

**To Reproduce**
Couldn't be reproduced. From one time to another the room overheats.

**Expected behavior**
Over mqtt the demand is changed, maybe this could be the problem?

**Additional context**
you can see this bug was on more devices see the pictures.
In the one device today I updated the firmware to a beta. I don't know if that solve the issue.

[image]
[image]
[image]
[image]

[image]

revk, 2025-01-29:
In the example of room being 23C what is the target temp sent to the Daikin at that point?

That should be in the MQTT status now on latest beta as actarget.

Ulf349, 2025-01-31:
Hi, Problem is again in the same room. The room heats up over target.

Here the mqtt values:

`{"protocol":"S21","ts":"2025-01-31T20:34:00Z","env":22.97,"target":18.00,"online":true,"control":true,"home":23.00,"heat":true,"fanrpm":350,"comp":14,"outside":1.50,"liquid":28.00,"anglev":23,"Wh":1270700,"power":true,"mode":"H","temp":20.50,"demand":50,"fan":"Q","swingv":false,"econo":false,"powerful":false,"comfort":false,"streamer":true,"sensor":true,"quiet":false}`

`{"ts":"2025-01-31T20:18:00Z","id":"68BxyzE20","up":true,"uptime":246881,"mqtt-up":72432,"mem":63836,"spi":2069040}`

{"protocol":"S21","env":23.0,"online":true,"control":true,"home":23.5,"heat":true,"fanrpm":350,"comp":14,"outside":1.5,"liquid":28.0,"anglev":23,"Wh":1270800,"power":true,"mode":"H","temp":20.5,"demand":50,"fan":"Q","swingv":false,"econo":false,"powerful":false,"comfort":false,"streamer":true,"sensor":true,"quiet":false,"ble":{"temp":22.99,"hum":51.47,"bat":2299},"autob":"A4C1385235A5","autor":0.5,"autot":18.5,"auto0":"00:00","auto1":"00:00","autop":false}

Can you see there a problem?

revk, 2025-02-01:
I am trying to follow.

Looking at that you have Faikin auto on, you have it set to 18.5C as target, yes?

Your aircon appears to think the room is 24.0C, your BLE sensor sees 22.89C.

The setting in the aircon appears to be heat to 20.5C

Setting heat to way below the current temp is turning heating off. i.e. setting heat to 20.5C when room is 24C.

Faikin auto will set the target temp to a few degrees below the temp the aircon sees, or a few degrees above what it sees to turn off or on the heating. It does the same the other way when cooling.

So in heat mode but not heating, waiting for it to cool down.

If it does not cool down it will eventually switch to a cooling mode.

There are settings to control how quickly it will switch from heating to cooling and other parameters if you need.

Ulf349, 2025-02-01:
Hey,

so I think I know how the Faikins working. I have 9 Daikins with 9 Faikins since 3 month and with this Faikin I have a crazy controlling. Yes, Faikin auto is active.
Yesterday I don´t tried to explain what´s the behavior: See last picture in last comment: The room temp BLE 22.87, the target was set to 18.5. And you can see the fan is on and the Daikin heats up the room. Nobody changed anything in the Faikin or over Onecta.
This 2 values BLE and target works on all other Faikins very well, but currently at this device not.

This is the permenent logging the last days in the problem room:
[image]

You can see below an other room. You can see the Faikin regulates perfect:

[image]

Your explained switching from heat to cool is disabled. I´m the master and decide you must heat or cool.

I ´m sure, if I restart the Faikin with the MQTT command, the Faikin controls the Daikin again perfectly, same as last time. But that could not be the solution to restart the module every day.

If you have any idea, what can I do, please help me. If loggings are necessary, nearly all values are permanend logged.

Thanks for your support.

revk, 2025-02-01:
OK, but settin disking to heat to 20.5 when it thinks the room is 24 should NOT be heating, clearly. If the daikin is still heating when it is 3.5 degrees too hot is silly.

You can change the margins used for fakin auto, to set even lower target to stop heating.

MUN0X, 2025-02-04:
> OK, but settin disking to heat to 20.5 when it thinks the room is 24 should NOT be heating, clearly. If the daikin is still heating when it is 3.5 degrees too hot is silly.
> 
> You can change the margins used for fakin auto, to set even lower target to stop heating.

It’s silly but some units do just that. My perfera heats until 4C above the setpoint by default. That why I requested to make “heatover” signed last year. A “heatback” value of at least 6 is required to avoid the unit from turning on when stepped back.

macmpi, 2025-02-25:
Had the same issue last nigh (and maybe also a few days before, I just not looked as closely at it): 
**Faikin auto** went south and was overheating over 18 (was positively blowing hot), while max temp (BLE) was set at 15.5 (thermostat & lockmode on), .

Interestingly HA (no automation set) showed proper info onscreen (that's how I remotely discovered the issue), but device [homepage was locked-up](https://github.com/revk/ESP32-Faikin/issues/299) (quite hard lockup, even settings were not accessible). Had to perform reboot on module pads. Back to normal after reboot (hence temp went down as expected).

On Faikin auto mode: how does Faikin takes over aircon built-in temp regulation?
Same when HA takes over (no my case here, HA is just monitoring for now): does Faikin auto also gets deactivated if MQTT order comes-in?

BTW, what is the "Sensor mode" for on web UI: I do not see that on mine (FTXS20G2V1B), neither (only have powerfull, & vert/horiz swing): how does Faikin decide which to show/enable/use?

PS: would be nice to provide a HA template/blueprint for Reboot button over MQTT

dahadd, 2025-02-25:
Reboot MQTT
command/xxxFaikinxxx/restart
xxx...your faikin-name

macmpi, 2025-03-09:
Had this again when Falkin-auto kicked-in (enabled on time schedule): web interface was also [down](https://github.com/revk/ESP32-Faikin/issues/299) - can't tell if was down earlier.
Same coincidence, symptoms with common root (4 days of use since previous reboot - not parameters changes)?
Restarted with [custom MQTT](https://github.com/revk/ESP32-Faikin/issues/657) button: seems fine since for now.

macmpi, 2025-03-09:
Again 3degrees north on faikin auto (web ok):
Faikin:
{"protocol":"S21","ts":"2025-03-09T23:05:00Z","env":[18.05,18.06,18.07],"target":[14.50,15.50],"online":true,"control":true,"home":18.50,"heat":true,"fanrpm":0,"comp":0,"outside":11.00,"liquid":[24.00,24.72,26.00],"anglev":66,"power":true,"mode":"H","temp":15.50,"fan":"A","swingh":false,"swingv":false,"powerful":false}

State status:
{"protocol":"S21","env":15.7,"online":true,"control":true,"home":14.5,"heat":true,"fanrpm":0,"comp":41,"outside":13.0,"liquid":28.5,"anglev":111,"power":true,"mode":"H","temp":15.5,"fan":"A","swingh":false,"swingv":false,"powerful":false,"ble":{"temp":15.73,"hum":62.17,"bat":95},"autob":"A4C138E68376","autor":0.5,"autot":15.0,"auto0":"05:55","auto1":"22:05","autop":false}

Jerther, 2025-03-22:
I'm experiencing the same problem. The actual temperature is always over the set point by 3 or 4 degrees. I'm not using MQTT. I'm using legacy REST API calls.

[image]

Is this a thing with the faikin module, or the daikin unit unable to regulate itself properly?

I have to admit I've had the faikin module very early after I got the daikin unit installed. So I don't remember the original behavior.

revk, 2025-04-15:
OK the image shows target 25C, current home temp 27C.

So that is what the air-con is told - heat to 25C, when it is 27C

So the aircon heating at that point is what the aircon has decided to do under those circumstances.

What happens if you do not have the Faikin, use the remote to set target to 25C and heating, does it keep heating even when 27C? It will, it is what the aircon has decided to do.

Utopia69, 2025-04-15:
> What happens if you do not have the Faikin, use the remote to set target to 25C and heating, does it keep heating even when 27C? It will, it is what the aircon has decided to do.

Will try later today

Utopia69, 2025-04-16:
I think this indeed the problem.
Will try with faikin-auto

Jerther, 2025-04-17:
As links tend to go dead, here are some relevant screenshots for a possible target temperature fix on the Daikin unit itself:

[image]

I have a different model, and this appears to be relevant to mine:

[image]

[image]
(setting 7)

I'll give it a try.