revk/xESP32-Faikout #640: [FEATURE] Web UI weekend schedule and temp - location

github0123456789, 2025-02-10:
**Daikin hardware**
Perfera 

**Is your feature request related to a problem?
I use the web UI a lot. Why? Simple and fast. 
But I’m missing a time setting/schedule for the weekends. Now my AC starts in the weekend at the same time as during weekdays. This is silly as the house is still asleep for 1 or 2 extra hours. For me the time setting is now a useless setting. My AC’s are now still scheduled with the Daikin online/app planner. And I want to get rid of this. 

**Describe the solution you'd like**
Add a weekend schedule in the Web UI to make it usable. A planner would even be better at the long run but also very complex. 

Thank you.

revk, 2025-02-10:
This is probably best done via Home Assistant to be honest. I am not that keen on making the main web interface too complex. As you say, it starts as just weekends, then some more comprehensive planner, then some calendar, then scripting language and so on, and I think HA is way better placed for that.

github0123456789, 2025-02-10:
Thank you for your reply. You are mentioning to use Home Assistant (HA) for scheduling the AC start. Great idea and I like the HA functions of Faikin. I implemented this but I have had bad experience. 

It happened me multiple times that HA was in an error. E.g. caused by a network failure, or other issue. HA did not trigger my AC to start heating. Resulting in a temperature of 12°C of the living room. A situation I would like to avoid.

I need to trust on my primary function of my AC (eg. Days scheduled On/Off). The Faikin module is part of the primary function. Faikin replaces the Daikin cloud service and should in my humble opinion, be able to run on its own / independent. So without the need of HA or other third party controllers. It is able to do that, but one schedule function seems to be missing to make it complete.

I really would like to use the weekdays schedule but also need a specific schedule for the weekend (Saturday/Sunday: time on - time off). 

I hope you can reconsider this following the explanation above. 

Thank you.

revk, 2025-02-10:
I'll consider.

bjappler, 2025-02-17:
I was also thinking about this feature a long time. Maybe you could start with an easy temperature schedule, where we could just set a target temperature, time and activation.
Of course we can use iobroker or HA for that. But this feature would also be very helpful, if there is not a possibility of using iobroker or HA.

Attached you will find a **proposal** of using an easy timed schedule. To avoid complexity you could add this below the timed on off section

[image]

Maybe you could consider this feature request. This would make Faikon much more comfortable out of the box.

revk, 2025-02-17:
This very quickly gets out of hand, on, off, temp, days of week, specific dates. Really HA is way better for it.

Not ruling it out, but it is a minefield.

github0123456789, 2025-02-17:
Thank you for your response. I believe that mentioning Home Assistant as the basis for scheduling the AC may come across as overly complex and not entirely reliable.

Considering your proposal, I think it includes all of the minimum requirements except for the specific date settings. Users can easily turn off the air-conditioning manually. I should skip that option. 😀 For your consideration only off course.