revk/xESP32-Faikout #641: [FEATURE] Web UI temperature minus too close to other buttons

github0123456789, 2025-02-10:
**Daikin hardware**
Perfera 

**Is your feature request related to a problem?
I use the web UI a lot. Why? Simple and fast. 

But I’m what I noticed is that the minus button for changing the temperature is too close to another button above it. At 2 locations of the web UI. When I lower my AC temperature with some touches on the web UI I accidentally switch off the setting for the allowed temperature differences below the text: “_Faikin-auto mode (sets hot/cold and temp high/low to aim for the following target), and timed and auto power on/off”_

This results in a situation that my AC tries to exactly stay at the set temperature and continues to run full day. 

[

[image]

](url)

**Describe the solution you'd like**
Please move this setting to the real settings screens so it cannot be changed accidentally nor by other users of the Web UI. 

Also please increase the size and location of the temperature settings when not in automode. 
See screenshot. Same issue as above. 

[image]

Thank you for your great product.

revk, 2025-02-10:
Odd, this is what it is meant to look like

[image]

If you have suggestions for css changes, let me know.

revk, 2025-02-12:
I have tweaked the slider size so that may be better now, try latest beta.

closed 2025-02-12 by revk

github0123456789, 2025-02-12:
Thank you. Just installed the beta and now the web UI fits my safari. Happy with your great work. 

Regarding the page colours on the screenshot: This is caused by running Safari with a plug-in that changes bright colours into less bright colours(dark mode).