revk/xESP32-Faikout #642: [BUG]

macmpi, 2025-02-10:
**Faikin hardware**
Faikin-S3-MINI-N4-R2: fcee8aa

**Daikin hardware**
FTXS20G2V1B or FVXS50FV1B / S21

**Describe the bug**
After OTA updates, some updated features may not work as expected. This is eventually solved after more reboots

**To Reproduce**
Request update, try use after reboot.

**Expected behavior**
Feature not activated as it was supposed to (ex [screenshots](https://github.com/revk/ESP32-Faikin/discussions/633#discussioncomment-12080685) after Hide Faikin auto)

**Additional context**
Maybe browser caches are not invalidated after OTA update. There are methods (ccs-versioning,& al) [to do so](https://stackoverflow.com/questions/9131037/how-to-invalidate-web-browser-cache-content), some of which may be included in page boilerplate.

revk, 2025-02-10:
This is not normally the case, and I suspect an issue with communications with the S21/aircon in your case. So will be very difficult to diagnose.

I suggest changing settings. Disable x50a and cnwired and swaptx and swaprx. That may help.

macmpi, 2025-02-10:
Will try next time, thanks.

Anyhow, making sure to invalidate browser cache after OTA (or even each reboot?) is probably quite necessary/classical measure: this will help for any UI change (icons, etc), and maybe other `javascript` stuff.
Always better to make sure to start clean & fresh.

On Falkin device side, is `reboot` really clearing all RAM (and eventual caches) for good?

revk, 2025-02-10:
Reboot restarts the code from scratch. There are no persistent RAM values over reboot.

But all active settings should then come back from the aircon.

closed 2025-02-12 by revk