revk/xESP32-Faikout #644: [FEATURE] Web UI: report BLE battery level information

macmpi, 2025-02-10:
**Daikin hardware**
 any, but vested interest for FTXS20G2V1B or FVXS50FV1B / S21

**Is your feature request related to a problem? Please describe.**
would be useful to report/display BLE battery information and level on WebUI homepage

**Describe the solution you'd like**
It could find its spot on the right side of Temp & Humidity info (additional icon? percent or value?)

**Describe alternatives you've considered**
none

**Additional context**
seems reported to HA https://github.com/revk/ESP32-Faikin/issues/155

closed 2025-02-11 by revk

macmpi, 2025-02-11:
Great, thanks!
Maybe items spacing could be tweaked a bit for improved readability?

[image]

macmpi, 2025-02-11:
😄 
I think BLE info display is a good suspect for[ homepage lock-up issue](https://github.com/revk/ESP32-Faikin/issues/299#issuecomment-2643419618), when you have time to look at it.

revk, 2025-02-11:
OK BLE on separate row, and yes, interesting theory on lockup.