revk/xESP32-Faikout #645: [FEATURE] Confirm "breaking" settings befor saving/applying them (e.g. when changing wifi settings, etc)

boecks, 2025-02-10:
**Daikin hardware**
Several Daikin Comfora on S21 Connector

**Is your feature request related to a problem? Please describe.**
Due to a WiFi change, I had to update the SSID and passphrase on all my Faikins. Unfortunately, in a moment of bad luck (and a mistimed Enter key press), I accidentally saved only the SSID change without updating the passphrase. As a result, my Faikin is now trying to connect to a non-existent WiFi network, leaving me without access.

**Describe the solution you'd like**
A confirmation popup before saving changes, or at least preventing the Enter key from automatically submitting the form, could help avoid situations like this. This would add a safeguard without significantly impacting usability.

**Describe alternatives you've considered**
Pay more attention when typing - though this might be error prone

**Additional context**
My Workaround will be to spawn the "wrong" wifi with an extra router and change the settings to the correct wifi.

revk, 2025-02-11:
If it cannot connect it should appear as an access point, just like when it first arrived, allowing you to change the SSID/password again.

If not, shorting the two pads on the back (fi you have them), three times, will factory reset.

[image]

closed 2025-02-11 by revk

boecks, 2025-02-11:
I was not aware, it would go in AP mode and did not scan for the Faikin wifi, though this could have obviously have helped solving my problem like I actually solved it. I created the wifi as described in my workaround and could fix the settings that way.

Having the hardware pads is good as a last resort, yet dissembling the AC unit to access them is always kind of a hassle. (at least do me) I would rather have to do an extra mouseclick for confirmation instead, before potentially breaking things... 😅

Thank you for your quick reply though.