revk/xESP32-Faikout #653: [BUG] FTX24AXVJU Communication Problems?

computerdude95, 2025-02-21:
**Faikin hardware**
Purcahsed 2x Faikin boards from Tindie - 2024-11-04 PCB revision.  Pictures of the Faikin board below, along with S21 port.

**Daikin hardware**
The Daikin unit is a FTX24AXVJU, connected via S21 port.  I have two of these units, connected identically.

**Describe the bug**

1. Initially on startup, the unit can sometimes be controlled, although sometimes it takes a settings toggle in the Faikin to start actually communicating.  The Faikin board is always available on wifi though. After a while – usually after around half an hour or so, I can no longer control the unit with the Faikin interface or Home Assistant.  This includes changing the setpoint, fan speed, turning the unit on or off, etc.
2. I have the following information in the MQTT server topics for both units.
        a. comms = {"protocol":"S21","mismatch":true,"data":"024700000000004703"}
        b. "failed-set = {"temp":21.5}"  or a similar "failed-set = {"fan":"5"}"
        c. cannot-tx = {"protocol":"S21","cmd":"DH","payload":"31303030","text":"1000"}
3. Additionally, after initial startup, on the Faikin webpage (and in Home Assistant), the Home and Outside temperature values never update until the Faikin is rebooted.

**To Reproduce**
1.	Connect the Faikin to the Dakin
2.	Power on the Daikin
3.     Sometimes settings cannot be changed at this time until a Faikin setting is modified. I will usually toggle the "Debug" option and it starts working.
4.	Wait a while (probably less than 30 minutes, although not exactly sure)
5.	Realize you cannot control the Daikin. Although the Faikin remains available on the network, the page will temporarily update with your requested change, then snap back to the original value shortly after.

**Expected behavior**
1.	Controls to continue to function immediately, as well as after 30 minutes after boot
2.	Current temperatures to update

Let me know what other information would be helpful!  Thank you for your work on this board, this seems to have incredible potential once I can get it working :)

[image]
[image]
[image]

revk, 2025-02-21:
This sounds like a communications problem.

A start is locking it down to just S21 anyway. Set nox50a and noswap settings so it does not try any others.

But also check cable, maybe make a shorter cable. Check all connections are good.

revk, 2025-02-21:
Communications issues are rare but people have found shorter cables to help in such cases. I have quite a long cable and no problems so it partly depends on the environment, electrical noise, etc, perhaps.