revk/xESP32-Faikout #654: Strange OFF/ON Faikin ?

DJJL, 2025-02-25:
Hi,
The Faikin card was acting strange today. In less than a second all its parameters were unavailable and then reconnected. This caused the air conditioning to stop with its instant start, causing an upset operation (with the sound of the air conditioning opening/closing mechanism blocking!). Do you have any idea about this malfunction? Here is the Home Assistant log confirming what happened (sorry it's in French, but we can clearly see that everything went unavailable and immediately returned to normal, wouldn't it be useful to allow a little delay before communicating with the air conditioning or other solution to avoid the conflict of commands in this type of situation?) Thank you

[image]

revk, 2025-02-25:
The web pages, settings/basic says the uptime, and will show if it rebooted, that may be a clue.

I have not see this issue though, even with a reboot, the air-con carries on and the fakin picks up the current setting. The air-con should not go off/on, etc, even in that case.

DJJL, 2025-02-25:
ok but why an untimely reboot ?

[image]

revk, 2025-02-25:
Could simply be that it updated software, though that would normally happen during the night. A reboot is not common otherwise, and without a serial debug connected it would be hard to know why.

revk, 2025-02-25:
It is not random, it is a software when new software is issued (rare).

It should not cause the aircon to go on/off or any such regardless of reason for reboot. It is odd that happened.

I would recommend s/w updates enabled, but up to you.

If it crashed for some other reason, disabling s/w updates won't help. Indeed, s/w updates are likely to mean you get better and more stable code over time.

closed 2025-02-25 by revk

DJJL, 2025-02-25:
ok, I don't change "auto upgrade" enabled,
and I'm keeping my fingers crossed that this doesn't happen too often because the air conditioning unit doesn't seem to like conflicting commands ;-)
Thank you for your answers

revk, 2025-02-25:
OK but it should not get conflicting commands even with a reboot, so no idea why.

And even so, it should not be upset - someone can press heat then cool then heat on a remote, and it should cope.

So sounds like something more complex happening. Maybe the air-con crashed and rebooted causing the Faikin to restart as a result?

DJJL, 2025-02-25:
yes normally the commands are sent "in series", a command is only activated once the previous one is finished, so no "annoyance" noise (eg: opening mechanically slowed by a closing or a "tac tac" noise from the air conditioning). In this case, it is as if the commands were sent in "parallel", or that there was no waiting for the end of the previous command to trigger the next one... (this assumption is just to understand the phenomenon). For information, this phenomenon has never been observed when the unit was used with the Infra red remote control

revk, 2025-02-25:
All commands are in series, it is as serial link, and I have never seen such an issue, so no idea what happened.

DJJL, 2025-02-25:
OK, maybe a faikin log file in the future ?
thanks

revk, 2025-02-25:
It logs lots to MQTT (e.g. s/w upgrades and commands) and serial connected debug.

DJJL, 2025-02-25:
yes, it's true. I see that in HA MQTT.
But this file "download diagnostics > config_entry-mqtt-01JFG0TSFB4AEWE5M9KRQHHT6K.json" is a bit too complex for me to decrypt it