revk/xESP32-Faikout #655: [BUG] no Faikin auto mode controls 7ca0777 2025-02-19T20:34:46

bjappler, 2025-02-25:
After getting the newest firmware update (7ca0777 2025-02-19T20:34:46) faikin auto mode controls are not available in the web front end.
In faikin preferences “nofaikinauto” is disabled. Faikins with firmeware 2024-11-30T are working fine. 

Can I downgrade again, or is there a hidden preference to show the faikin auto control section again?

[image]

bjappler, 2025-02-25:
[image]

revk, 2025-02-26:
So you have a password on the settings?

revk, 2025-02-26:
Ok, the logic was that the auto mode allowed changes to settings - BLE and so on, and password locked settings should not allow that.

I could perhaps have some setting allowing some changes still. What would you like exposed in such case? Temp target and margin?

bjappler, 2025-02-26:
Yes this would be very important for us. We are using faikin in auto mode only, and it has to be possible to change target temperature as well as margin without a password.
I use two of my Faikins in an appartment, and I want to avoid that the users could change the main settings like WLAN passwort or more settings than just target temp or margin

Could you do this again? Or could you push me the last firmware back again, that I could deactivate auto updates?

revk, 2025-02-26:
Best thing would be a setting allowing this. I can work on that and make a new beta for you.

Hopefully today.

bjappler, 2025-02-26:
This would be awesome. Thanks in advance!

bjappler, 2025-02-26:
Just one more important thing. After disabling password auto faikin mode controls are back again, but status of BLE, Huminity and Battery are not displayed. These values can be displayed all time (with or without password protection)

[image]

revk, 2025-02-26:
That is unexpected! The BLE does not show signal level in pull down either. It is correct is it?

bjappler, 2025-02-26:
just the one which is not selected.

[image]

bjappler, 2025-02-26:
after relaoding the faiking front end, its displaying all values

macmpi, 2025-02-26:
> after relaoding the faiking front end, its displaying all values

browser cache issue maybe? (ref https://github.com/revk/ESP32-Faikin/issues/642)
You mean you restarted `failkin` or just reloaded browser page?

bjappler, 2025-02-26:
I just reloaded the browser page.

revk, 2025-02-26:
Ah ok, the page serves on basis of having BLE data or not. If no BLE data (yet) it won't show when the BLE starts because those fields are not in the page.

Should only be an edge case just after a reboot.

Sorry for confusion.

macmpi, 2025-02-26:
> I just reloaded the browser page.

Thanks for confirmation.
@revk browser cache issues are frequent after OTA: this is just another example for https://github.com/revk/ESP32-Faikin/issues/642
Invalidating cache on reboot after OTA would probably solve all those edge-cases at once.

revk, 2025-02-26:
Not a cache issue. The page is served with no BLE fields because no BLE seen yet. The page is set non cached anyway.

I could serve BLE fields if BLE configured even if not seen, and I may change to do that.

revk, 2025-02-26:
Issuing a new beta shortly, do try.

closed 2025-02-26 by revk