revk/xESP32-Faikout #657: [FEATURE] HA integration: add Failkin restart button

macmpi, 2025-02-26:
**Daikin hardware**
any, but vested interest for FTXS20G2V1B or FVXS50FV1B / S21

**Is your feature request related to a problem? Please describe.**
Restarting Failkin device remotely is not easy for average user through MQTT cryptic command.

**Describe the solution you'd like**
A simple restart button on HA UI would be much more convenient.
This could be set within default MQTT integration as **MQTT button**.

**Describe alternatives you've considered**
Custom restart button can be constructed as HA template in `configuration.yaml` as per below: however this still requires user know-how.
Default integration would be much straightforward.
```
mqtt:
  - button:
      unique_id: button_faikin_restart_<faikin-name>
      command_topic: "command/<faikin-name>/restart"
      payload_press: ""
      device_class: "restart"
```

**Additional context**
see[ MQTT Button](https://www.home-assistant.io/integrations/button.mqtt/).
Note: `availability` option might be worth to set too.
(maybe reset to factory and firmware upgrades management could eventually have similar button)

revk, 2025-02-26:
I have added to web interface already.

I generally dislike reboot buttons as they are a failure. Why would you need to reboot? It is a sledge hammer approach.

But it is there on web interface now, does that help?

macmpi, 2025-02-26:
> But it is there on web interface now, does that help?

It does a bit...as long as web interface can load!
We know there are still many cases with [HTTP bug](https://github.com/revk/ESP32-Faikin/issues/299): only resort then is via MQTT...or open-up aircon and short pads.
Even when HTTP bug is eventually fixed, it would be convenient for people switching to HA to have such recovery option too at hand (they may/may not choose to display it to their own UI panel).

macmpi, 2025-03-10:
adding [this](https://github.com/revk/ESP32-Faikin/issues/299#issuecomment-2709794584) & other diagnosing logs might be helpful too to lower bar for providing meaningful debugging info