revk/xESP32-Faikout #658: [FEATURE] Support Shelly BLU H&T BLE temperature sensors

RadekHvizdos, 2025-02-27:
**Is your feature request related to a problem? Please describe.**
I am trying to add Shelly BLU H&T BLE temperature sensor to my Faikin setup. I have enabled Faikin BLE, but the temperature sensor does not show up in the BLE dropdown. I don't think it is supported (yet). This is the first time I am trying to add a BLE sensor to Faikin, so I am not sure what the right procedure is.

**Describe the solution you'd like**
Please add support for these sensors to Faikin. They report their readings in BTHome v2 format; support both plaintext and encrypted payloads if possible.

**Additional context**
[Shelly BLU H&T KB](https://kb.shelly.cloud/knowledge-base/shelly-blu-h-t)

Thanks!

revk, 2025-02-28:
We handle a number of formats, I thought BTHome v2 format was included. Needs to be unencrypted.

Only way would be my getting one.

RadekHvizdos, 2025-03-02:
Hi @revk,

Thank you for your feedback. I had a look at the relevant [file](https://github.com/revk/ESP32-BLE-Env/blob/24a371cb13fbee0e19cb7a50ce840f0eea8ab961/main/bleenv.c) and compared the logic to my sample advertisement data I have captured from Shelly BLU H&T BLE: `fc d2 44 00 20 01 64 2e 28 45 ff 00`

Your code is looking for object id's 0x02 and 0x03 for temperature and humidity properties:
```
            if (*d == 0x02 && d + 3 <= n)
            {                   // Temper
               temp = d + 1;
               d += 3;
               continue;
            }
            if (*d == 0x03 && d + 3 <= n)
            {                   // Humidity
               hum = d + 1;
               d += 3;
               continue;
```
But my advertisement data is using values **0x45** and **0x2e** instead. The [BT Home V2 format](https://bthome.io/format/) supports both encodings:

> Object id | Property | Data type | Factor | Example | Result | Unit
> 0x57 | temperature | sint8 (1 byte) | 1 | 57EA | -22 | °C
> 0x58 | temperature | sint8 (1 byte) | 0.35 | 58EA | -7.7 | °C
> 0x45 | temperature | sint16 (2 bytes) | 0.1 | 451101 | 27.3 | °C
> 0x02 | temperature | sint16 (2 bytes) | 0.01 | 02CA09 | 25.06 | °C
>
> 0x03 | humidity | uint16 (2 bytes) | 0.01 | 03BF13 | 50.55 | %
> 0x2E | humidity | uint8 (1 byte) | 1 | 2E23 | 35 | %
 
Would it be possible to extend the logic in [bleenv_gap_disc ](https://github.com/revk/ESP32-BLE-Env/blob/24a371cb13fbee0e19cb7a50ce840f0eea8ab961/main/bleenv.c#L38C1-L38C17)function to handle the additional object id's while taking into account different number of bytes and scale factors?

I can assist with debugging if needed.

Thanks,
Radek

macmpi, 2025-03-03:
Any chance [original BT Home parser](https://github.com/Bluetooth-Devices/bthome-ble/tree/main/src/bthome_ble) or [Theeng's](https://github.com/theengs/decoder) may be used for maximum compatibility and ease of maintenance, or would that be overkill?
(feature-creep warning: down the road, this may also allow Faikin to relay any other BLE sensors around, and act as a gateway!).

revk, 2025-03-03:
Yeh it is getting more complex but does the job mostly. I have an app for collecting BLE temperatures over a large area and multiple receivers reporting over MQTT.

revk, 2025-03-04:
Try latest beta.

closed 2025-03-04 by revk