revk/xESP32-Faikout #660: [FEATURE] GPIO extra usage

Ulf349, 2025-02-28:
**Daikin hardware**
All with Faikin software

**Is your feature request related to a problem? Please describe.
Problem is the motion eye from Daikin Perfera, which couldn't be used in Faikin via S21.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
My idea is to use an external PIR sensor on an gpio as input. Over mqtt this gpio will be visible, when a motion is in the room.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Not only an gpio input should be available. Maybe also an gpio output, controlled by mqtt for buzzer, beep, alarm. 

Out of topic... thank you so much for your good support and great development. Your integration is very helpful to have the commands on local network. 
BR

revk, 2025-03-03:
What would the PIR sensor actually do?

I thought the air-con adjusted louvres to allow for it or something.

Ulf349, 2025-03-04:
The PIR sensor can use as motion or presence detection in the room.
This motion detection is visible via mqtt

Result over mqtt: 
Motion detected=true
No motion=false
Only readable

Gpio output:
Alarm signal to an relay, optocoupler how ever with big load can be written via mqtt

Example: 
External device true = gpio active
External device false =gpio inactive

revk, 2025-03-04:
Surely this would be simpler with a PIR and something like a Shelly i4 or i4DC or some such. It is not really a Faikin thing.

As for output, a Shelly Plus 1 is ideal as a proper relay, and input.

Both can do DC (12V) or mains, and can run tasmota that will play with HA and MQTT nicely.

closed 2025-03-04 by revk