revk/xESP32-Faikout #661: Power icon not showing in web UI

manfi001, 2025-03-02:
**Faikin hardware**
Selfmade ESP32 

**Daikin hardware**
Perfera

**Describe the bug**
Power icon not showing in web UI

**To Reproduce**
Steps to reproduce the behavior:
Tried different browsers on different platforms - same result

**Expected behavior**
Power icon should be shown when selected in settings

[image]

revk, 2025-03-02:
That is your browser lacking that unicode character. There is a setting not to show icons.

manfi001, 2025-03-02:
I know the setting not to show icons, but I prefer the setting with icons.
Which browser would you recommend then?
So far I tried Firefox, Chrome, Edge, Samsung (tablet)
In the 8266 branch the issue is apparently solved that the text "Power" is shown instead of the missing unicode character:

[image]
I think, this looks better.
Thanks for consideration

revk, 2025-03-02:
The power icon has been in unicode for some time, you may be able to upgrade your system fonts.

If I try and allow for every device that does not have latest unicode there would be no icons, which is why I made a no icon option.