revk/xESP32-Faikout #662: [BUG]

benjamin-dcs, 2025-03-03:
**Faikin hardware**
Amazon - Faikin-S3-MINI-N4-R2

**Daikin hardware**
Daikin Perfera 2.0kw

**Describe the bug**
Home Assistant / MQTT entitities using `rpm` as unit should not be of device_class `frequency`. They should have no device class at all (as there is no device_class available for `rpm`)
```
2025-03-03 10:32:13.181 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.faikin_kantoor_comp (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'rpm' which is not a valid unit for the device class ('frequency') it is using; expected one of ['GHz', 'MHz', 'Hz', 'kHz']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-03-03 10:32:13.181 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.faikin_kantoor_fanfreq (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'rpm' which is not a valid unit for the device class ('frequency') it is using; expected one of ['GHz', 'MHz', 'Hz', 'kHz']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-03-03 10:32:13.182 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.faikin_floren_comp (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'rpm' which is not a valid unit for the device class ('frequency') it is using; expected one of ['GHz', 'MHz', 'Hz', 'kHz']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-03-03 10:32:13.182 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.faikin_floren_fanfreq (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'rpm' which is not a valid unit for the device class ('frequency') it is using; expected one of ['GHz', 'MHz', 'Hz', 'kHz']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
```

**Expected behavior**
No device_class for entities with `rpm` as unit

revk, 2025-03-03:
If I am right, it can be frequency if we use `Hz` which is an option, so really only needs removing when using `rpm`.

benjamin-dcs, 2025-03-03:
I see, than you are right, should be removed when using `rpm` :)

commit 56ef07c3 (2025-03-03): Fix #662
closed 2025-03-03 by revk