revk/xESP32-Faikout #665: [BUG]

YangChen-mechanical, 2025-03-04:
**Faikin hardware**
Please advise if this is a Faikin you purchased (e.g. from Amazon), had built, or some other device.
Ideally, please include picture of both sides of the board itself as this will help confirm the version.

**Daikin hardware**
Please advise the model of your Daikin, and how you are connected (e.g. S21 lead, X50 lead, etc).

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

YangChen-mechanical, 2025-03-06:
> If you have GND and 12V the right way that should power it.
> 
> But if you got from JLC, how did you program it with the Faikin code?

Thanks for reply.
I didn't program it. I just use this website (https://adafruit.github.io/Adafruit_WebSerial_ESPTool/) flash your code to Faikin.

The code flash to Faikin is choose from your release(https://github.com/revk/ESP32-Faikin/tree/main/ESP/release), base this(https://github.com/revk/ESP32-Faikin/tree/main/ESP/release)

Did I make some mistakes above?

revk, 2025-03-06:
OK, so you did program flash) it, good. What is the picture of the USB/serial adapter? The Faikin board looks like it has TC2030 USB pads. Do you have a picture of the back of the Faikin board?

When the flashing completed did the LED come on? It should be solid green and then start flashing.

YangChen-mechanical, 2025-03-06:
> OK, so you did program flash) it, good. What is the picture of the USB/serial adapter? The Faikin board looks like it has TC2030 USB pads. Do you have a picture of the back of the Faikin board?
> 
> When the flashing completed did the LED come on? It should be solid green and then start flashing.

I think I made some mistakes when I flashed the code. The light is on now. Thanks a lot. Thanks for spending time on a rookie like me.