revk/xESP32-Faikout #668: [FEATURE] Verbose data on the serial port to help diagnose wifi issues

Jerther, 2025-03-09:
**Daikin hardware**
My Daikin unit is a FTX12NMVJUA
I have the faikin module connected according to this thread I made: https://github.com/revk/ESP32-Faikin/issues/208

**Is your feature request related to a problem? Please describe.**
Related to https://github.com/revk/ESP32-Faikin/issues/211

**Describe the solution you'd like**
More verbose stuff about wifi activity over serial port

**Describe alternatives you've considered**
See related issue

**Additional context**
Hi!!!

The faikin module has been randomly disconnecting from the network (see related issue. However it's been quite stable for a while. It disconnected again yesterday and stayed that way until today when I tried to do some troubleshooting.

So following what I've learned through https://github.com/revk/ESP32-Faikin/issues/590 I'Ve decided to check the serial output using an [ESP-Link](https://github.com/jeelabs/esp-link) which provides a serial console over wifi.

First thing I learned: the 14V on the S403 port has not enough power to handle both the faikin module and the ESPLink (ESP8266 NodeMCU). As soon as I connected the ESPLink, I heard a whining noise from the Faikin unit, and the Faikin module rebooted. So much for diagnosing the wifi problem... Next I used an external 5V power supply. That worked fine.

Then I noticed there's not much stuff happening on the serial port after the module has finished booting. I doubt I would've found anything when the wifi wasn't working.

Is there an existing feature flag for this? Or can it be added?

Here's a picture of my setup.

[image]

revk, 2025-03-09:
Build code with debug level logging enabled and flash that!

Shows way more detail on serial.

Jerther, 2025-03-09:
I'll have a go. Is such a build just as stable?

revk, 2025-03-09:
Should be stable. You can fine tune some of the debug, eg for wifi. Tends to get a bit verbose.

Jerther, 2025-03-13:
Ok so I managed to have the project build using `make`. Now I'm trying to enable some debug logging. I have to say that at this point I'm a bit lost...

I found these lines in the 2588 lines `sdkconfig` file, and I guess this is where I can customize some flags, then I just do `make` again?

```
# CONFIG_ESP_WIFI_DEBUG_PRINT is not set  <-- I guess I should change this to =y
CONFIG_LOG_DEFAULT_LEVEL_ERROR=y  <-- then comment this
CONFIG_LOG_DEFAULT_LEVEL=1  <-- this I have no idea
```

I feel I'm close, but that's a lot to swallow. What would be a good combination to enable debugging of WIFI connection through the serial port?

revk, 2025-03-13:
I would use `menuconfig` rather than hand editing `sdkconfig`

Jerther, 2025-03-13:
Man there's a lot in there...

Anyways I found this in Component config, Log output:

```
    Default log verbosity (Debug)  --->
```

It was set to error, I set it to Debug. I guess it's enough change and I can compile and flash to get stuff over the serial port, right?

But before I do, I also found this in the RevK section, near other WiFi configurations:

```
(3600) Reset if we are offline too long
```

I'm not sure this is working. Indeed it would be nice if the faikin module reset itself when wifi is not connected for an hour. I personnaly would lower that to 10 minutes, but anyways it doesn't seem to work. Unless it means something else?

revk, 2025-03-14:
I think the problem with that setting is the Faikin stays in AP mode if no WiFi as client, so not seen as "offline". That settings is used in some other applications and works well. But I can take a look and see if that is the case for Faikin, and maybe change it.

Jerther, 2025-03-14:
10 minutes of ap mode then reboot would be fine for me. 

But I haven't thought of checking if the faikin module was in ap mode when it's wifi connection dropped. Good point.

revk, 2025-03-15:
I may have to do more testing, it looks like this should in fact work. I'll make a slight change anyway.

Jerther, 2025-03-26:
I just got around to flashing the firmware of the faikin module and I thought saw a firmware upload form on the faikin web interface, but, well, apparently I need glasses :(

Is there a way to upload a custom firmware on the faikin module OTA, aka without having to pull it out and hard-wire it? A web form would be so nice!!! Or did I miss it?

revk, 2025-03-27:
It has a setting for where it checks and loads over the air updates, which defaults to ota.revk.uk, you can change that.

Jerther, 2025-03-28:
[image]

🤣 

Through https://github.com/revk/ESP32-Faikin/issues/335 I found out that the direct URL should be http://my_server/Faikin-S3-MINI-N4-R2.bin

I do have some way of hosting a file over http so I'll give that a try.