revk/xESP32-Faikout #673: [FEATURE] New models tested and working

githrdw, 2025-03-11:
**Daikin hardware**
* Daikin Emura wall-unit
FTXJ50A2V1BW
* Daikin Perfera wall-unit
FTXM20A5V1B
* Daikin Multisplit outside-unit
2MXM50A2V1B9

**Is your feature request related to a problem? Please describe.**
I felt adventurous and ordered two Faikin chips right after the above Daikin aircons were installed.
After some struggles to remove the case and wire the Faikin I got it to work in the end!
Both models are using the `BRP069C4x` wifi chip and the `S21` port was not in use yet.
Connecting dupont directly to `S21` did not work but I some `PA 2mm PAP to fem. dupont 5-pin` cables from Ali did the trick!

**Describe the solution you'd like**
I think these two models can be added to the list of confirmed working air con units?

**Additional context**
Thank you very very much for this project ❤

revk, 2025-03-12:
Thanks.

The wiki listing models is is editable

closed 2025-03-12 by revk

dahadd, 2025-03-12:
@githrdw you had on pin 4 power ? because my model FTXJ35A2V1BW has no power on Pin 4