revk/xESP32-Faikout #681: [FEATURE] Heatover/Heatback as decimals

xudabit, 2025-03-16:
Hi, 

would it be possible to make heatover and heatback decimals? At the moment I'm using a heatback of 3. With this the ACs slow down, but run a little too long after the temperature is reached. With 4 they shut off too early. With a value like 3.5 it could be just enough to let the AC run in lower power, but it would also shutdown in a good range. This could help increase the efficiency by not shutting off too often.

Maybe this should also be done with coolback and coolover to be equal.

Seems like a minor change without complicating the current configuration to me, but maybe I'm missing something.

Thanks

revk, 2025-03-17:
It seems a minor change I agreed, except the stored values in units now would for wrong. I store decimals in settings as integers so your 3 would become 0.3.

It is fine if default value as not stored but anyone who has changed these would need to edit them.

I could force to defaults by changing name. But that would upset someone!

A name change and migrate old settings if not default, as specific code, is not impossible. But as you see it starts to get quite complex!

xudabit, 2025-03-17:
Thanks for your reply. I get that. Seams like it would be easier to add a seperate value for the decimals alone. But this would make the settings more complicated. Too bad it's that complicated. Would love to make my ACs more efficient. The Faikin-Auto with thermostat-mode works really well with my ACs besides the heatback is now fine enough.

revk, 2025-03-17:
I'll think about it, see if I can come up with a neater way

xudabit, 2025-03-17:
That would be great. Thanks.

Just an idea: would it be possible to allow two decimal places in code, but only allow one decimal place in the settings-view?

So after an Update someone would have a 0.06 instead of 6, right? This way you could identify "wrong" values after an update and correct them by multiplying them with 100.

revk, 2025-03-17:
That would mean the values no longer fit in a byte (or would be +/- 1.27) so change to uint16. I am not 100% sure that would not lose the value as change if type.

You see these things are never as simple as you expect.

I suspect the only way is a change of name, keep old name (deprecated) and if new value not set and old value set then restore as new value rescaled only.

So then do I make a generic way to do that in library settings or a special for Faikin code.

Arg!

xudabit, 2025-03-17:
I'm developing in Java. Same with my colleagues. Asking for a "minor" change which should be made in 1-2 hours. After analysing it I except 1-2 days. Now I'm on the opposite side. I don't know much about C, so I mostly understand your code, but I wouldn't feel good about making any changes. Especially because so far I don't have any option to test it easily. My ACs replace the central heating, so I can't do without any of them at the moment...

Thanks for your time spent so far.