revk/xESP32-Faikout #682: [BUG] Faikin reporting incorrect lifetime energy consumption data (2 indoor units connected to 1 outdoor unit)

RadekHvizdos, 2025-03-16:
**Daikin hardware**
2 Indoor units (FTXM25A2V1B) using S21 lead connected to common external unit (3MXM52A2V1B9)

**Describe the bug**
I have both Onecta and Faikin reporting energy consumption data to HA. I have noticed that the data does not agree. For example I was only using the Orange colored AC unit, and Onecta shows an increase of the energy usage of this unit only. Faikin shows an increase for both units, although I wasn't using the other one:
[image]

I think the different energy scale is due to Onecta doing it's yearly energy value resets, and I am fine with Faikin not doing this.

I am also not sure why both Faikin readings seem to be offset by about 3kWh, I would expect them to be the same in case they report the same value from the outdoor unit.

**To Reproduce**
Steps to reproduce the behavior:
1. Have 2 indoor units connected to 1 outdoor unit
2. Turn one of the indoor units on
3. Observe lifetime energy values of both faikin units

**Expected behavior**
I would expect Faikin would report energy consumption values which are specific to each indoor unit.

Thank you for developing this fantastic solution.
Radek

revk, 2025-03-17:
It just comes from the S21 data.

I don't know if there is an S21 command to provide the indoor unit specific usage. If so, that could be added. There may be.

There is an issue/discussion on known S21 commands which may help.

But basically I just need to know which S21 command to add.

GeeCee007, 2025-04-20:
Good day. 
Thanks for the alsmost perfect solution with Faikin.
My problem seems to be similar to Radek's s I'll post here.

I have Faikin 8 devices on 8 Perfecta indoor units, four on each outdoor unit which heat and cool the entire house.
This setup has been installed since late 2023 and are all on autoupdate.

March this year all four Perferas on the one outdoor unit reset their lifetime energy sensors back to zero.
The four units reset as follow: 

One on 14th March 21:00 CET
One on 15th March 04:00 CET
One on 25th March 19:00 CET
One on 28th March 21:00 CET

The other four Perferas on the second outdoor unit are still tracking correctly.
Both outdoor units are tracking their total energy correctly. I have these on a Shelly EM3 Pro which tracks energy used.

I have not changed anything

[image]

Could you let me know what I should do to correct this problem or if you need further information?
Thanks again
Garry

revk, 2025-04-20:
The usage does "wrap" as far as I know, not sure at what level. There is an S21 debug setting that will dump what we get from the aircon.

The top is not, by chance 6553.5 Wh?

GeeCee007, 2025-04-20:
The "Top" is indeed always 6553,5 (now where do I remember that number from)

revk, 2025-04-20:
The S21 message is 4 byte hex, so 16 bits, in 0.1kWh steps, so it will wrap at FFFF which is 6553.5.

For HA, etc, you need to tell it that it is a cyclic counter so it knows to carry on when it wraps.

closed 2025-04-20 by revk

GeeCee007, 2025-04-20:
Thanks for the prompt reply. I'll do that. Have a good holiday.

epiniguin, 2025-06-06:
I have the same problem: two indoor and one outdoor unit.
Onecta tells that lifetime energies are 8.1kWh for the first indoor unit and 6.2kWh for the second one.
Faikin tells 8.1kWh for both indoor units.
I turned on a debug. There is the same GM value for both units, and it is 4500.

`{"protocol":"S21¬Tx¬Rx","dump":"02474D343530305D03","GM":"4500"}`

revk, 2025-06-07:
I think the reading is not per unit, but one onecta probably had a different start and has seen the value cycle fewer times.

I may be wrong, what does manual for onecta say?

epiniguin, 2025-06-08:
> I think the reading is not per unit, but one onecta probably had a different start and has seen the value cycle fewer times.
> 
> I may be wrong, what does manual for onecta say?

I can't find onecta's manual. There is only a page https://www.daikin.eu/en_us/product-group/control-systems/daikin-online-controller.html.

Currently my Faikin lifetime energy is 10.9kWh, and it does not correspond with any number from Onecta.

These are Onecta data at the moment:

Indoor 1 unit: lifetime 9.9, monthly 5.8
Inddor 2 unit: lifetime 7.2, monthly 4.7
Outdoor unit: lifetime 17.1, monthly 10.5

vogtmh, 2025-12-29:
It's actually not a bug and quite easy to explain: the data reported by Faikin is per **outdoor** device, which is the only.existing metric. The numbers of two Faikins might differ, because they start at 0 when you set them up and will only get the diff. The unit you installed first will show a bigger number.

The data from Onecta on the other hand is just a calculated value. It will check how much energy was used by the outdoor device and divide it based on the runtime of each unit.

You can test it very simple: Create a template sensor with all indoor units (connected to the same outdoor unit) from Onecta. Then create a utility meter with a daily reset for both this template sensor and one of your Faikins. At the end of the day, both should show the same value.

The utility meter for two Faikins will show the same daily usage.