revk/xESP32-Faikout #683: [FEATURE] HA fans setting compatibility

kameo4242, 2025-03-17:
Hi Revk,

The possible fan settings in Daikin's integration for HA are "Auto, Silence, 1, 2, 3, 4, 5".
The possible fan settings in Faikin for HA are "auto, low, lowMedium, medium, mediumHigh, high, night".

Could it be possible to adopt the "Auto, Silence, 1, 2, 3, 4, 5" notation which would make scripts dealing with mixed faikin/daikin setup easier? 

Otherwise, just deployed 2 units out of 8; with BLE temp probes, so far, it's a game changer, thanks for your work.

Bests,

revk, 2025-03-17:
This is getting silly, the settings are based on previous comments specifically asking for settings names like this.

Does HA not have a "standard" for this? Are we following it?

kameo4242, 2025-03-17:
Oh sorry, I didn't know that. I think the code is pulling fan modes from the integration, meaning it can be whatever, so I'm not sure there is any guideline here. The Daikin integration dev chose silence, 1,2,3,4,5 but 

[HA Core - Climate entity](https://developers.home-assistant.io/docs/core/entity/climate/)

I don't know who's maintaining it on GH for HA.
[HA Climate on GH](https://github.com/home-assistant/core/tree/dev/homeassistant/components/climate
)

No problem for me; I dupped my code and used numerals for Daikin and strings for Faikin.