revk/xESP32-Faikout #684: [BUG]  switching wifi SSID does not work

Dentist27685, 2025-03-19:
**Faikin hardware**
Two Faikin modules, both giving same behavior: "Faikin-S3-MINI-N4-R2: 7ca0777 2025-02-19T20:34:46" and "Faikin-S3-MINI-N4-R2: 9f65e5b 2024-11-30T08:32:55" purchased from [Elfatronics](https://www.tindie.com/products/elfatronics/faikin-alternative-daikin-wifi-controller/)

**Daikin hardware**
S21 lead into (dunno which Daikin at the moment, but tihs is most likely immaterial to this bug)

**Describe the bug**
For a few weeks after initial install and initial setup, all worked fine for both Faikin modeules. Now something has happened with wifi1 router (maybe it has stopped handing out iIPv4 addresses) and it is technically infeasible to debug the wifi1 router. Therefore it became a necessity to switch Faikin onto "wifi2" == another wifi SSID (on another router). However, after Faiking poweroff + poweron (via disconnect + connect to S21 cable ), I can connect to Faikin AP to specify the new wifi2 SSID and password for the new wifi2, but after clicking Save, not a single packet is received by the wifi2 router (running tcpdump on it). After clicking Save, the Faikin module returns into settings screen and is only accessible on wifi1 by resolving its IPv6 address.   Other mobile devices connect to wifi2 without a glitch, get IPv4 address and function fully via wifi2, but not Faikins.
After random times the faikin1 is pingable for a while via IPv4 address and then it looses its pingability via IPv4, IPv6 pingability remains.

**To Reproduce**
Steps to reproduce the behavior:
1. disable on your wifi1 ipv4 DHCP server ( I am guessing not having functional IPv4 DHCP server part of the problem, but I have no way to verify it )
2. start `tcpdump -enn -i wlan0 -c 100` on your wifi2 router
3. ping faikin1.local on windows, getting pings : "Reply from fe80::9a3d:aeff:fef1:abcd%12: time=19ms "
5. in Edge browser go to "faikin1"  (this resolves from windows machine on the same network - probably using IPv6)
6. observe text "faikin1 [WiFi not connected]", observe that Wifi SSID + WIFI password input fields are prefilled with wifi1 SSID and stars for password. Obviously WIFI is connected via IPv6, because the faikin module responds to HTTP request. So in a way this "[WiFi not connected]" is not quite true. Maybe text "[WiFi IPv4 not connected]" would be better.
7. write into the presented  fakin1.local input fields new wifi2 SSID and wifi2 password
8. Click on Save
9. observe that in 3 to 5 seconds you are back to same screen which automatically again has wifi1 (!not wifi2!) SSID pre-filled and Wifi password is showing stars
10. observe that the above tcpdump produces no lines
11. Tell another wifi device to connect to  wifi2 SSID using same wifi2 password
12. observe that other non-Faikin devices produce many lines out of above tcpdump

revk, 2025-03-19:
Using the settings in AP mode it will try to connect before saving them. If it fails it stays on old settings.

So, you say it can be accessed via IPv6? You could edit the settings in the non AP mode and they should save.

Now, the question is why are the new settings not working.

Two thoughts.

1. There was older cost that did not like long pass phrases. Is this up to date code?

2. It only works on 2.4GHz WiFi, is the new WiFi 2.4GHz capable?

revk, 2025-03-19:
Oh and finally, if new enough for the two round contacts on the back, shorting those three times will factory reset and make it forget the old WiFi totally. See if that helps.

Dentist27685, 2025-03-19:
Thanks for the super fast response.

The wifi2 router claims to be in "band: bg", which I understand is only 2.4GHz mode. Is "only b" or "only g" mode likely to resolve the issue?

I am editing the settings in the non AP mode (by accessing the Faikin via IPv6, I believe) and the "wifi2" and pass I enter do not save after clicking "Save". "Wifi1" comes back as prefilled in the WIDI SSID field when the settings page is reloaded. In non-AP mode, is it also true that "...it will try to connect before saving them. If it fails it stays on old settings." ? The thing is that I am not seeing a single [tcpdump] packet. I'll see if I can get  Airodump-ng or wifidump to work  to show me any frames. I guess not seeing tcpdump packets is logical with the message "WiFi did not connect, try again" which shows up after those about 10 seconds after clicking Save.

Both wifi2 name and password are 9 chars long. Is that too long?  The http page claims that the  version is "Faikin-S3-MINI-N4-R2: https://github.com/revk/ESP32-Faikin/commit/7ca07776aa4411919a0b3ee68bff32727de5c00a 2025-02-19T20:34:46" so I would like to think that this includes a recent code.

Re: the tip on reset - I'll be back at the house within a couple of days, and will take a look at the physical device(s) to see if there are such round contacts.

revk, 2025-03-20:
In the normal edit mode it should definitely just save with no problem. So there is something amiss.

And 9 characters is fine. I think you said you are on recent enough code anyway.

Does changing any other setting work?

The factory reset is worth a try anyway, will go to AP mode.

pimvullers, 2026-01-04:
I had similar issues today with 1 out of three faikouts. It showed up on the router but only with ipv6. For some reason it did not get an ipv4.
By assigning it a reserved ip address (gl-inet naming for dhcp pre-defined mac-ip binding), it worked fine.

pimvullers, 2026-01-04:
It now actually gets the IP through DHCP, just in the dhcp server it is a predefined one that is handed out. So i actually guess it is a router issue.