revk/xESP32-Faikout #685: [FEATURE] Set a hysteresis in faikin auto mode

hue8i, 2025-03-20:
**Daikin hardware**
S21 Port and sending actual room temperature over MQTT  ENV messages

**Is your feature request related to a problem? Please describe.**
Now when my unit is working in faikin auto mode, I have on off switching, when the room temperatures falling and rising about 0.1 degrees.

**Describe the solution you'd like**
It would be nice I can adjust the cap between on and off in faikin auto mode. I don't want to use the thermostat mode. 

**Describe alternatives you've considered**
Maybe I can do it by changing some settings now. But I don't know which ones.

**Additional context**
It would be nice to have a deeper explaination about:
thermref -> which sensors are involved. What does a low / high value
tpredicts -> what does a low / high value 
tpredictt -> what does a low / high value 
tsample -> what does a low / high value

revk, 2025-03-20:
The manual goes in to a lot of detail on this

hue8i, 2025-03-20:
Ok that's bad in thermostat mode you loose the feature with pushtemp and switchtemp. 
So the outdoor unit is running on max  to aim the setpoint and not slow down.

eMeF1, 2025-03-20:
I'm not sure the pushtemp would help with that either, as it also only adjusts the setpoint.

I was trying to achieve the same, but to me, it seems like the power of the external unit is managed solely by Daikin based on several factors, primarily by the difference between the true setpoint and current temperature measured by Daikin (reference). As far as I understood, the only way you can influence that is by playing with temp track, temp adjust, heat over, and heat back. 

But even when you think you've achieved reasonable performance, you can start over again once the outdoor conditions start changing with the season.

I haven't tried out the Demand yet, but I sense that a dynamic setting of that parameter (based on outdoor conditions) will be the way to go.

eMeF1, 2025-03-21:
Let us know how it went

MUN0X, 2025-03-21:
> Ok that's bad in thermostat mode you loose the feature with pushtemp and switchtemp. So the outdoor unit is running on max to aim the setpoint and not slow down.

The setpoint calculation in thermostat mode can be tuned quite well, but not with pushtemp and switchtemp. Play around with heatback/heatover and temptrack/tempadjust. If set correctly the unit will scale back once the upper setpoint range is reached.

Demand will limit maximum powerdraw of the unit.

Its all documented in the manual

hue8i, 2025-03-24:
Hello, 
thanks for your comments. I tested this weekend a lot with the settings in Thermostat mode. 

If you set:
heating
set temp 22.5°C
hysteresis +/- 0.2  -> min/on  22.3°C  max/off  22.7 °C
heat over to  -3 

Happens when Set Temp < -1     //  AC Home - AC Target = -1 compressor ramps up
Happens when Set Temp -1       //  AC Home - AC Target = -1.5 compressor ramps up
Happens when Set Temp -0.6   //  AC Home - AC Target = -2 compressor ramps up
Happens when Set Temp -0.4   //  AC Home - AC Target = -2.5 compressor stay 
Happens when Set Temp + 0.1  //  AC Home - AC Target = -3  compressor ramps down fast to min reps

My proposal is to make  this setting changeable.

_Happens when Set Temp_ **+ 0.1**  _//  AC Home - AC Target = -3  compressor ramps down fast to min reps_

So the compressor ramps down earlier before it reaches the setpoint. That makes it more efficient.

MUN0X, 2025-03-24:
> Hello, thanks for your comments. I tested this weekend a lot with the settings in Thermostat mode.
>
> If you set: heating set temp 22.5°C hysteresis +/- 0.2 -> min/on 22.3°C max/off 22.7 °C heat over [...]

Now that you have heat over tuned,
 Cant you get the desired effect by changing pushtemp/switchtemp? One +0.1, other -0.1. This will move the hysteresis-range up or down relative to your setpoint.

hue8i, 2025-03-24:
No sorry,  in thermostat mode pushtemp  has no effect. You cannot go to minus. I want to go to -3 earlier than 0.1°C away from setpoint. And that's not possible right know. 

In automatik mode it is the same.

revk, 2025-03-24:
Does not sound too hard to apply to thermostats mode. If anyone fancies doing an MR.

hue8i, 2025-03-24:
So you will integrate something like this?  This would be fine. Thanks!