revk/xESP32-Faikout #692: Issue with Connecting to MQTT on Home Assistant (Synology NAS)

sjaakie09, 2025-03-26:
We are currently facing an issue with connecting to MQTT on our Home Assistant installation, which is running on a Synology NAS. We have entered the correct IP address, but we keep receiving the message: "MQTT not connecting."

We have checked several settings, but the issue persists. Could you assist us in resolving this problem, or are there any specific steps we can follow to establish the connection?

sjaakie09, 2025-03-26:
2025-03-26 10:45:13.602 DEBUG (MainThread) [homeassistant.components.mqtt.client] Error re-connecting to MQTT server due to exception: [Errno -3] Try again
2025-03-26 10:45:28.611 DEBUG (MainThread) [homeassistant.components.mqtt.client] Error re-connecting to MQTT server due to exception: [Errno -3] Try again

jinxlal, 2025-04-10:
Are you running Home Assistant in a virtual machine or in a docker container? I am running HassOS in a VM on my Synology and don’t have any issues with the faikin communicating over MQTT.

sjaakie09, 2025-05-02:
I want to buy the BLE sensor with screen flashed from you for the faikin esp. Is this possible en what is the cost total.

revk, 2025-05-02:
This maybe? https://www.tindie.com/products/revk/faikin-remote-aircon-control-display-dev-board/

sjaakie09, 2025-05-02:
No i meant this one. Pre flashed by you. 

[image]

revk, 2025-05-02:
I don't sell those, order one and reflash yourself.

sjaakie09, 2025-05-02:
Ok, do you have an manual for me so i can flash the ble

revk, 2025-05-02:
I was sure I did a link to someone that did this.

sjaakie09, 2025-05-13:
I have purchased the govee smart thermo hygrometer how do we connect it to the faikin? We tried to put the ble function on the page on and off but the ble doesnt show up..

revk, 2025-05-13:
Turn on ble, and reboots, then main page should allow you to select from BLE devices it can see.

sjaakie09, 2025-05-13:
We have tested the bluetooth of the device with connectin it to our Proxmox VM home assistant and that works. But we still dont see the bluetooth device in : Daikin Faikin back end page we have refreshed and rebooted the page aswell do you have any suggestions we use this device: 
Govee Bluetooth Hygrometer Thermometer H5075

revk, 2025-05-14:
Also try latest beta anyway, no significant changes, but gives me a baseline for work on this.

sjaakie09, 2025-05-14:
Hi,

Thank you for your quick response.

We were wondering if there's a way to diagnose whether the Faikin BLE is functioning correctly. For example, is it possible to use a smartphone as a BLE device to check if the Faikin module is broadcasting?

We’d also appreciate it if you could verify on your end whether the Faikin BLE is operational.

Additionally, we’ve tried working with the small square BLE modules from AliExpress, but we're experiencing difficulties flashing them following the provided tutorial.

Our main goal is to integrate with a Daikin air conditioner, which we’ve noticed has a temperature tolerance of approximately +2 degrees compared to the actual room temperature.

Looking forward to your advice.

sjaakie09, 2025-05-15:
Hi,

I've conducted some research and successfully connected the Govee device to the Govee app. I wanted to verify the stability and reliability of the connection, and I've gathered some initial data for reference.

Additionally, I've come across several GitHub issues that may be relevant to this topic:

https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+govee_ble%22

https://github.com/home-assistant/core/issues/142305

https://github.com/home-assistant/core/issues/129461

https://github.com/home-assistant/core/issues/138000

I've also included some screenshots from a BLE scanner app called nRF Connect, which show how the device appears during scanning.

If there's any way I can assist further, please don't hesitate to let me know.

We never have used the BLE function on the faikin how does it exactly work and how should it look when it works?

[image]

[image]
[image]
[image]
[image]
[image]
[image]
[image]

revk, 2025-05-15:
I'll check the raw data now

revk, 2025-05-15:
That raw data is indeed odd, manufacturer EC88 is not valid. Interestingly the model I have uses 0001 which is Nokia!!! They also have an Apple beacon for some reason.

I don't suppose they can provide a spec for what they send. I don't understand why they would use either 0001 or EC88, given that a Bluetooth manufacturer ID is free to get even!

Was it 26.4C and 43.0% at the time?

sjaakie09, 2025-05-15:
I'm using an additional phone—an Apple device I'm referring to as the "Apple Beacon"—to connect to the Govee device. Without it, the Govee doesn’t seem to show up for some reason. This workaround has helped in making the device visible and accessible.

Regarding the device specifications: I haven’t been able to find any detailed technical information from Govee’s side. I’ve already contacted their support to request more details about the specs, and I'm currently waiting for a response.

Yes, the data readings are correct—26.4°C and 43% humidity.

Also, did you have a chance to review the BLE integration info on GitHub and the Home Assistant site? These might offer some valuable insights:

https://github.com/home-assistant/core/tree/dev/homeassistant/components/govee_ble

https://www.home-assistant.io/integrations/govee_ble

Do you happen to know of any other tools or methods we could try to improve or troubleshoot the connection?

revk, 2025-05-15:
Try latest beta. I have no clue why Gove are not doing it properly though.

sjaakie09, 2025-05-15:
i have downloaded the latest: Faikin-S3-MINI-N4-R2: 8fc8ad1 2025-05-15T11:01:06
this the one we bought its very cheap maybe you can also try one it looks good it would be great if we could manage also for your customers and your tool:
https://eu.govee.com/products/govee-bluetooth-hygrometer-thermometer-h5075?srsltid=AfmBOorQJoX5xCHnAmP7hfNBvpiPyjwVankEtTPKBnJVvVGovNsb5Aw0

revk, 2025-05-15:
Does the latest beta work with it?

sjaakie09, 2025-05-15:
No i dont see any device when i click on the drop down menu

revk, 2025-05-15:
I don't understand the apple aspect - can you show the raw data advertisement for the Govee thing itself?

The code I have added should handle `09 FF 88 EC 00 04 08 EE 64 00` and see name `GVH5075_FA3D`.

But those details shown total 53 bytes which is too long to actually be the advertisement from the Govee.

sjaakie09, 2025-05-15:
Still no device in the list en last update was an hour ago

sjaakie09, 2025-05-15:
can you maybe show me how it looks on your end what we should see

revk, 2025-05-15:
Serial logging via TC2030 port of every advertisement maybe?

But that image still has apple stuff and still looks too long.

There is a "BLE Scanner" app which may be clearer... Looks like this

[image]

sjaakie09, 2025-05-15:
Here

[image]

revk, 2025-05-15:
New beta issued, try that.

sjaakie09, 2025-05-15:
I have updated it now i see the devices what now? :)

[image]