revk/xESP32-Faikout #693: [BUG] Mode AUTO heat_cool not allowed with CN_WIRED

davideciarmiello, 2025-03-27:
**Faikin hardware**
Custom with ESP32-S1

**Daikin hardware**
DAIKIN FAN COIL FWT04GATNMV1 connected with CN_WIRED

**Describe the bug**
The mode AUTO heat_cool does not work, as it is not present in the air conditioner

**To Reproduce**
Steps to reproduce the behavior:
1. Switch on Mode AUTO
2. Switch on Main power
3. Unit not power on

**Expected behavior**
Power on the unit

**Additional context**
There is also a problem with the shutdown. If I turn it on with another mode, set it to AUTO, and turn it off, it does not turn off

alanb2, 2025-11-28:
Hello @revk is it possible to have a switch "Unit does not have heat/cool mode"?
As stated above CN_WIRED models do not have this mode.
Thanks

revk, 2025-11-28:
Tha auto mode? Interesting. Does it cause problems, or basically just cosmetic?

alanb2, 2025-11-28:
It causes problems.
When auto mode is selected, the aircon can't switch on/off.
Also for example, by default Apple Home (via HA and Homekit bridge) powers it on auto mode and so in reality the air con does not power on.

alanb2, 2025-11-28:
Sure, no rush but it would be cool to have it.
Thanks for your work!

revk, 2025-11-28:
How many fan levels does it have? My ducted units have no auto but I think died to the fan levels setting.

revk, 2025-11-29:
I have
```

   return fantype == REVK_SETTINGS_FANTYPE_3_LEVEL_AUTO || (fantype == REVK_SETTINGS_FANTYPE_DEFAULT
                                                            && proto_type () == PROTO_TYPE_CN_WIRED);
```

Which suggests we have auto on PROTO_TYPE_CN_WIRED - so yours does not have? I assume some do.

alanb2, 2025-11-30:
Thanks for looking into this.

Let me clarify my use case and what I’m actually asking for, because I think I didn’t explain it clearly enough.

My units are hydronic fan coils (FWT04CATNMV1 and FWT02CATNMV1 on CN_WIR). On the original wall controller I only have the following modes:
- HEAT
- COOL
- DRY
- FAN

There is no native AUTO heat_cool mode on this controller.

With Faikout, the `AUTO` / `heat_cool` mode is still exposed:
- in the Web UI as a selectable mode
- in MQTT as one of the `hvac_modes`

When I select AUTO and then turn the main power on, the unit does not start.
If I turn it on in a supported mode, then change to AUTO and then switch power off, the unit does not really switch off either. This matches the fact that CN_WIR(ED) for this model simply does not support AUTO.

What I am asking for is something similar to what already exists for horizontal swing (`noswingh`):
- a configuration flag, e.g. `noauto` or `nomodeauto`
- when this flag is enabled:
  * the Web UI should not show the AUTO mode button at all
  * the MQTT discovery should not include `auto` in `hvac_modes`

This way:
- split units that really have AUTO continue to work exactly as today
- hydronic CN_WIR(ED) units like mine can be configured so that AUTO is never offered to the user and never sent to the unit

This is the same pattern already used for options like `noswingh`/`noswingv`, `nopowerful`, etc., and would solve the bug for installations where AUTO simply does not exist on the Daikin controller.

Just to avoid any confusion:
- I am not asking to change anything about the fan speeds.  The fan levels on my unit are correct as implemented in Faikout: `Low`, `Medium`, `High`, `Auto`, and `Quiet`. Fan Auto is fine and should stay.  
- this issue is not about Faikin “auto thermostat” logic; I keep that disabled because it doesn’t make sense on a hydronic system. This is only about the Daikin AUX mode `AUTO heat_cool` being exposed in UI/MQTT while my CN_WIR fan coil does not support it at all.

Thanks