revk/xESP32-Faikout #697: [BUG] - heating or cooling continues 3 degrees past the set thermostat temperature

peter-forrest, 2025-04-07:
**Faikin hardware**
Purchased from Tindie in Nov 2024.

[image]
[image]

**Daikin hardware**
FTXM25YVMA
Connected like in this photo...except plugged in properly! (this photo was from when it wasn't working properly and I forgot to take a better photo once I straightened up those connectors):

[image]
[image]

**Describe the bug**
Looks at the below screenshot - the heating has continued 2 degrees past what the thermostat has been set to. It got up to 23 degrees after the screenshot was taken. This also happens for cooling.

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Set to heat or cool
2. Set your desired temperatures
3. Wait to be thoroughly disappointed

**Expected behaviour**
When set to heat to 20.0 degrees, the heating stops at 20.0 degrees and then waits until the temperatures drops to 19.5 degrees before heating resumes.

**Additional context**
Add any other context about the problem here.

revk, 2025-04-07:
There are a lot of settings to control how it works, all in the manual.

The question is what has the temperature target been set to in the air-con unit.

What does the Faikin web interface show in this case?

Are you using the "Faikin auto" mode? Or just setting a target temp.

If the latter then your aircon is doing this, not the Faikin.

peter-forrest, 2025-04-07:
ok, I'll have a look in the manual. Is there a setting that controls how close it gets to the target before turning off?

I don't use auto because sometimes it goes too far in one direction (say it's set to heat), then it turns around and starts cooling.

I don't use the Daikin remote, I only use Faikin in Home Assistant.

peter-forrest, 2025-04-07:
When using the Faikin-auto mode, it performs quite strangely. It always shows "heat/cool" in Home Assistant on the app, it doesn't change modes. When I go from off to "heat" on the app, it goes to "cool" in the Faikin web GUI (not pictured in the video below).

https://share.icloud.com/photos/02cLkJhMnSrv5R8oJFWYz3IVw