revk/xESP32-Faikout #700: Daikin Ururu Sarara Control

4x4works-uk, 2025-04-11:
Hi there, Daikin just recommended Faikin to me! What i need is the ability to control the HUMIDIFY and HUMID HEATING functions - which Daikin's Onecta app does not have. Is this possible?

revk, 2025-04-11:
Wow!

I am not sure it is - unless it has been found in S21 commands.

sm-Fifteen, 2025-04-25:
I don't believe anyone in #408 had an FTXZ**N unit. That heatpump+heat-exchanger combo is pretty much the highest price tier Daikin has, and it doesn't look like it's available in all markets (it's not in the Americas, at least). It would take some amount of probing to figure out how that one is controlled.

[This wired remote installation guide](https://www.youtube.com/watch?v=C2fdTc3oruM) shows FTXZ-N does have an S21 connector, though, so that could be done. Actually, I thought that there was no such thing as an S21 wired remote, but the BRC073A1 exists and seems to support a range of recent models including FTXZ-N, but the user manual doesn't mention Humidify and Humid Heat settings, so it wouldn't help here.

4x4works-uk, 2025-04-26:
Hey @sm-Fifteen , thanks for the reply! Do you think there is any feasible way to achieve what I'm hoping to do? There's not a problem with controlling the other settings with devices such as Sensibo, but nothing i've seen can control the Humidify and Humid Heat settings.

sm-Fifteen, 2025-04-26:
@4x4works-uk : It's almost certainly doable, it's just that it will take some amount of help from your end to figure out what the correct knobs are, since you'd be the first person here to have these features.

4x4works-uk, 2025-04-26:
@sm-Fifteen I'm happy to help, what info would you need? BTW, Daikin told me that this is a really popular product in Japan, so it might be that any solution is valuable to Japanese users!

sm-Fifteen, 2025-04-26:
@4x4works-uk ~~If they're at the point where they're outright reccomending Faikin to customers, some official [protocol documentation](https://github.com/revk/ESP32-Faikin/wiki/S21-Protocol) would certainly be nice~~ Right now, my best bet for a starting point would be to set the indoor unit to HUMIDIFY  and check what the `F1` command returns afterwards, since it's the main command for reading the current mode and setpoint. [In order to do that](https://github.com/revk/ESP32-Faikin/blob/main/Manuals/Advanced.md), though, I believe you need to have an MQTT server configured, like people use to integrate Faikin with Home Assistant.

If you *don't* have that yet... I'm not sure there's currently an alternative. @revk : Is there any non-MQTT method for sending raw commands, like an HTTP route?

4x4works-uk, 2025-04-26:
@sm-Fifteen I don't have an MQTT server configured, no.

kjedamzik, 2025-07-23:
@gretzke already found the time to check the F1 command Response?
I think it supports different speed settings as well

sm-Fifteen, 2025-07-23:
> @kjedamzik Delivery is taking longer unfortunately, I will be able to start working on it early August it seems. How do I check command responses btw? Is there some documentation I can look into?

You have to connect to your Home Assistant MQTT server with any MQTT client, you can send commands using the `command/myFaikinUnit/send` topic, sending either one command at a time in plain text, like `F1`, or a list of commands as a JSON array, like `["Ra", "RI", "Rb", "RzB2"]`. You also need to subscribe to a separate topic to read the replies, but I forget what it was on top of my head. Will edit this message later when I can check on my home computer.

revk, 2025-07-24:
If you entered your wifi SSID and password the AP would go and it connects to your WiFi now.

svobodin321, 2025-07-24:
I could not enter the name so cannot access. How do I reset it by shorting it?

svobodin321, 2025-07-24:
Great thank you. It works now.

svobodin321, 2025-07-24:
Also can you connect Google Home and control it via voice commands?

sm-Fifteen, 2025-07-24:
> I have a question about to the specific Ururu Sarara model settings. The original remote (now broken) allowed me to  point the air sideways and keep it there. Is it possible in Faikin?

That's always possible by turning swing on and off with specific timing. Some units support snapping to preset positions as well (written with DR, read with FR, see S21 protocol wiki page), but I don't know if Faikin supports that snapping mode.

Also, see if you can't get the remote working, it would be massively useful in figuring out how to access the more unique features of the FTXZ\*\*N units, because you can just change the indoor unit configuration with it and check to see what commands now return different results.

gretzke, 2025-08-02:
Okay, so I sent the F1 command, but rx and tx are firing so many messages, I can't really say which one is the response. In tx I found a response mentioning F1, so maybe that's it?

```json
{"protocol":"S21","dump":"0246317703","F1":""}
```

sm-Fifteen, 2025-08-02:
@gretzke: F1 is the command you sent, the unit is expected to reply [G1XXXX](https://github.com/revk/ESP32-Faikin/wiki/S21-Protocol#d1f1-command).

[image]

gretzke, 2025-08-02:
Ohh yeah I had debug logs enabled haha, I wrote this script to track changes whenever I press a button on the remote
https://github.com/gretzke/s21-monitor

This is the output I get
```
----------------------------------------------------------------------------------------------------
CMD      DIR  HEX VALUE            DUMP                      TIME      
----------------------------------------------------------------------------------------------------
F1       TX                        0246317703                20:41:49  
F3       TX                        0246337903                20:41:49  
F5       TX                        0246357B03                20:41:49  
F6       TX                        0246367C03                20:41:16  
F7       TX                        0246377D03                20:41:16  
FM       TX                        02464D9303                20:41:16  
G1       RX   31334441             024731313344416103        20:41:49  
G3       RX   30FEFE00             02473330FEFE00A603        20:41:49  
G5       RX   30303B80             02473530303B809703        20:41:49  
RG       TX                        0252479903                20:41:49  
RH       TX                        0252489A03                20:41:44  
RI       TX                        0252499B03                20:41:45  
RL       TX                        02524C9E03                20:41:47  
RN       TX                        02524EA003                20:41:49  
Ra       TX                        025261B303                20:41:46  
Rd       TX                        025264B603                20:41:48  
SG       RX   41                   02534741DB03              20:41:49  
SH       RX   3030322B             0253483030322B5803        20:41:44  
SI       RX   3034312B             0253493034312B5C03        20:41:45  
SL       RX   343530               02534C3435303803          20:41:47  
SN       RX   3371312B             02534E3371312B6103        20:41:49  
Sa       RX   3535312B             0253613535312B7A03        20:41:46  
Sd       RX   363030               0253643630304D03          20:41:48  
cmd      RX   464D                                           20:41:16  
nak      RX   54727565                                       20:41:16  
```
What's interesting is that my remote has a button for the streamer and fresh air ventilations. The 4 options on my remote are On/On (Clean/Fresh), On/High, On/Off and Off/Off. However I don't see any changes to any commands when I press the button. Are there any other commands I'm missing?

sm-Fifteen, 2025-08-03:
@gretzke: FYI, most (if not all) commands reply in ASCII, so having the ASCII output tends to be more readily practical than hex, outside of some very specific cases, so converting the replies makes for results that are easier to check at a glace.

> What's interesting is that my remote has a button for the streamer and fresh air ventilations. The 4 options on my remote are On/On (Clean/Fresh), On/High, On/Off and Off/Off. However I don't see any changes to any commands when I press the button. Are there any other commands I'm missing?

[This remote?](https://kyrodimos.gr/wp-content/uploads/2021/05/ftxz6-600x600-1.png) The "Clean" part of "clean/fresh" would likely have something to do with "Streamer mode", which I understand to be queried by F6 (and likely set via D6), not sure about "Fresh" (something about letting in outside air?), might also be under the same command. Can you confirm that "Cool", "Heat", "Dry" and "Humidify" all affect the output for F1, though? Cool, Heat and Dry are all fairly well understood, but "Humidify" is the main unknown, here.

gretzke, 2025-08-03:
Yes, that's my remote. The ASCII was producing weird characters so I thought I'd convert it to hex, so it's more consistent. But I return ASCII now.

Here are the G1 results (G5 and G9 was also changing when changing these settings, I can send them too if they are interesting):
Auto => Off: 10D7 → 00D7
Off => Auto: 00D7 → 10D7
Auto => Cooling: 10D7 → 13B7
Cooling => Heat: 13B7 → 14R7
Heat => Dry: 14R7 → 31328041
Dry => Humidify: 31328041 → 3134C037
Humidify => Fan Only: 3134C037 → 16R7
Cooling => Dry Cooling: 13B7 → 13BA

Humid heating does not change the G1 code, it will remain on heat (14R7). Here the G5 code changes.
OFF => LOW: G5 30303080 → 30303880
LOW => STD: G5 30303880 → 30303980
STD => HI: G5 30303980 → 30303A80
HI => CONT: G5 30303A80 → 3030FF80
CONT => MOISTURISING: G1 14R7 → 14RA, G5 3030FF80 → 30303080

pressing humidify while heating will set humidifying to continuous mode (G5 3030FF80)

Here are the changes for cooling and humidity (cooling + humidity off: 13B7) (also saw some G9 changes here, I think these are the fan positions):
OFF => HI: G1 13B7 → 13BA, G5 30303080 → 30303C80
HI => STD: G5 30303C80 → 30303B80
STD => LOW: G5 30303B80 → 30303A80
LOW => CONT: G5 30303A80 → 3030FF80
CONT => MOISTURISING: G5 3030FF80 → 30303080 (no G1 changes like we saw with heating)

I'll figure out the fresh air supply tomorrow, it's a bit trickier because the G9 response seems to also take into account whether the 24h fresh air supply is activated or not.
Anything else that could be interesting? There are like 7 different vent positions I can set, both vertically and horizontally, is that interesting potentially?

sm-Fifteen, 2025-08-03:
@gretzke: Ok, interesting that Humidify and Humid Heat are both registered by `F1` as heating operation and Dry Cooling as regular cooling (except for Humidify having a "target temperature" of 0xC0, which is new). Probably because of a compatibility concern, as these are all cutting-edge features, whereas all the other modes have been standard on heatpumps for 20 years or more by now. Probably the same reason "Quiet fan" mode isn't reported by F1 either, even though `RG` acknowledges it. That actually has me wonder if `RB` might not recognize these new modes in a similar fashion.

> Anything else that could be interesting? There are like 7 different vent positions I can set, both vertically and horizontally, is that interesting potentially?

I think preset louver (vertical) positions are command `FR` (reading) and `DR` (for setting), but we've not found anything similar for preset fin (horizontal) positions.

It's likely your unit would have a lot of commands we've never seen before. @Sonic-Amiga had a [collect-profile.py](https://github.com/revk/ESP32-Faikin/blob/main/Tools/Simulators/collect_profile.py) script he used to scan all known commands and use dumps of these to test how official controllers would react when encountering them. That would probably be helpful to figure out a direction.

sm-Fifteen, 2025-08-03:
> Here are the changes for cooling and humidity (cooling + humidity off: 13B7) (also saw some G9 changes here, I think these are the fan positions):
> * OFF => HI: G1 13B7 → 13BA, G5 30303080 → [...]

Also, interesting that "OFF" and "MOISTURISING" modes have the same reading on F5, but all the other settings are different. G1 13B7 → 13BA, would mean that the fan has switched from high speed to automatic setting, but the observed mode (according to F1) is still Cooling. I don't have enough information of how the whole humidity thing works to guess, but looking at the [service manual], that might be some kind of "humidifier unit fan speed" of sorts.

gretzke, 2025-08-03:
Here is the output of the python script you sent me, I needed to add authentication and it seems to get stuck on VS00M
```
Connected with result code Success
F8 30 31 00 00 
FC 41 32 41 30 
F2 3C 3B 00 92 
F3 30 FE FE 00 
F4 30 00 80 00 
FB 30 30 30 30 
FG <NAK>
FK <NAK>
FN <NAK>
FP <NAK>
FQ <NAK>
FR <NAK>
FS <NAK>
FT <NAK>
FV <NAK>
M 31 30 44 33 
V 30 30 45 30 
VS000M 
```

sm-Fifteen, 2025-08-03:
> Here is the output of the python script you sent me, I needed to add authentication and it seems to get stuck on VS000M
> 
> ```
> Connected with result code Success
> F8 30 31 00 00 
> ```

Woah, wait, what? `F8 01 \x00 \x00`?? That's *completely* unexpected, that would mean like "S21 protocol v1", which everyone thought was a) probably fairly old and no longer in use and b) had been completely skipped because all the units we checked either returned `F8 0 \x00 \x00 \x00` or `F8 02 \x00 \x00`.

This is fascinating! It also means we probably won't have to look *super* far to find the commands and signals used by your unit, given your unit stops recognizing F commands at either FC or FG (do try seeing if FC returns anything, though), and likely won't reach all the way into the FU\*\* or FY\*\* range.

----

Also, I've looked a bit more into it and I'm seeing how Ururu Sarara moisturizing and drying works, it's pretty interesting.

https://www.youtube.com/watch?v=-J13TB3tWiY

[image]

"Sarara Drying" (unlike regular drying, called "Cool Drying") works by heating and cooling parts of the indoor unit simultaneously in order to force condensation. That's something to be on the lookout for in the new return values. Humidifying seems to use condensation from the outside air on a "humidifying rotor" passing through a new "humidifying hose", which is also probably going to be reported somewhere.

sm-Fifteen, 2025-08-03:
> ```
> F8 30 31 00 00 
> FC 41 32 41 30 
> F2 3C 3B 00 92 
> ```

Also, your result for [F2](https://github.com/revk/ESP32-Faikin/wiki/S21-Protocol#f2-command) foes indicate that humidifying operation is available (byte 3 is 92, so `0b1001_0010`). I'd have to check to see if the matrixes in our doc (which Sonic-Amiga decompiled from the mobile app) match what your service manual says. That would give credance to that one bit in F5 being humidifier control. Interestingly, "MOISTURIZE" isn't reported in any of the matrices either.

sm-Fifteen, 2025-08-03:
> I see a response FC on my end (A2A0) but do not see FG. so it probably stops after FC. To me it seems like MOISTURIZE is a different mode entirely because on the remote I also see that it changes a bunch of other things

Think you could set it to "Sarara dry" and/or Humidify/Moisturize and [probe all the R\* commands](https://github.com/revk/ESP32-Faikin/wiki/S21-Protocol#sensor-reading-commands-r) for me? I'm specifically interested in RB (unit mode), RF,  RW (unknown) and Rb ("compressor pull", given how "Sarara Dry" is stated to work). You might also have R\* commands that the other units we've seen don't ACK, so it might be worth just trying all letters, just in case. R commands are only for reading, so there's no risk in doing that.

gretzke, 2025-08-04:
Okay so here are the codes I see when I set it to dry:
```
G1       RX   31328041             024731313280419C03        15:00:09  
G2       RX   3C3B0092             0247323C3B00928203        15:00:09  
G3       RX   30FEFE00             02473330FEFE00A603        15:00:09  
G4       RX   30008000             024734300080002B03        15:00:09  
G5       RX   30303C80             02473530303C809803        15:00:09  
G8       RX   30310000             02473830310000E003        15:00:09  
G9       RX   ACAC6730             024739ACAC67306F03        15:00:09  
GA       RX   7020                 024741373032305103        15:00:09  
GB       RX   0000                 024742303030304903        15:00:09  
GC       RX   A2A0                 024743413241306E03        15:00:09  
SA       RX   1                    02534131C503              15:00:10  
SB       RX   3                    02534233C803              15:00:10  
SC       RX   220+                 0253433232302B5503        15:00:10  
SD       RX   000                  0253443030302703          15:00:10  
SE       RX   750                  0253453735303403          15:00:10  
SF       RX   00                   0253463030F903            15:00:10  
SG       RX   A                    02534741DB03              15:00:10  
SH       RX   522+                 0253483532322B5F03        15:00:10  
SI       RX   570+                 0253493537302B6303        15:00:10  
SK       RX   660                  02534B3636303A03          15:00:10  
SL       RX   230                  02534C3233303403          15:00:10  
SM       RX   334+                 02534D3333342B6503        15:00:10  
SN       RX   471+                 02534E3437312B6803        15:00:10  
SW       RX   00                   02535730300A03            15:00:11  
SX       RX   522+                 0253583532322B6F03        15:00:11  
Sa       RX   022+                 0253613032322B7303        15:00:11  
Sb       RX   400                  0253623430304903          15:00:11  
Sd       RX   820                  0253643832305103          15:00:11  
Se       RX   750                  0253653735305403          15:00:11  
Sg       RX   1                    02536731EB03              15:00:11  
Sz       RX   72C3                 02537A37324333AC03        15:00:11  
```

Here are the codes when I set it to humidify
```
G1       RX   3134C037             0247313134C037D403        15:01:43  
G2       RX   3C3B0092             0247323C3B00928203        15:01:43  
G3       RX   30FEFE00             02473330FEFE00A603        15:01:43  
G4       RX   30008000             024734300080002B03        15:01:43  
G5       RX   3030FF80             0247353030FF805B03        15:01:43  
G8       RX   30310000             02473830310000E003        15:01:43  
G9       RX   ACAC6730             024739ACAC67306F03        15:01:43  
GA       RX   7020                 024741373032305103        15:01:43  
GB       RX   0000                 024742303030304903        15:01:43  
GC       RX   A2A0                 024743413241306E03        15:01:43  
SA       RX   1                    02534131C503              15:01:44  
SB       RX   4                    02534234C903              15:01:44  
SC       RX   910+                 0253433931302B5B03        15:01:44  
SD       RX   000                  0253443030302703          15:01:44  
SE       RX   750                  0253453735303403          15:01:44  
SF       RX   00                   0253463030F903            15:01:44  
SG       RX   7                    02534737D103              15:01:44  
SH       RX   022+                 0253483032322B5A03        15:01:44  
SI       RX   041+                 0253493034312B5C03        15:01:44  
SK       RX   801                  02534B3830313703          15:01:40  
SL       RX   701                  02534C3730313703          15:01:40  
SM       RX   491+                 02534D3439312B6903        15:01:40  
SN       RX   510+                 02534E3531302B6203        15:01:40  
SW       RX   00                   02535730300A03            15:01:41  
SX       RX   011+                 0253583031312B6803        15:01:41  
Sa       RX   022+                 0253613032322B7303        15:01:41  
Sb       RX   000                  0253623030304503          15:01:41  
Sd       RX   000                  0253643030304703          15:01:41  
Se       RX   850                  0253653835305503          15:01:41  
Sg       RX   0                    02536730EA03              15:01:41  
Sz       RX   7264                 02537A37323634A003        15:01:41  
```

sm-Fifteen, 2025-08-04:
@gretzke : Thanks, will take a closer look when I'm back from work.

One thing I'm noticing, though: Most units have a 3 minutes cooldown on the compressor when switching modes or shutting down, and `Sg` in your second dump indicates that the compressor wasn't running. I don't know if that's normal for Humidify mode, but just to be sure, could you leave your unit on humidify mode for 3-5 minutes and take another reading?

gretzke, 2025-08-04:
Dry after 5 minutes:
```
G1       RX   31328041             024731313280419C03        15:41:58  
G2       RX   3C3B0092             0247323C3B00928203        15:41:58  
G3       RX   30FEFE00             02473330FEFE00A603        15:41:58  
G4       RX   30008000             024734300080002B03        15:41:58  
G5       RX   30303C80             02473530303C809803        15:41:58  
G8       RX   30310000             02473830310000E003        15:41:58  
G9       RX   A8AC7130             024739A8AC71307503        15:41:58  
GA       RX   8020                 024741383032305203        15:41:58  
GB       RX   0000                 024742303030304903        15:41:58  
GC       RX   A2A0                 024743413241306E03        15:41:59  
O1       RX   31328041             024F31313280419C03        15:36:55  
O5       RX   30303C80             024F3530303C809803        15:36:55  
SA       RX   1                    02534131C503              15:41:59  
SB       RX   3                    02534233C803              15:41:59  
SC       RX   020+                 0253433032302B5303        15:41:59  
SD       RX   000                  0253443030302703          15:41:59  
SE       RX   750                  0253453735303403          15:41:59  
SF       RX   00                   0253463030F903            15:41:59  
SG       RX   A                    02534741DB03              15:41:59  
SH       RX   002+                 0253483030322B5803        15:41:59  
SI       RX   080+                 0253493038302B5F03        15:41:59  
SK       RX   850                  02534B3835303B03          15:41:59  
SL       RX   750                  02534C3735303B03          15:41:59  
SM       RX   311+                 02534D3331312B6003        15:41:59  
SN       RX   311+                 02534E3331312B6103        15:41:59  
SW       RX   00                   02535730300A03            15:42:00  
SX       RX   002+                 0253583030322B6803        15:42:00  
Sa       RX   032+                 0253613033322B7403        15:42:00  
Sb       RX   400                  0253623430304903          15:42:00  
Sd       RX   230                  0253643233304C03          15:42:00  
Se       RX   860                  0253653836305603          15:42:00  
Sg       RX   1                    02536731EB03              15:42:00  
Sm       RX   860                  02536D3836305603          15:41:38  
St       RX   030                  0253743033304A03          15:40:08  
Sz       RX   72C3                 02537A37324333AC03        15:42:00  
WG       RX   A                    02574741DB03              15:38:36  
```

Humidify after 5 minutes:
```
G1       RX   3134C037             0247313134C037D403        15:35:39  
G2       RX   3C3B0092             0247323C3B00928203        15:35:39  
G3       RX   30FEFE00             02473330FEFE00A603        15:35:39  
G4       RX   30008000             024734300080002B03        15:35:39  
G5       RX   3030FF80             0247353030FF805B03        15:35:39  
G8       RX   30310000             02473830310000E003        15:35:39  
G9       RX   A8AA7630             024739A8AA76307803        15:35:39  
GA       RX   8020                 024741383032305203        15:35:39  
GB       RX   0000                 024742303030304903        15:35:39  
GC       RX   A2A0                 024743413241306E03        15:35:40  
SA       RX   1                    02534131C503              15:35:40  
SB       RX   4                    02534234C903              15:35:40  
SC       RX   910+                 0253433931302B5B03        15:35:40  
SD       RX   000                  0253443030302703          15:35:40  
SE       RX   750                  0253453735303403          15:35:40  
SF       RX   00                   0253463030F903            15:35:40  
SG       RX   7                    02534737D103              15:35:40  
SH       RX   002+                 0253483030322B5803        15:35:40  
SI       RX   571+                 0253493537312B6403        15:35:40  
SK       RX   801                  02534B3830313703          15:35:40  
SL       RX   701                  02534C3730313703          15:35:40  
SM       RX   491+                 02534D3439312B6903        15:35:40  
SN       RX   491+                 02534E3439312B6A03        15:35:41  
SW       RX   00                   02535730300A03            15:35:41  
SX       RX   011+                 0253583031312B6803        15:35:41  
Sa       RX   512+                 0253613531322B7703        15:35:41  
Sb       RX   000                  0253623030304503          15:35:41  
Sd       RX   000                  0253643030304703          15:35:41  
Se       RX   470                  0253653437305303          15:35:41  
Sg       RX   0                    02536730EA03              15:35:41  
Sz       RX   7264                 02537A37323634A003        15:35:41  
```

sm-Fifteen, 2025-08-04:
Ok, so no compressor in Humidify mode seems like the expected vehavior, then. Good to know.

Return commands Sm and St are new, those will be worth monitoring to try and figure out what they mean. I'm also seeing O1, O5 and WG, which look like duplicates of G1, G5 and SG respectively? Are those mistakes/a bug on your end or did they actually get returned in your tests?

Otherwise, the third byte on F1, the third byte on F5, the retirn value of Re (already known to be indoor humidity) and for good measure these new Rm and Rt seem like the most promising ones to keep an eye on. I'm also wondering if any of these new features might show in RzB2 (Read unit state) and RzC3 (Read system state).

What I had built when I was doing my own reverse-engineeiring efforts was a script that cycles through a list of commands every 30 seconds, and saves each cycle as a row in a CSV file, with one column per command, hiding values that hadn't changed since the last cycle. I would leave that script running for several hours at a time to try and spot trends in the replies of any unknown command. It wasn't very complicated and I could pull it up if needed, but it looks like it probably wouldn't be too hard to alter what you built to have a similar effect.

kjedamzik, 2025-08-04:
The humidification system operates independently from the compressor, suggesting it may be implemented through separate fan modes. The outdoor unit features a desiccant-based disc with a dedicated motor for rotation, along with a fan and heater assembly designed to transfer moist air to the indoor unit. It has a separate vacuum cleaner like hose to the indoor unit for that: https://www.youtube.com/watch?v=-J13TB3tWiY

maybe interesting for decoding sensor packets: 
* it has a humidity sensor, the service manual mentions one on the indoor unit pcb , 
* also a indoor heat exchanger thermistor  
* outdoor unit has thermistorsas well:
* outdoor temperature
* outdoor heat exchanger
* discharge pipe
* and a "liquid pipe" thermistor

[image]

[page 12](https://www.daikin.eu/content/dam/document-library/operation-manuals/ac/split/ftxz-n/FTXZ-N_3P338603-1G_Operation%20manual_English.pdf)

Some other noteworthy differences between the FTXZ and other models:
- the Intelligent eye has 3 areas (not 2 as with C/FTXM-A and FTXTM-S) but not the fancy ir imaging thing that other units have might be important for sensor readings
- it has a self cleaning filter and the cleaning operation can be triggered on the remote
- its the only unit that does not come with a wifi module nowadays so getting faikin to support it would be very very nice ❤️

sm-Fifteen, 2025-08-04:
> Is Wa a valid response? I also saw that one in my MQTT explorer, so it shouldn't be a parsing bug
> ```
> {"protocol":"S21","dump":"0257613032322B7303","Wa":"022+"}
> ```

W isn't a prefix I've ever seen before, but I don't know if anyone ever tried testing arbitrary 2 letter combinations, in case there's one that on its own can do anything bad. Do you know what command it was that you sent before getting that reply?

gretzke, 2025-08-04:
I commented out the Rz* commands and now I see less unusual responses. Still getting some unexpected responses, Gt (same response as G4) and gB (same response as GB). We found out that the F commands go up to FC on my device, any way for me to figure out what R commands my device supports?

sm-Fifteen, 2025-08-04:
> I commented out the Rz* commands and now I see less unusual responses. Still getting some unexpected responses, Gt (same response as G4) and gB (same response as GB). We found out that the F commands go up to FC on my device, any way for me to figure out what R commands my device supports?

Every device we've tested so far has supported all known R\* commands. Rz\*\* is less well charted, and Rz alone returning something is a sign that there's a missing length check on these commands, which is why Rz alone never returns anything reliable.

Really odd how you seem to be recieving replies that don't match the expected format for the commands you're sending, almost as if there was noise on the line or a bad contact. Maybe your Faikin unit isn't plugged properly?

gretzke, 2025-08-04:
I'm sending them in one array `["F1", ..., "FC", "RA", ..., "Rg", "RzB2", ..., "Rz72", "V", "M", "VS000M"]`

sm-Fifteen, 2025-08-04:
@gretzke  F6 and F7 didn't work before either, since replies G6 and G7 didn't show up in your dumps previously, so that's all normal. V and M, I think Faikin's send MQTT command requires a minimum length of 2 bytes or it will reject the command, if I'm not mistaken? But those don't matter as much. The rest all looks as expected.

gretzke, 2025-08-04:
Okay makes sense, what are the next steps now? Anything else on your end you'd be interested in seeing?

sm-Fifteen, 2025-08-04:
@gretzke: Long timelines would probably be most be useful. [Here's the script](https://gist.github.com/sm-Fifteen/65bdb0117e03394ab5970a7c8e1bbfbb) I was using last year as a reference, though you could probably run it mostly as-is. It's not my proudest code and has proven slightly more unstable than I'd like (mostly timing issues because I'm really not used to dealing with async MQTT), but it generates the kind of file I like to deal with.

You start the script, maybe do a `tail -f` on the output file, fiddle with the remote a little, write down what you did and roughly when, go about your day, come back and fiddle with the remote some more, etc. It gives all readings time to settle or show variation, and lets you cross-reference the values each command returned at roughly the same time. To reduce visual noise, I also don't log values that haven't changed.

You could probably remove the `rz_commands` from the list or replace it with like `["RzB2", "RzC3", "Rz52", "Rz72"]` like you did earlier, since it's unlikely the knobs we're looking for are in there right now. You may also want to add a few more F\* commands to the list, like F5 and F9, maybe F4.

sm-Fifteen, 2025-08-05:
@gretzke : Faikin auto-mode works by monitoring a temperature sensor (usually the indoor unit's, but it can be an external one) and "manually" changing the temperature setpoint to try and keep it at a specific target temperature. The actual Auto setting on Daikin units works fine and can actually modulate PI demand control and such, so I don't personally think there's much point in enabling it unless you have an external thermometer for it to track.

Daikin's built in method of hitting and keeping a temperature setpoint when heating or cooling is to calculate a target temperature (`RX`,  which is usually 2 degrees higher than the setpoint to account for rising heat) and for each half-degree Celcius of difference between the target and the measured indoor temperature (`RH`), it will count one point of demand (`Rb`), and adjust the fan and compressor frequency in accordance. Demand is 2 when the measure and the target are equal, IIRC, so the unit only actually stops if the target is exceeded by one or two degrees. The built-in auto mode on top of that switches mode when the target is overshot by something like 4 degrees, though I'd have to check to make sure.

sm-Fifteen, 2025-08-22:
> > Fan mode cannot bee set with the remote with a moisturisation mode
> 
> Fan mode (in the manual referred to as airflow rate) cannot be set when any drying mode is set (dry or dry cooling) including the moisturizing mode. In the humidify mode (humidify or humid heating) the fan mode can be set, except when in moisturizing mode

Huh, it hadn't actually occured to me that your unit had no mode cycling button or any indication that it could be set in "fan only with no compressor" mode. Fan mode is "D1/G1 16**" and it [didn't come up in your tests before](https://github.com/revk/ESP32-Faikin/issues/700#issuecomment-3148597372). How does your unit respond to something like `D116R7` or similar?

> > Does Auto mode do anything with humidity
> 
> I don't think so, when in auto mode, the remote doesn't allow me to change the humidity

It won't let you set a target humidity level at all? Interesting...

> > unless your remote can read that from the device, I think the manual hinted at that
> 
> From what my tests show the remote can only read the temperature, humidity and power data from the device. I set the device to heating, then in home assistant I changed it to cooling. When reading the data from the device, it did not update what was shown on the remote to me.

That's unfortunate for our testing, but on the other hand, it's still really impressive that it can read from the unit like that at all.

> Also here is another fun thing I figured out, when heating is activated, pressing the humidify button toggles between heating and humid heating, to activate pure humidification, I need to switch to cooling first, might be a quirk with the remote though.

I think I'm going to need a state machine chart to keep track of all these modes, it's making my head spin.

> Also it seems like I cannot activate moisturizing mode in pure humidfy mode

That's weird, it's the one mode where you would expect it to be available. Is that limitation described in your user manual?

> Here is my data changing the modes regarding heating and humidify (this one also includes F2 data now, the data for cooling/dry did not include that data)

Will look into those over the weekend, thanks!

sm-Fifteen, 2025-10-06:
I'm aware I said "over the weekend" and have not given any sign of life for over a month; I'm just bad at following up on this sort of thing. I haven't forgotten, though!

gretzke, 2025-10-06:
No worries, I actually missed your previous message, so good reminder for me to follow up here.
> Huh, it hadn't actually occured to me that your unit had no mode cycling button or any indication that it could be set in "fan only with no compressor" mode. Fan mode is "D1/G1 16**" and it https://github.com/revk/ESP32-Faikin/issues/700#issuecomment-3148597372. How does your unit respond to something like D116R7 or similar?
In the [photo of the remote] the fan mode can be set using the button in the lower right. All modes have to be set explicitly, so they can't really be cycled. I will check the response to D116R7.

>  It won't let you set a target humidity level at all? Interesting...
According to the manual this sets the device to heat/cool based on the current temperature, the manual explicitly states that the humidity settings are not available in auto mode

> That's weird, it's the one mode where you would expect it to be available. Is that limitation described in your user manual?
This seems to be a special operation, that's why it's limited maybe. To quote the manual: "Keeps the humidity high and does not blow air directly on the body to make the operation gentle to your skin. The operation is recommended when you are worries about the dryness of your skin. The airflow rate cannot be changed."

zen2, 2025-11-23:
In case that can help understand the different control modes and related limitations:
https://www.daikin.eu/content/dam/document-library/operation-manuals/ac/split/ftxz-n/FTXZ-N_3P338603-1G_Operation%20manual_English.pdf

sm-Fifteen, 2026-01-19:
@4x4works-uk Ah, I believe you have the wrong response message, that looks like the periodic Faikout status message. You'll want to add a topic filter to only keep `info/myFaikoutUnit/rx` or `info/#/rx`, which contain the replies to commands sent manually.

sm-Fifteen, 2026-01-19:
Yeah, ok, that matches with what we had. The "moisturizing mode" appears undistinguishable from OFF with the commands we have currently, and otherwise it's just 3 settings of humidity control.

I'll check a few things with how Home Assistant expects moisture settings to be sent, and I'll try to have an outline of how it should be implemented in Faikout later today or tomorrow.

sm-Fifteen, 2026-01-20:
> Humidity off (in heating mode only)
>    "protocol": "S21",
>     "dump": "024735303030808C03",
>     "G5": "000"
> 
> Humid Heating LOW
>   "protocol": "S21",
>     "dump": "024735303038809403",
>     "G5": "008"
> 
> Humid Heating STD
>   "protocol": "S21",
>     "dump": "024735303039809503",
>     "G5": "009"
> 
> Humid Heating HI
>     "protocol": "S21",
>     "dump": "02473530303A809603",
>     "G5": "00:"
> 
> Humid Heating CONSTANT
>   "protocol": "S21",
>     "dump": "0247353030FF805B03",
>     "G5": "00ÿ"

Wait, I actually just noticed that the humidity settings values returnes by @4x4works-uk in humid heating mode this week are different from the values returned by @gretzke this summer in (dry?) cooling mode. Here we have 0x38, 0x39 and 0x3A, but before it was 0x3A, 0x3B and 0x3C.

sm-Fifteen, 2026-01-20:
Re-reading the thread, it turns out that this switch up between dry and humidify [was reported before](https://github.com/revk/ESP32-Faikout/issues/700#issuecomment-3148597372) and I actually failed to notice it:

> Humid heating does not change the G1 code, it will remain on heat (14R7). Here the G5 code changes. OFF => LOW: G5 30303080 → 30303880 LOW => STD: G5 30303880 → 30303980 STD => HI: G5 30303980 → [...]

| G5 (third byte) | Heat mode name     | Cool mode name   |
|-----------------|--------------------|------------------|
| 0x38            | Humid heating LOW  |                  |
| 0x39            | Humid heating STD  |                  |
| 0x3A            | Humid heating HI   | Dry cooling LOW  |
| 0x3B            |                    | Dry cooling STD  |
| 0x3C            |                    | Dry cooling HI   |
| 0xFF            | Humid heating CONT | Dry cooling CONT |

So... does that mean the byte is some form of target moisture level (except for continuous)?

> I should also report that, when setting up the Ururu Sarara in Home Assistant with the standard Daikin WiFi board, we have a STREAMER function on/off. However, once the Daikin WiFi board has been replaced with Faikout, the STREAMER option is no longer present.

@4x4works-uk Streamer mode is (as far as we know) controlled by D6/F6, and gretzke made it seem like that command wasn't supported on that unit. If you're saying that it wa supported before, we'd need to snoop on the signal between the official wi-fi adapter and the indoor unit to know what alternate signals are being used to probe for it and to control it. I'm not equipped to do that myself, so I can't help you with that, but I recall that RevK once had [dual-port Faikout boards](https://github.com/revk/ESP32-Faikout/blob/main/Manuals/Advanced.md#debug) for that sort of purpose.

4x4works-uk, 2026-01-20:
@sm-Fifteen "pressing humidify while heating will set humidifying to continuous mode (G5 3030FF80)" - this is not the case with my units. If I press HUMIDIFY while the unit is in HEAT mode, it enters HUMID HEATING and returns to the most recently set HUMIDIFY state.

Re STREAMER mode, I have 3x Ururu Sarara units, 2x with standard Daikin WiFi boards and 1x with a Faikout.  All 3x are configured in Home Assistant and the 2x with standard boards show a streamer switch in HA (albeit disabled) automatically.

Let me know if any further data is needed from my end.

revk, 2026-01-21:
This looks promising, we need to check we can control it as well as just report it I guess, but should be able to. I suppose now you want a humidify setting on Faikout :-)