revk/xESP32-Faikout #701: [FEATURE]

github0123456789, 2025-04-12:
I would like to have a new switch or button in the Faikin console to manually disable Auto mode. There are days when I prefer not to follow the same schedule—for example, when the house is already sufficiently warm or will naturally be heated by the sun.

Currently, I need to set the time values to 00:00 hours to deactivate Auto mode, which is not very convenient. Additionally, even with the 'Auto' switch disabled, the air conditioning always starts at the time set in the 'On' field. After conducting multiple tests at different times with 'Auto' switched off, the device continues to start automatically. This behavior is unexpected and may indicate a bug in the faikin?

Could you please verify this issue and or implement a user-friendly option, such as a new switch or button, to easily disable Auto mode? Thank you very much in advance!

[image]

revk, 2025-04-13:
There are two things

- Times - if set then the unit goes on/off on those times
- Auto power - this is separate and turns on at other times if temperature too far out, and turn off it staying within target temp.

revk, 2025-04-13:
OK I have made a beta, can you test for me as my HA is busted. You need the *extra HA switches* enabled to see the extra switch. It is an overall enable for *Faikin auto mode* including times, etc.

closed 2025-04-13 by revk

github0123456789, 2025-04-13:
Switch shows up but somehow a strange behaviour. Pressing Switch Auto mode does not give any logging in HA. It cannot stay on and does not communicate with Faikin. Nothing switched on on faikin. Changes on both sides are not effected. 
For comfort mode switch on HA changes the faikin to on, it stays on at my faikin but it switches to of on HA. 
[image]

revk, 2025-04-13:
OK I'll have to try and get my HA working and let you know.

revk, 2025-04-14:
Yeh, my HA is buggered, I need to re-build it I expect.

revk, 2025-04-14:
OK try new beta

github0123456789, 2025-04-14:
I tested this in HA, and it worked flawlessly. Thank you so much!